espressif/esp-idf #5388: ESP32 S2 chip not able to talk over I2C with MFi chip

iDarshan, 2020-06-03:
I am using esp-idf master branch along with esp-homekit-sdk on ESP32 S2 WROOM chip. MFi 2.0 chip is connected to ESP32 S2 over I2C using GPIO 10 and 11. On bootup, ESP32 S2 prints logs that its trying MFI read over I2C SW and its failing after 200 retries. Same hardware connection works with ESP32 chip on release/v3.0 branch of esp-homekit-sdk and 463a9d8b7 commit of esp-idf.

So it seems like something is broken in master branch of esp-idf which breaks I2C communication with MFi chip.

shahpiyushv, 2020-08-14:
@iDarshan , can you try changing the MFi I2C data rate from 400Kbps to 100Kbps and see if it solves the issue?

mythbuster5, 2020-10-12:
Can you provide more information and source code?

iDarshan, 2020-10-12:
@mythbuster5 I have already given information about which codebase I am using in the description. You can use any example application on top of that to reproduce the issue.

shahpiyushv, 2020-10-13:
@iDarshan since this is about the MFi variant of the esp-homekit-sdk, esp-idf GitHub issues isn't the right place for this issue anyways. Even then, just to add, the release/v3.0 was never intended for ESP32-S2. Please check the documentation again and if you face any issues, file it on the MFI variant of esp-homekit-sdk.

Alvin1Zhang, 2021-08-25:
Thanks for reporting, will close now due to short of feedback, feel free to reopen or create a new issue in the right issue tracker. Thanks.

closed 2021-08-25 by Alvin1Zhang