espressif/esp-idf #5391: Wifi unable to connect again after a long run with AP turned off. Memory Error.

enricop, 2020-06-04:
## Problem Description

After 12 hours running with a Wifi AccessPoint turned off, the wifi component fails allocating RAM for connecting again to the AP when it is turned back on.

In our setup we have mqtt component repeatedly trying to use the network (connecting to the broker) even if wifi is not connected

We are using esp-idf commits on branch r4.0 just before r4.0.1. It's :  b0f053d82

## Debug Logs

[Long_Run.txt](https://github.com/espressif/esp-idf/files/4728249/Long_Run.txt)
"minimum internal_dma free mem: 2336" is printed every time on the "disconnect" event using the print function suggested in #3592 

## SdkConfig

[sdkconfig.txt](https://github.com/espressif/esp-idf/files/4728261/sdkconfig.txt)
out "CONFIG_SPIRAM_MALLOC_RESERVE_INTERNAL" setting is 32K

##Ideas

Could it be that these continuous connection/disconnection retries increases fragmentation of the internal memory?
So that even if 2k of internal memory are free, even 76bytes are not allocatable....

Or another reason could be memory leaks...of components or of application code. But we do not notice memory leaks in our application code when it is connected correctly to the wifi.

thanks for your support

espxiehang, 2020-06-04:
Hi @enricop ,Can you tell me the detailed steps to reproduce this problem？
You can use ”heap_caps_get_free_size( MALLOC_CAP_DEFAULT | MALLOC_CAP_INTERNAL )“ where the problem occurs to view the remaining interval memory of the system.

enricop, 2020-06-04:
The step for trying to reproduce are the following:
- set a valid ssid and password in wificonfig
- turn off the AP
- trigger connections attempts periodically to the AP (every x seconds) causing a disconnect event
- leave the esp32 in this state for 13 hours or so.
- check the console for wifi lib memory errors

Regarding the print function to use you have this in the disconnect event, as you can see from the previously attached logs.

`std::cout << "minimum internal_dma free mem: " << heap_caps_get_minimum_free_size(MALLOC_CAP_8BIT|MALLOC_CAP_DMA) << std::endl;;`

espxiehang, 2020-06-05:
Thanks for feedback.
This problem is because the system's internal memory is exhausted.
WiFi dynamic buffer, use wifi_malloc() to allocate memory.
It uses "MALLOC_CAP_DEFAULT|MALLOC_CAP_INTERNAL".
Therefore, we need to check if there is a memory leak.

espxiehang, 2020-06-05:
Hi @enricop ,During the test, the application ran those businesses. Can you describe it in detail?
After the application runs, how much internal memory does the system have?

enricop, 2020-07-20:
hello @projectgus @espxiehang 

what does the first point of [this commit message](https://github.com/espressif/esp-idf/commit/8ae3c733db7e7525358a0c0584573f2c1ee51710) refers to? Have you resolved some kind of memory allocation issue in wifi lib?

thanks for your support,
Best Regards,

espxiehang, 2020-07-20:
Hi @enricop ,Fixed an issue where setting TX DMA buffer above 32 would not take effect.

enricop, 2020-07-27:
hello @espxiehang : we can perform a further long run test.

Before that, we would like to know what we have to monitor when the following log occurs:
```
W (51791738) wifi:alloc eb len=76 type=2 fail, heap:2892620
W (51791738) wifi:m f probe req l=0
```
We will use "MALLOC_CAP_DEFAULT|MALLOC_CAP_INTERNAL" caps to print free mem. Any other advice?

Note that in our sdkconfig we have CONFIG_SPIRAM_USE_MALLOC=y , so MALLOC_CAP_DEFAULT already point to SPIRAM

Why wifi library fails to allocate 76bytes even if wifi_malloc() uses "MALLOC_CAP_DEFAULT|MALLOC_CAP_INTERNAL"?

Thanks for your support,
Enrico

espxiehang, 2020-07-27:
Hi @enricop ，
The reason for this log is: 
1, Buffer of "type=2" has been used to its maximum.
2,Insufficient internal system memory.
Therefore, when there is a problem, we need to observe the margin of the internal memory.
When a problem occurs, you can use the esp_get_free_internal_heap_size() function to print out the remaining capacity of the internal memory.

Alvin1Zhang, 2020-08-18:
@enricop Thanks for reporting, would you please help share if any updates for this issue? Thanks.

enricop, 2020-08-28:
Hi all, we don't have any updates yet.

could this leak be the reason of the problem above? https://github.com/espressif/esp-idf/commit/b50bbe3a5b1e18372605a222240397591c21bfec

thanks

espxiehang, 2020-08-28:
Hi @enricop ，May be related to this.

When the above exception occurs, the system's internal memory margin may be small, resulting in insufficient internal memory when sending management frames.

wgge, 2020-09-01:
Observed samething in v4.0.1, is there any possible fix?

espxiehang, 2020-09-04:
Hi @wgge ,The v4.0.1 version does not include b50bbe3, you can try the latest version/v4.1 version.

wgge, 2020-09-22:
@espxiehang The patch is for a regression introduced after v4.0.1. Hence the patch won't fix the title issue which is also observed in v4.0.1.

espxiehang, 2020-10-27:
Hi @wgge @enricop , Does this issue still exits? You can use the latest IDF to verify.

enricop, 2020-10-28:
hi @espxiehang . we are aligning the esp-idf to v4.0 branch last commit. you can close this issue. if needed we will open a new ticket. thank you.

AxelLin, 2020-11-05:
I hit similar issue with on a device build with v4.3-dev-1197-g8bc19ba893e5.

W (22501094) wifi:alloc eb len=76 type=2 fail, heap:5424
W (22501094) wifi:m f probe req l=0
W (22501215) wifi:alloc eb len=76 type=2 fail, heap:5424
W (22501215) wifi:m f probe req l=0
W (22501336) wifi:alloc eb len=76 type=2 fail, heap:5424
W (22501336) wifi:m f probe req l=0
W (22501457) wifi:alloc eb len=76 type=2 fail, heap:5424
W (22501457) wifi:m f probe req l=0
W (22501578) wifi:alloc eb len=76 type=2 fail, heap:5424
W (22501579) wifi:m f probe req l=0
W (22501699) wifi:alloc eb len=76 type=2 fail, heap:5424
W (22501700) wifi:m f probe req l=0
I (22501821) app: Disconnect reason: 201
I (22501821) app: Retry connecting AP: TEST-AP
W (22501822) wifi:alloc eb len=76 type=2 fail, heap:5336
W (22501826) wifi:m f probe req l=0
W (22501950) wifi:alloc eb len=76 type=2 fail, heap:5416
W (22501950) wifi:m f probe req l=0
W (22502071) wifi:alloc eb len=76 type=2 fail, heap:5416

In normal operation my device should has 80KB+ free memory.
I don't know what happend make it show heap:5424.

espxiehang, 2020-11-05:
Hi @AxelLin ,We have malloc a buffer, which will cost 1600 memory.
I guess when this happens,the system doesn't have enough contiguous memory.

Alvin1Zhang, 2020-12-09:
Thanks for reporting and sorry for the slow turnaround, the fix is available at https://github.com/espressif/esp-idf/commit/5a0124a103f06a5bbad6bc11a3e6b203ea70b88c, feel free to reopen if the issue still happens. Thanks.

closed 2020-12-09 by Alvin1Zhang