espressif/esp-idf #5392: Wifi: sta connection maximum SSID 31char

brazoayeye, 2020-06-04:
Hello, I reported a bug to a side project regarding this topic: maximum password length is 31char but sometimes routers have built-in password longer than 31 chars. 

Isn't possible to increase the maximum size to the standard 63char to avoid problems with strange configurations?

Thanks

https://github.com/martin-ger/esp32_nat_router/issues/17

negativekelvin, 2020-06-04:
https://github.com/espressif/esp-idf/issues/3762

HarveyRong-Esp, 2020-11-09:
Hi @brazoayeye，
802.11-2016 stipulates that the maximum SSID is 32, which will not reach 64. At present, the maximum SSID length that can be configured with ESP32 is also 32.
[image]

closed 2022-01-06 by jack0c