espressif/esp-idf #5393: esp-idf-tools-setup-2.3.exe rebooted my Windows without first asking

elektronika-ba, 2020-06-04:
This is not related to actual ESP-IDF, but for Windows installation tool **esp-idf-tools-setup-2.3.exe**

The install wizard rebooted my windows without asking, closing all my active applications (losing my current work).

Not very nice.

igrr, 2020-06-04:
@elektronika-ba Sorry about that! The installer code doesn't contain the logic to reboot the computer (even as an option), but it is possible that there is a bug in the installer which has triggered this. Have you selected the option to install Windows Defender exclusions during the installation? How far did the installation process go (if you can recall)? Is the installation log present in the Temp directory under your user profile?

elektronika-ba, 2020-06-04:
> Have you selected the option to install Windows Defender exclusions during the installation? How far did the installation process go (if you can recall)? Is the installation log present in the Temp directory under your user profile?

I unchecked (deselected) that option regarding the Windows Defender, because I am on my work PC and didn't want to make any exclusions in the Firewall.

The installation took some time and at the end it said that it failed, if I remember correctly something as a status code 1. It asked me if I wanted to view the log file, and I said YES, and then it closed all my apps and started to reboot.

I was looking for the log file, but couldn't find it in the destination directory (which I selected in totally different folder and different drive on my PC from the default proposed destination).

BTW I already had Python 3.7 installed on my PC and I selected it during the installation.

Do you have any idea where I can find the installation log file to share it here?

igrr, 2020-06-04:
The log should be in a directory named `C:\Users\username\AppData\Local\Temp\is-XXXXX.tmp`

elektronika-ba, 2020-06-04:
No files with that name in the location you mentioned.

All I have is "is-K11O7.tmp" and it is an empty folder.

Nevermind, I will repeat the process on my home PC. I guess it will work, but this still remains a mistery.

elektronika-ba, 2020-06-05:
I installed the IDF on my home PC (Windows 10) without an issue.
P.S. It does ask me to reboot the PC after the installation, which is just fine.

[image]

I don't know what caused my Windows 8.1 to reboot after failed installation without asking first.

georgik, 2021-04-14:
@elektronika-ba Thank you for reporting the problem. Just to confirm: Installer 2.3 on Windows 8 is really invoking restart even in case of failed installation. This is indicated by message:
```
Need to restart Windows? Yes
```
That's interesting because installation of ESP-IDF has no reason to call restart.

georgik, 2021-04-14:
@elektronika-ba  Inno Setup in PrepareToInstall phase is checking whether some files are required to be replaced.
This phase occurs in our installer after downloading all requirements. If everything is ok, it should log following message:
```
Calling RestartManager's RmGetList.
RmGetList finished successfully.
RestartManager found no applications using one of our files.
Starting the installation process.
```

The problem might occur when there is pending Windows update. We recommend to check Windows for missing updates or pending restarts. The similar failure was also noticed when installing Git in version of installer 2.3.

Please use more recent version of installer which does not require elevation of privileges on Git or Python installation.
For Windows 8 we recommend version 2.6 which contains Embedded Git and Embedded Python:  https://dl.espressif.com/dl/esp-idf-tools-setup-online-2.6.exe

closed 2021-04-15 by Alvin1Zhang