espressif/esp-idf #5396: Release/v3.3: Building Bootloader fails when flash encryption and/or hardware secure boot enabled

S3phe, 2020-06-04:
Environment

- Development Kit:     none
- Module or chip used:  ESP32-WROVER
- IDF version: v3.3.2-208-gd7e6eecd6

- Build System:         Make
- Compiler version: xtensa-esp32-elf-gcc (crosstool-NG crosstool-ng-1.22.0-80-g6c4433a) 5.2.0
- Operating System:     macOS

- Using an IDE?: Yes: VSCODE
- Power Supply:         external 3.3V

## Problem Description

In "make menuconfig", the building of the bootloader fails if:
- Security Options -> "Enable hardware secure boot in bootloader" = ENABLED ("SECURE_BOOT_ENABLED")
- AND/OR: Security Options -> "Enable flash encryption on boot" = ENABLED ("FLASH_ENCRYPTION_ENABLED")

//Detailed problem description goes here.
The following message appears after building with "make bootloader"

espsecure.py v2.8
secure_boot_signing_key.pem public key extracted to /build/bootloader/bootloader_support/signature_verification_key.bin
EMBED build/bootloader/bootloader_support/signature_verification_key.bin.bin.o
AR build/bootloader/bootloader_support/libbootloader_support.a
LD build/bootloader/bootloader.elf
/build/bootloader/efuse/libefuse.a(esp_efuse_utility.o):(.literal.esp_efuse_utility_burn_efuses+0x18): undefined reference to `esp_clk_apb_freq'
/build/bootloader/efuse/libefuse.a(esp_efuse_utility.o): In function `esp_efuse_utility_burn_efuses':
/esp/esp-idf/components/efuse/src/esp_efuse_utility.c:214: undefined reference to `esp_clk_apb_freq'
collect2: error: ld returned 1 exit status
make[1]: *** [/build/bootloader/bootloader.elf] Error 1
make: *** [/build/bootloader/bootloader.bin] Error 2

### Expected Behavior

Building a bootloader with secureboot and/or flash-encryption support

### Actual Behavior

Building fails

### Steps to reproduce
1. Get ESP-IDF / Install Toolchain like docs tell you on release/v3.3 page
2. Create new Hello-World project
3. make menuconfig -> security options -> Enable Hardware Secure boot and/or flash encryption
4. Save and Build with "make bootloader"
5. *crash*
[sdkconfig.txt](https://github.com/espressif/esp-idf/files/4731102/sdkconfig.txt)

Alvin1Zhang, 2020-06-05:
@S3phe Thanks for reporting, we will look into.

S3phe, 2020-06-08:
@Alvin1Zhang :
Is there something I can do, to help you? 

In my opinion this problem is important, because Release/v3.3 is a long-term supported release, which is also meant for devices in production.

KonstantinKondrashov, 2020-06-08:
Hi @S3phe! Thanks for the report. It was fixed in 4.0 and was not ported to 3.3. I will do it.

KonstantinKondrashov, 2020-06-08:
I will provide the patch for this here.

KonstantinKondrashov, 2020-06-08:
@S3phe Could you try this patch. [9093.zip](https://github.com/espressif/esp-idf/files/4744867/9093.zip)
The fix for master was in fb72a6f629a62c0655e8eff1d2d3f67d7aa9b62c.

S3phe, 2020-06-08:
@KonstantinKondrashov :
Thank you very much! Patch is working!

Alvin1Zhang, 2020-07-03:
@S3phe Thanks for reporting and sorry for slow turnaround, the fix is now available at https://github.com/espressif/esp-idf/commit/5e84ee108c526a0cfb562acb5ceba926bf2bc27b, feel free to reopen if the issue still happens. Thanks.

closed 2020-07-03 by Alvin1Zhang