espressif/esp-idf #5397: (BLE mesh) Get Node from Name

JacekDob, 2020-06-04:
Please add function similar to `esp_ble_mesh_provisioner_get_node_index` which based on `name` returns `esp_ble_mesh_node_t` (like in `esp_ble_mesh_provisioner_get_node_with_uuid` or `esp_ble_mesh_provisioner_get_node_with_addr`).

Currently having node `name` is not possible to relate it to node (`esp_ble_mesh_node_t`).
Currently it is possible when using `esp_ble_mesh_provisioner_get_node_index` to get `node index`, but there is not function to use this `node index`.

Alvin1Zhang, 2020-06-05:
@JacekDob Thanks for reporting this feature request.

JacekDob, 2020-06-05:
Thank you for that.

Please also add possibility to iterate through provisioned nodes.

Currently I see only possibility to get number of provisioned nodes (`esp_ble_mesh_provisioner_get_prov_node_count`), but there is no access to that/not possible to iterate.

Campou, 2020-06-08:
Hi @JacekDob 

Thanks for the feedback. I have submitted a MR for this internally, and since the code is just in review, please have a try with this patch [get_node_with_name.zip](https://github.com/espressif/esp-idf/files/4744234/get_node_with_name.zip)

Thanks.

JacekDob, 2020-06-09:
Thank you,

how to apply it?

I am cloning then newest version and trying to apply and I get:

```
$ git clone --recursive https://github.com/espressif/esp-idf.git
$ cd esp-idf
$ git apply 0001-ble_mesh-Get-node-info-with-pre-configured-node-name.patch
error: patch failed: components/bt/esp_ble_mesh/mesh_core/provisioner_main.c:622
error: components/bt/esp_ble_mesh/mesh_core/provisioner_main.c: patch does not apply
```

Campou, 2020-06-12:
Hi @JacekDob

I generated the patch file based on the latest internal idf, but since the esp-idf in Github has not been synced for more than 20 days, this causes failing to apply the patch. 

Please apply this one [0001-ble_mesh-get-node-info-with-preset-node-name.patch.tar.gz](https://github.com/espressif/esp-idf/files/4768949/0001-ble_mesh-get-node-info-with-preset-node-name.patch.tar.gz)

Thanks.

JacekDob, 2020-06-12:
Thank you, tested, everything works fine.

Alvin1Zhang, 2020-06-30:
Thanks for reporting, the fix has been available at https://github.com/espressif/esp-idf/commit/00148cd0c6873cc65aa7edfab83566b517832155, feel free to reopen. Thanks.

closed 2020-06-30 by Alvin1Zhang