espressif/esp-idf #5398: Using RMT_CHANNEL_6 instead of 0 in led_strip_main.c causes interrupt watchdog crash

lucastcox, 2020-06-04:
## Environment

- Development Kit: [none]
- Module or chip used: ESP32-WROVER-B
- IDF version (run ``git describe --tags`` to find it): v4.2-dev-318-g605da33c3
- Build System:        CMAKE
- Compiler version : xtensa-esp32-elf-gcc (crosstool-NG esp-2019r2) 8.2.0
 Operating System:     [Linux]
- Power Supply:         [external 3.3V]

## Problem Description

In my own project, I was trying to use all 8 rmt channels for controlling ws2812 strips. It kept crashing, so I tried to narrow it down by using the example found [here](https://github.com/espressif/esp-idf/blob/master/examples/peripherals/rmt/led_strip/main/led_strip_main.c) and made one change. I changed 

Eventually, I figured out that I could reproduce the crashing by only making the following change:

```c
#define RMT_TX_CHANNEL RMT_CHANNEL_0
```
to
```c
#define RMT_TX_CHANNEL RMT_CHANNEL_6
```

### Expected Behavior

The led example runs normally.

### Actual Behavior

The example crashes. (see debug log below)

### Code to reproduce this issue

led_strip_main.c

## Debug Logs

```
I (13) boot: ESP-IDF v4.2-dev-318-g605da33c3 2nd stage bootloader
I (13) boot: compile time 14:26:26
I (13) boot: chip revision: 1
I (16) boot_comm: chip revision: 1, min. bootloader chip revision: 0
I (23) boot.esp32: SPI Speed      : 40MHz
I (28) boot.esp32: SPI Mode       : DIO
I (32) boot.esp32: SPI Flash Size : 2MB
I (37) boot: Enabling RNG early entropy source...
I (42) boot: Partition Table:
I (46) boot: ## Label            Usage          Type ST Offset   Length
I (53) boot:  0 nvs              WiFi data        01 02 00009000 00006000
I (60) boot:  1 phy_init         RF data          01 01 0000f000 00001000
I (68) boot:  2 factory          factory app      00 00 00010000 00100000
I (75) boot: End of partition table
I (79) boot_comm: chip revision: 1, min. application chip revision: 0
I (87) esp_image: segment 0: paddr=0x00010020 vaddr=0x3f400020 size=0x0609c ( 24732) map
I (105) esp_image: segment 1: paddr=0x000160c4 vaddr=0x3ffb0000 size=0x0218c (  8588) load
I (108) esp_image: segment 2: paddr=0x00018258 vaddr=0x40080000 size=0x00404 (  1028) load
0x40080000: _WindowOverflow4 at /home/lucas/Projects/RemoteIO/esp-idf/components/freertos/xtensa/xtensa_vectors.S:1789

I (114) esp_image: segment 3: paddr=0x00018664 vaddr=0x40080404 size=0x079b4 ( 31156) load
I (135) esp_image: segment 4: paddr=0x00020020 vaddr=0x400d0020 size=0x159f8 ( 88568) map
0x400d0020: _stext at ??:?

I (168) esp_image: segment 5: paddr=0x00035a20 vaddr=0x40087db8 size=0x058ac ( 22700) load
0x40087db8: spi_flash_chip_generic_read at /home/lucas/Projects/RemoteIO/esp-idf/components/spi_flash/spi_flash_chip_generic.c:150

I (186) boot: Loaded app from partition at offset 0x10000
I (186) boot: Disabling RNG early entropy source...
I (186) cpu_start: Pro cpu up.
I (190) cpu_start: Application information:
I (195) cpu_start: Project name:     led_strip
I (200) cpu_start: App version:      1
I (204) cpu_start: Compile time:     Jun  4 2020 14:26:28
I (210) cpu_start: ELF file SHA256:  020c1bbaa8737abb...
I (216) cpu_start: ESP-IDF:          v4.2-dev-318-g605da33c3
I (223) cpu_start: Starting app cpu, entry point is 0x400810d4
0x400810d4: call_start_cpu1 at /home/lucas/Projects/RemoteIO/esp-idf/components/esp32/cpu_start.c:274

I (223) cpu_start: App cpu up.
I (233) heap_init: Initializing. RAM available for dynamic allocation:
I (240) heap_init: At 3FFAE6E0 len 00001920 (6 KiB): DRAM
I (246) heap_init: At 3FFB29C8 len 0002D638 (181 KiB): DRAM
I (253) heap_init: At 3FFE0440 len 00003AE0 (14 KiB): D/IRAM
I (259) heap_init: At 3FFE4350 len 0001BCB0 (111 KiB): D/IRAM
I (265) heap_init: At 4008D664 len 0001299C (74 KiB): IRAM
I (272) cpu_start: Pro cpu start user code
I (290) spi_flash: detected chip: gd
I (291) spi_flash: flash io: dio
W (291) spi_flash: Detected size(8192k) larger than the size in the binary image header(2048k). Using the size in the binary image header.
I (301) cpu_start: Starting scheduler on PRO CPU.
I (0) cpu_start: Starting scheduler on APP CPU.
Guru Meditation Error: Core  0 panic'ed (Interrupt wdt timeout on CPU0)
Core 0 register dump:
PC      : 0x40084b10  PS      : 0x00050034  A0      : 0x400825c0  A1      : 0x3ffb0640  
0x40084b10: rmt_ll_get_err_interrupt_status at /home/lucas/Projects/RemoteIO/esp-idf/components/soc/src/esp32/include/hal/rmt_ll.h:242
 (inlined by) rmt_driver_isr_default at /home/lucas/Projects/RemoteIO/esp-idf/components/driver/rmt.c:672

0x400825c0: _xt_lowint1 at /home/lucas/Projects/RemoteIO/esp-idf/components/freertos/xtensa/xtensa_vectors.S:1164

A2      : 0x3ffb63b0  A3      : 0x00040000  A4      : 0x00000000  A5      : 0x00000000  
A6      : 0x00000008  A7      : 0x3ffb63bc  A8      : 0x00000000  A9      : 0x3ffb05e0  
A10     : 0xd3e7e5ba  A11     : 0x3ffb2278  A12     : 0xd3e7e5ba  A13     : 0xd3e7e5ba  
A14     : 0xd3e7e5ba  A15     : 0x3ffb2278  SAR     : 0x00000008  EXCCAUSE: 0x00000005  
EXCVADDR: 0x00000000  LBEG    : 0x00000000  LEND    : 0x00000000  LCOUNT  : 0x00000000  
Core 0 was running in ISR context:
EPC1    : 0x400d7cd5  EPC2    : 0x00000000  EPC3    : 0x00000000  EPC4    : 0x40084b10
0x400d7cd5: esp_pm_impl_waiti at /home/lucas/Projects/RemoteIO/esp-idf/components/esp32/pm_esp32.c:486

0x40084b10: rmt_ll_get_err_interrupt_status at /home/lucas/Projects/RemoteIO/esp-idf/components/soc/src/esp32/include/hal/rmt_ll.h:242
 (inlined by) rmt_driver_isr_default at /home/lucas/Projects/RemoteIO/esp-idf/components/driver/rmt.c:672

ELF file SHA256: 020c1bbaa8737abb071c664b7dc35efcb1ae0b4037c768b205912d7553bc6e14

Backtrace: 0x40084b0d:0x3ffb0640 0x400825bd:0x3ffb0680 0x400d7cd2:0x3ffb4f20 0x400d2156:0x3ffb4f50 0x4008b9cf:0x3ffb4f80 0x4008a21a:0x3ffb4fb0
0x40084b0d: rmt_ll_get_err_interrupt_status at /home/lucas/Projects/RemoteIO/esp-idf/components/soc/src/esp32/include/hal/rmt_ll.h:241
 (inlined by) rmt_driver_isr_default at /home/lucas/Projects/RemoteIO/esp-idf/components/driver/rmt.c:672

0x400825bd: _xt_lowint1 at /home/lucas/Projects/RemoteIO/esp-idf/components/freertos/xtensa/xtensa_vectors.S:1164

0x400d7cd2: esp_pm_impl_waiti at /home/lucas/Projects/RemoteIO/esp-idf/components/esp32/pm_esp32.c:484

0x400d2156: esp_vApplicationIdleHook at /home/lucas/Projects/RemoteIO/esp-idf/components/esp_common/src/freertos_hooks.c:63

0x4008b9cf: prvIdleTask at /home/lucas/Projects/RemoteIO/esp-idf/components/freertos/tasks.c:3385 (discriminator 1)

0x4008a21a: vPortTaskWrapper at /home/lucas/Projects/RemoteIO/esp-idf/components/freertos/xtensa/port.c:143

Core 1 register dump:
PC      : 0x400d7cd5  PS      : 0x00060e34  A0      : 0x800d2159  A1      : 0x3ffb5690  
0x400d7cd5: esp_pm_impl_waiti at /home/lucas/Projects/RemoteIO/esp-idf/components/esp32/pm_esp32.c:486

A2      : 0x00000000  A3      : 0xd3e7e5ba  A4      : 0x00000001  A5      : 0x80000001  
A6      : 0x400d0e48  A7      : 0x00000000  A8      : 0x3ffb2278  A9      : 0xd3e7e5ba  
0x400d0e48: dport_access_init_core at /home/lucas/Projects/RemoteIO/esp-idf/components/esp32/dport_access.c:149

A10     : 0x00000000  A11     : 0x00060e23  A12     : 0x00060e20  A13     : 0x00060e23  
A14     : 0x00060e20  A15     : 0x00000001  SAR     : 0x00000000  EXCCAUSE: 0x00000005  
EXCVADDR: 0x00000000  LBEG    : 0x00000000  LEND    : 0x00000000  LCOUNT  : 0x00000000  

ELF file SHA256: 020c1bbaa8737abb071c664b7dc35efcb1ae0b4037c768b205912d7553bc6e14

Backtrace: 0x400d7cd2:0x3ffb5690 0x400d2156:0x3ffb56c0 0x4008b9cf:0x3ffb56f0 0x4008a21a:0x3ffb5720
0x400d7cd2: esp_pm_impl_waiti at /home/lucas/Projects/RemoteIO/esp-idf/components/esp32/pm_esp32.c:484

0x400d2156: esp_vApplicationIdleHook at /home/lucas/Projects/RemoteIO/esp-idf/components/esp_common/src/freertos_hooks.c:63

0x4008b9cf: prvIdleTask at /home/lucas/Projects/RemoteIO/esp-idf/components/freertos/tasks.c:3385 (discriminator 1)

0x4008a21a: vPortTaskWrapper at /home/lucas/Projects/RemoteIO/esp-idf/components/freertos/xtensa/port.c:143

Rebooting...
```

Alvin1Zhang, 2020-06-05:
@lucastcox Thanks for reporting, we will look into. @suda-morris

suda-morris, 2020-06-05:
@lucastcox Could you please try installing RMT driver from CPU core 1 (i.e. create a temporarily pin-to-core task, where invoking rmt driver install), in this way, the RMT interrupt will be installed on Core1. 
BTW, I didn't reproduced your issue in my local.

lucastcox, 2020-06-05:
Hmmm...tried that by setting cpu_start of "main task" to be pinned to 1. Different issue now. It gets stuck and then the task WDT resets it. I've attached two elf files: 1 with app_main pinned to 0, and one with app_main pinned to 1
[elf files.zip](https://github.com/espressif/esp-idf/files/4736119/elf.files.zip)

```
rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0030,len:4
load:0x3fff0034,len:7248
ho 0 tail 12 room 4
load:0x40078000,len:14352
load:0x40080400,len:4448
0x40080400: _init at ??:?

entry 0x400806f0
I (30) boot: ESP-IDF v4.2-dev-318-g605da33c3 2nd stage bootloader
I (30) boot: compile time 08:10:43
I (30) boot: chip revision: 1
I (35) boot_comm: chip revision: 1, min. bootloader chip revision: 0
I (42) boot.esp32: SPI Speed      : 40MHz
I (46) boot.esp32: SPI Mode       : DIO
I (51) boot.esp32: SPI Flash Size : 2MB
I (55) boot: Enabling RNG early entropy source...
I (61) boot: Partition Table:
I (64) boot: ## Label            Usage          Type ST Offset   Length
I (72) boot:  0 nvs              WiFi data        01 02 00009000 00006000
I (79) boot:  1 phy_init         RF data          01 01 0000f000 00001000
I (87) boot:  2 factory          factory app      00 00 00010000 00100000
I (94) boot: End of partition table
I (98) boot_comm: chip revision: 1, min. application chip revision: 0
I (105) esp_image: segment 0: paddr=0x00010020 vaddr=0x3f400020 size=0x06068 ( 24680) map
I (123) esp_image: segment 1: paddr=0x00016090 vaddr=0x3ffb0000 size=0x0218c (  8588) load
I (127) esp_image: segment 2: paddr=0x00018224 vaddr=0x40080000 size=0x00404 (  1028) load
0x40080000: _WindowOverflow4 at /home/lucas/Projects/RemoteIO/esp-idf/components/freertos/xtensa/xtensa_vectors.S:1789

I (133) esp_image: segment 3: paddr=0x00018630 vaddr=0x40080404 size=0x079e8 ( 31208) load
I (154) esp_image: segment 4: paddr=0x00020020 vaddr=0x400d0020 size=0x13eb0 ( 81584) map
0x400d0020: _stext at ??:?

I (184) esp_image: segment 5: paddr=0x00033ed8 vaddr=0x40087dec size=0x02e50 ( 11856) load
0x40087dec: rtc_clk_cpu_freq_get_config at /home/lucas/Projects/RemoteIO/esp-idf/components/soc/src/esp32/rtc_clk.c:651

I (196) boot: Loaded app from partition at offset 0x10000
I (196) boot: Disabling RNG early entropy source...
I (197) cpu_start: Pro cpu up.
I (200) cpu_start: Application information:
I (205) cpu_start: Project name:     led_strip
I (210) cpu_start: App version:      1
I (215) cpu_start: Compile time:     Jun  5 2020 08:10:33
I (221) cpu_start: ELF file SHA256:  840748b38bde0816...
I (227) cpu_start: ESP-IDF:          v4.2-dev-318-g605da33c3
I (233) cpu_start: Starting app cpu, entry point is 0x400810a8
0x400810a8: call_start_cpu1 at /home/lucas/Projects/RemoteIO/esp-idf/components/esp32/cpu_start.c:274

I (0) cpu_start: App cpu up.
I (244) heap_init: Initializing. RAM available for dynamic allocation:
I (250) heap_init: At 3FFAE6E0 len 00001920 (6 KiB): DRAM
I (256) heap_init: At 3FFB29C0 len 0002D640 (181 KiB): DRAM
I (263) heap_init: At 3FFE0440 len 00003AE0 (14 KiB): D/IRAM
I (269) heap_init: At 3FFE4350 len 0001BCB0 (111 KiB): D/IRAM
I (276) heap_init: At 4008AC3C len 000153C4 (84 KiB): IRAM
I (282) cpu_start: Pro cpu start user code
I (300) spi_flash: detected chip: gd
I (300) spi_flash: flash io: dio
W (300) spi_flash: Detected size(4096k) larger than the size in the binary image header(2048k). Using the size in the binary image header.
I (310) cpu_start: Starting scheduler on PRO CPU.
I (0) cpu_start: Starting scheduler on APP CPU.
ets Jun  8 2016 00:22:57

rst:0x7 (TG0WDT_SYS_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0030,len:4
load:0x3fff0034,len:7248
ho 0 tail 12 room 4
load:0x40078000,len:14352
load:0x40080400,len:4448
0x40080400: _init at ??:?

entry 0x400806f0
I (61) boot: ESP-IDF v4.2-dev-318-g605da33c3 2nd stage bootloader
I (62) boot: compile time 08:10:43
I (62) boot: chip revision: 1
I (66) boot_comm: chip revision: 1, min. bootloader chip revision: 0
I (73) boot.esp32: SPI Speed      : 40MHz
I (78) boot.esp32: SPI Mode       : DIO
I (83) boot.esp32: SPI Flash Size : 2MB
W (87) boot.esp32: PRO CPU has been reset by WDT.
W (93) boot.esp32: WDT reset info: PRO CPU PC=0x400e3b82 (waiti mode)
0x400e3b82: esp_pm_impl_waiti at /home/lucas/Projects/RemoteIO/esp-idf/components/esp32/pm_esp32.c:486

W (100) boot.esp32: WDT reset info: APP CPU PC=0x40083edb
0x40083edb: rmt_ll_get_err_interrupt_status at /home/lucas/Projects/RemoteIO/esp-idf/components/soc/src/esp32/include/hal/rmt_ll.h:242
 (inlined by) rmt_driver_isr_default at /home/lucas/Projects/RemoteIO/esp-idf/components/driver/rmt.c:672
```

suda-morris, 2020-06-08:
Hi @lucastcox I reproduced your issue successfully. But could you also try the example against the lasted commit on master branch? It seems like the issue doesn't exist there.

lucastcox, 2020-06-09:
@suda-morris, thanks. I'll try on master and let you know.

lucastcox, 2020-06-22:
@suda-morris Haven't forgotten about this. Just haven't had a chance to test yet. I'll update you when I do.

lucastcox, 2020-07-17:
@suda-morris can confirm it is resolved in master.

lucastcox, 2020-07-24:
@suda-morris I spoke to soon. the code I "confirmed it" on was still avoiding channel 6 of the rmt module. Even on the latest master, I still have the issue

suda-morris, 2022-06-13:
Sorry for the late response. Unfortunately, I test the led_strip example (the one from release/v4.2 branch) against the master branch of esp-idf SDK. Can channel 6 can work well.

closed 2022-06-13 by suda-morris