espressif/esp-idf #5399: Code Hang's on Wifi Startup

jonhunterking66, 2020-06-05:
## Environment

- Development Kit: ESP32-DevKitC
- Kit version (for WroverKit/PicoKit/DevKitC): v4
- Module or chip used:  ESP32-WROOM-32D
- IDF version (run ``git describe --tags`` to find it):
v4.1-dev-2140-g2976f5dae
- Build System: idf.py
- Compiler version (run ``xtensa-esp32-elf-gcc --version`` to find it):
xtensa-esp32-elf-gcc (crosstool-NG esp-2019r2) 8.2.0
- Operating System:     Linux
- Using an IDE?: VS Code 
- Power Supply: I work with several supplies. USB and battery are most common

## Problem Description

After a sleep of 5 seconds during the initialization of wifi the code gets stuck and hangs indefinitely. It is an intermittent issue. My code cycles through sleeping, then waking up and starting wifi. It usually takes about 90 cycles for this to occur. This is equivalent to 15 minutes. The issue is fixed by pressing the reset button or re-flashing the unit. 

### Expected Behavior

The code would not get stuck and initialize successfully. 

### Actual Behavior

The program gets stuck at the same place repeatedly upon wifi startup. 

// If possible, attach a picture of your setup/wiring here.

### Code to reproduce this issue

```cpp
void app_main(void)
{
  bleInit();
  createWifiStruct();
  gpio_init();
  adc_init();
  motorsInit();
  xTaskCreate(&read_task, "read", 1024*10, NULL, 1, &read_handle);
  vTaskSuspend(read_handle);
  xTaskCreate(&main_task, "MAIN", 1024 *80, NULL, 2, &main_handle);
}

void createWifiStruct()
{
  esp_err_t ret;
  ESP_ERROR_CHECK(nvs_flash_init());
  tcpip_adapter_init();
  ESP_ERROR_CHECK(esp_event_loop_init(event_handler, NULL));
  wifi_init_config_t wifi_init_config = WIFI_INIT_CONFIG_DEFAULT();
  ESP_ERROR_CHECK(esp_wifi_init(&wifi_init_config));
  ESP_ERROR_CHECK(esp_wifi_set_mode(WIFI_MODE_STA));
}

```
## Debug Logs

```
I (671) esp_image: segment 6: paddr=0x0016f590 vaddr=0x400c0000 size=0x00064 (   100) 
I (687) boot: Loaded app from partition at offset 0x20000
I (687) boot: Disabling RNG early entropy source...
I (688) cpu_start: Pro cpu up.
I (691) cpu_start: Application information:
I (696) cpu_start: Project name:     mair-template
I (701) cpu_start: App version:      2602e703-dirty
I (707) cpu_start: Compile time:     Jun  5 2020 08:36:13
I (713) cpu_start: ELF file SHA256:  949c6d0071992b91...
I (719) cpu_start: ESP-IDF:          v4.1-dev-2140-g2976f5dae
I (725) cpu_start: Starting app cpu, entry point is 0x4008148c
0x4008148c: call_start_cpu1 at /home/jonhunterking/esp/esp-idf/components/esp32/cpu_start.c:275

I (0) cpu_start: App cpu up.
I (736) heap_init: Initializing. RAM available for dynamic allocation:
I (743) heap_init: At 3FFAFF10 len 000000F0 (0 KiB): DRAM
I (749) heap_init: At 3FFB6388 len 00001C78 (7 KiB): DRAM
I (755) heap_init: At 3FFB9A20 len 00004108 (16 KiB): DRAM
I (761) heap_init: At 3FFBDB5C len 00000004 (0 KiB): DRAM
I (767) heap_init: At 3FFD2C60 len 0000D3A0 (52 KiB): DRAM
I (773) heap_init: At 3FFE0440 len 00003AE0 (14 KiB): D/IRAM
I (780) heap_init: At 3FFE4350 len 0001BCB0 (111 KiB): D/IRAM
I (786) heap_init: At 4009E818 len 000017E8 (5 KiB): IRAM
I (792) cpu_start: Pro cpu start user code
I (811) spi_flash: detected chip: generic
I (811) spi_flash: flash io: dio
I (811) cpu_start: Starting scheduler on PRO CPU.
I (0) cpu_start: Starting scheduler on APP CPU.
I (907) BTDM_INIT: BT controller compile version [fd2a63b]
I (909) system_api: Base MAC address is not set, read default base MAC address from BLK0 of EFUSE
I (928) phy: phy_version: 4180, cb3948e, Sep 12 2019, 16:39:13, 0, 1
I (1298) BLE: REGISTER_APP_EVT, status 0, app_id 0

I (1300) NVS: Returned Unit Number is oto66
I (1316) BLE: CREATE_SERVICE_EVT, status 0,  service_handle 40

I (1339) BLE: SERVICE_START_EVT, status 0, service_handle 40

I (1340) BLE: ADD_CHAR_EVT, status 0,  attr_handle 42, service_handle 40

E (1343) BLE: IN ONE
I (1346) BLE: the gatts demo char length = 3

I (1351) BLE: prf_char[0] =11

I (1355) BLE: prf_char[1] =22

I (1358) BLE: prf_char[2] =33

I (1362) BLE: ADD_CHAR_EVT, status 0,  attr_handle 44, service_handle 40

E (1370) BLE: IN TWO
I (1373) BLE: the gatts demo char length = 3

I (1378) BLE: prf_char[0] =11

I (1382) BLE: prf_char[1] =22

I (1385) BLE: prf_char[2] =33

I (1389) BLE: ADD_CHAR_EVT, status 133,  attr_handle 0, service_handle 40

I (1398) BLE: ADD_DESCR_EVT, status 0, attr_handle 45, service_handle 40

I (1404) BLE: ADD_DESCR_EVT, status 133, attr_handle 0, service_handle 40

I (1412) wifi: wifi driver task: 3ffdb56c, prio:23, stack:3584, core=0
I (1419) system_api: Base MAC address is not set, read default base MAC address from BLK0 of EFUSE
I (1428) system_api: Base MAC address is not set, read default base MAC address from BLK0 of EFUSE
I (1459) wifi: wifi firmware version: 551ce8d
I (1459) wifi: config NVS flash: enabled
I (1459) wifi: config nano formating: disabled
I (1460) wifi: Init dynamic tx buffer num: 32
I (1464) wifi: Init data frame dynamic rx buffer num: 32
I (1469) wifi: Init management frame dynamic rx buffer num: 32
I (1474) wifi: Init management short buffer num: 32
I (1479) wifi: Init static rx buffer size: 1600
I (1483) wifi: Init static rx buffer num: 10
I (1487) wifi: Init dynamic rx buffer num: 32
```

Alvin1Zhang, 2020-06-08:
@jonhunterking66 Thanks for reporting, we will look into.

liuzfesp, 2020-06-15:
Hi @jonhunterking66, does this issue happen if the BLE is not initialized?

jonhunterking66, 2020-06-15:
@liuzfesp I have restructured my code such that BLE is initialized before wifi, I have not seen the issue since.

Alvin1Zhang, 2020-06-16:
Glad to hear the issue didn't happen. Feel free to reopen if it still happens. Thanks.

closed 2020-06-16 by Alvin1Zhang