espressif/esp-idf #5401: Connection Issue With ESP32 When using OTA

jt-moriarty, 2020-06-05:
Hello, 

I'm attempting to incorporate OTA update capability into an existing project that uses the ESP32. I've copied the relevant code from this example (https://github.com/espressif/esp-idf/tree/master/examples/system/ota/advanced_https_ota) into my project. I've included the cert for my server in the project, and I have a firmware .bin up on the server at the address that the ESP looks for when it runs. I'm running v4.2-dev-1660-g7d75213 of the ESP IDF.

I get the following output when I run the program (after I've established a working connection via wifi or ethernet):
`I (7553) esp_https_ota: Starting OTA...
I (7553) esp_https_ota: Writing to partition subtype 16 at offset 0x110000
D (7553) HTTP_CLIENT: esp_transport_read returned:-1 and errno:128 
E (7553) esp_https_ota: Connection closed, errno = 128
E (7563) esp_https_ota: Complete headers were not received
E (7573) ota-updater: esp_https_ota_read_img_desc failed
E (7583) ota-updater: ESP_HTTPS_OTA upgrade failed 0`

It seems to start grabbing the image description but disconnects almost immediately. I've seen posts about this issue on earlier IDF versions, but the posts seem to indicate that the issue has been resolved. Happy to provide additional information if necessary that will help me resolve this problem. Thanks!

shubhamkulkarni97, 2020-06-08:
Hi @jt-moriarty,

Can you try running `advanced_https_ota` example in ESP-IDF and check if you get same errors?

If you are facing issue with `advanced_https_ota` example as well, please provide us more information about server used to perform OTA update and host OS on which server is running.

Please check if any of the suggestions in https://github.com/espressif/esp-idf/issues/5240#issuecomment-639292839 work for you.

Also check if you are able to download correct firmware image (using `curl`) on host machine.

jt-moriarty, 2020-06-09:
As a followup, troubleshooting using the example we were able to identify a problem with our server that was the cause of the error, thank you for your assistance.

closed 2020-06-09 by jt-moriarty