espressif/esp-idf #5402: Ability to extend available HTTP methods in esp_http_client

kaelef, 2020-06-05:
**Is your feature request related to a problem? Please describe.**

I'm trying to convert a project that uses Arduino-ESP32 to one that's pure ESP-IDF.  The project communicates with WebDAV, so it needs to be able to send HTTP requests with methods like PROPFIND, MOVE, etc.  When using the HttpClient class in Arduino-ESP32, the sendRequest(TYPE, payload, size) is available.  Since TYPE is a string, any method name can be specified.  esp_http_client uses an enum, esp_http_client_method_t, and provides no way to override the HTTP method string with something not in the enum list.

**Describe the solution you'd like**

Either the esp_http_client_method_t enum can be extended to include the standard WebDAV methods, or a way to specify an arbitrary method the way HttpClient in Arudino-ESP32 allows would solve the problem.

esp_http_client_set_method() could be overloaded to allow use of a char pointer in place of a esp_http_client_method_t value.

**Describe alternatives you've considered**

I've looked at the source for esp_http_client, and the method string is placed into the outgoing buffer before the usual HTTP headers are added.  I don't see a way to fix this without making my own version of esp_http_client that doesn't have this limitation.

Alvin1Zhang, 2020-06-08:
@kaelef Thanks for evaluating this feature request.

shubhamkulkarni97, 2020-06-09:
@kaelef,

It is not possible to change API signature to allow specifying arbitrary method. The only way is to add methods in `esp_http_client_method_t` enum.

Can you elaborate what all methods need to be added in esp_http_client_method_t?

kaelef, 2020-06-09:
WebDAV is described well here: https://en.wikipedia.org/wiki/WebDAV
The additional methods it specifies are:

- COPY
- MOVE
- LOCK
- UNLOCK
- PROPFIND
- PROPPATCH
- MKCOL

kaelef, 2020-06-11:
Yes, this will work - thank you!

closed 2020-07-10 by espressif-bot