espressif/esp-idf #5403: I2S Slave bit set to 1 in Master mode?

niclash, 2020-06-06:
https://github.com/espressif/esp-idf/blob/63cd2b06137ddfab9a3df0a9905698b526f4e87d/components/soc/src/hal/i2s_hal.c#L239

Shouldn't both SLAVE bits be set to "0" when in master mode?

koobest, 2020-06-08:
Hi, @niclash 
In master loop back mode(master TX and RX sharing the same BCK and WS), the receiver should set to slave mode.

niclash, 2020-06-08:
Ok, but that isn't the only way it can operate, right? 
Nevertheless, is there a complete(!) example of streaming from I2S external encoder, into processing that stream and out to an external I2S encoder?? I can't find one...

koobest, 2020-06-08:
Hi, @niclash
The driver does not support this feature because the transmitter and receiver may have different sampling rates..  you can use I2S0 for TX and I2S1 for RX on ESP32, but if you use ESP32-S2, the only way is to rewrite the driver.

thanks !!

Alvin1Zhang, 2020-06-24:
Thanks for reporting, feel free to reopen if you still have the issue, or please help file a new ticket for new issue. Thanks.

closed 2020-06-24 by Alvin1Zhang