espressif/esp-idf #5405: httpd_ws_send_frame_async makes illegal frames

jkent, 2020-06-06:
## Environment

- Development Kit:      none
- Module or chip used:  ESP32
- IDF version (run ``git describe --tags`` to find it):
    v4.2-dev-1660-g7d7521367
- Build System:         idf.py
- Compiler version (run ``xtensa-esp32-elf-gcc --version`` to find it):
    xtensa-esp32-elf-gcc (crosstool-NG esp-2020r1) 8.2.0
- Operating System:     Linu
- Using an IDE?:    Yes, Visual studio code
- Power Supply:         USB

## Problem Description

When using httpd_ws_send_frame_async, a simple client using python websocket_client says that frames are invalid.  Non-async frames are fine.

### Expected Behavior

Data would be received as normal

### Actual Behavior

send: b'\x81\x87w\x0eS\xd1?k?\xbd\x18.c'
recv: Hello 0
send: b'\x81\x87\xb7\xe4\xb3\x92\xff\x81\xdf\xfe\xd8\xc4\x82'
error: Illegal frame
send: b'\x88\x82\xf7\xf3w\t\xf4\x1b'
closed

### Steps to reproduce

Run a websocket server.  Run python client.  Trigger an async event (such as a push button).

### Code to reproduce this issue

This should also error out with the Espressif ws_echo_server as well.

```cpp
static void async_work(void *arg)
{
    connection_data_t *n;
    SLIST_FOREACH(n, &connection_data_head, next) {
        static const char *data = "Hello async world!";
        httpd_ws_frame_t ws_pkt = {
            .payload = (uint8_t *)data,
            .type = HTTPD_WS_TYPE_TEXT,
            .len = strlen(data),
        };

        httpd_ws_send_frame_async(server, n->fd, &ws_pkt);
    }
}

static void async_task(void *pvParameters)
{
    while (true) {
        xEventGroupWaitBits(io_event_group, IO_BUTTON0_DOWN, pdFALSE,
            pdFALSE, portMAX_DELAY);

        httpd_queue_work(server, async_work, NULL);

        xEventGroupWaitBits(io_event_group, IO_BUTTON0_UP, pdFALSE,
            pdFALSE, portMAX_DELAY);
    }
}
```

Python test code is here: https://pypi.org/project/websocket_client/#long-lived-connection

Alvin1Zhang, 2020-06-08:
@jkent Thanks for reporting, we will look into.

david-cermak, 2020-06-18:
Hi @jkent 

The reason why the async frames are treated as invalid is that they miss the `FIN` flag. Please update the snippet to:
``` c
        httpd_ws_frame_t ws_pkt = {
            .payload = (uint8_t *)data,
            .type = HTTPD_WS_TYPE_TEXT,
            .len = strlen(data),
            .final = true, 
        };
```
should make the websocket client receive the async frames correctly.

There's a bug in the example code and the documentation is not very clear on the asynchronous send. This will be fixed.

baldhead69, 2020-07-16:
Hello,

Not fixed yet, neither example code nor documentation.

"There's a bug in the example code and the documentation is not very clear on the asynchronous send. This will be fixed."

https://github.com/espressif/esp-idf/issues/5602

Thank's.

baldhead69, 2020-07-16:
Hi again,

How to use "httpd_ws_send_frame_async" without "static esp_err_t trigger_async_send(httpd_handle_t handle, httpd_req_t *req)"  ?

I need to use "httpd_queue_work(handle, ws_async_send, resp_arg);" ?

In the function "esp_err_t httpd_ws_send_frame_async(httpd_handle_t hd, int fd, httpd_ws_frame_t *frame)", is not very clear to me 
which one httpd_handle_t hd and which one int fd to use.

Thank's.

closed 2020-08-03 by espressif-bot