espressif/esp-idf #5406: esp_http_server websocket send message to all connected clients

ghtech, 2020-06-06:
Can not find any function in esp_http_server ->httpd_ws.c to send message to all connected client.
by adding following function in httpd_ws.c work for me 

esp_err_t httpd_ws_send_frame_to_all_clients(httpd_req_t *req, httpd_ws_frame_t *frame)
{
    esp_err_t ret = httpd_ws_check_req(req);
    if (ret != ESP_OK)
    {
        return ret;
    }
    struct httpd_data *hd = req->handle;
    int fd = httpd_req_to_sockfd(req);

    int i;
    for (i = 0; i < hd->config.max_open_sockets; i++)
    {

        ESP_LOGI(TAG, "INTO MY");
        if (fd != hd->hd_sd[i].fd)
        {
            httpd_ws_send_frame_async(hd->hd_sd[i].handle, hd->hd_sd[i].fd, frame);
        }
    }
    return ESP_OK;
}

In above function  struct httpd_data *hd = req->handle; give me access to all connected clients and we can send message through httpd_ws_send_frame_async

BUT if if try to access  struct httpd_data *hd = req->handle; in my project file mean outside of http server component give me 
**error: dereferencing pointer to incomplete type 'struct httpd_data'** 

I think httpd_data defined as private or I am doing something wrong .

**Describe alternatives you've considered**
or declare struct httpd_data public or make connected client public so we can send message to all clients if we want

linrx1511, 2020-06-25:
Hi, ghtech

Maybe when you call  the function `esp_err_t httpd_start(httpd_handle_t *handle, const httpd_config_t *config)`, you can set the open_fn and the close_fn in the `httpd_config_t ` structure. Then you can maintain these HD/FD by youself.
```
    /**
     * Custom session opening callback.
     *
     * Called on a new session socket just after accept(), but before reading any data.
     *
     * This is an opportunity to set up e.g. SSL encryption using global_transport_ctx
     * and the send/recv/pending session overrides.
     *
     * If a context needs to be maintained between these functions, store it in the session using
     * httpd_sess_set_transport_ctx() and retrieve it later with httpd_sess_get_transport_ctx()
     *
     * Returning a value other than ESP_OK will immediately close the new socket.
     */
    httpd_open_func_t open_fn;

    /**
     * Custom session closing callback.
     *
     * Called when a session is deleted, before freeing user and transport contexts and before
     * closing the socket. This is a place for custom de-init code common to all sockets.
     *
     * Set the user or transport context to NULL if it was freed here, so the server does not
     * try to free it again.
     *
     * This function is run for all terminated sessions, including sessions where the socket
     * was closed by the network stack - that is, the file descriptor may not be valid anymore.
     */
    httpd_close_func_t close_fn;
```

baldhead69, 2020-07-22:
Hi,

I need the same requirement.

https://github.com/espressif/esp-idf/issues/5602

closed 2020-09-16 by espressif-bot

Zauzolkov, 2021-12-22:
Here is possible solution (at least it's working):

```c++
httpd_handle_t server_handle = nullptr; // this variable should be set by httpd_start(…) function

esp_err_t httpd_ws_send_frame_to_all_clients(httpd_ws_frame_t *ws_pkt) {
    static constexpr size_t max_clients = CONFIG_LWIP_MAX_LISTENING_TCP;
    size_t fds = max_clients;
    int client_fds[max_clients] = {0};

    esp_err_t ret = httpd_get_client_list(server_handle, &fds, client_fds);

    if (ret != ESP_OK) {
        return ret;
    }

    for (int i = 0; i < fds; i++) {
        auto client_info = httpd_ws_get_fd_info(server_handle, client_fds[i]);
        if (client_info == HTTPD_WS_CLIENT_WEBSOCKET) {
            httpd_ws_send_frame_async(server_handle, client_fds[i], ws_pkt);
        }
    }

    return ESP_OK;
}
```