espressif/esp-idf #5407: Change the number of esp_http_client authorization retries

kaelef, 2020-06-06:
**Is your feature request related to a problem? Please describe.**

esp_http_client currently retries authorization attempts up to the maximum number of redirects.  This is a problem because if an incorrect password is provided, esp_http_client will retry that incorrect password 10 times by default.  This is likely to trigger an account lock-out on many servers.

**Describe the solution you'd like**

esp_http_client should only retry an authorization failure once, and only if the current authorization type is set to HTTP_AUTH_TYPE_NONE.  It can then attempt to set DIGEST or BASIC once based on the server response.

This change can be made easily by adding a  check in esp_http_client_add_auth()
`    if (client->connection_info.auth_type != HTTP_AUTH_TYPE_NONE)
        return;`

**Describe alternatives you've considered**

The current implementation provides no alternatives, so I have made a copy of esp_http_client and modified it.  I would like not to have my own version of esp_http_client, though.

**Additional context**

Alvin1Zhang, 2020-06-08:
@kaelef Thanks for raising this feature request.

shubhamkulkarni97, 2020-06-10:
@kaelef,

Can you try patch attached below?

It adds a configuration in `esp_http_client_config_t` to set maximum number of authorization attempts (by default, its value is 10).

Setting `max_authorization_retries` to 1 will solve your issue.

Please let us know if the patch works for you.
[authorization_retries.patch.zip](https://github.com/espressif/esp-idf/files/4756974/authorization_retries.patch.zip)

kaelef, 2020-06-11:
Yes, this will work - thank you!

closed 2020-07-10 by espressif-bot