espressif/esp-idf #5411: V4.1和V4.2不支持blufi吗？

guo652917087, 2020-06-08:
你好，
     我测试了esp-idf v4.0版本可以，v4.1和v4.2不行。报错：BLUFI_EXAMPLE: blufi_dh_negotiate_data_handler make public failed -12416。定位到dhm_check_range( &ctx->GY, &ctx->P ) 不符合。安卓和ios 一样报错，不支持4.1和4.2版本。

WCCWCC, 2020-06-08:
Hi @guo652917087 ,

* I use **App version 1.3.1**, **Blufi lib version 2.0.1**, **Support protocol version** and master branch can work normally.

*  Can you tell me your idf commit id, and app version information, I need to reproduce your problem.

guo652917087, 2020-06-08:
Hi,@WCCWCC
  安卓app：https://github.com/EspressifApp/EspBlufiForAndroid/releases ，v1.5.3
  ios: 苹果store 下载，版本1.1.0
 esp-idf:  release/v4.2 测试安卓和ios报错一样 make public failed -12416
               release/v4.1 测试安卓和ios报错一样 make public failed -12416
              release/v4.0 测试ok.

WCCWCC, 2020-06-08:
Hi @guo652917087 ,

* 我使用与你相同的 app 版本, 代码分支在 release/v4.1 ， 我依然没有复现这个问题。 

* 请问你的  release/v4.1 的commit 是多少？ 我的 commit id 是 d35226. 

* 为了方便复现问题，建议你把你使用的 安卓 手机型号告知一下。

guo652917087, 2020-06-08:
你好，
V4.2 ：cd4fa46f5372a552a67452313d246d4d09d2b39f
V4.1：84b51781c80740fda92784dafcfc96c13b0d8b66

我刚刚再次用iphone X测试确认问题存在。安卓手机要等明天上班，同事确认下手机型号。

使用V4.0：4c81978a3e2220674a432a588292a4c860eef27b
iphone x手机测试正常。

guo652917087, 2020-06-09:
@WCCWCC ，你好
    安卓测试，默认连进去是明文通信的，你要使用加密通信，触发密钥协商，就出现这个问题了。IOS app默认连进去就是使用加密通信，所以一连进去，blufi的例子，串口就打印报错，APP端发送自定义数据app就闪退了。

WCCWCC, 2020-06-09:
Hi @guo652917087 ,

我已经复现问题了，会尽快提供修复版本给你。

WCCWCC, 2020-06-10:
Hi @guo652917087 .

你将配置项 Component config -> mbedTLS -> Enable hardware MPI(bignum) acceleration 关闭就能正常工作了。我们会将这个问题报给维护这个功能的同事。

guo652917087, 2020-06-11:
@WCCWCC ，你好，
  我测试了，安照你的方法，虽然可行。但我的应用层反映好慢，其他的任务栈的程序都卡得很严重！！

WCCWCC, 2020-06-11:
Hi @guo652917087 ,

* 你可以先使用提供的 blufi demo 看一下是否会变慢呢，v4.0 与之后的版本 blufi 这边没有做任何改动呢。

* 描述应用层反映变慢的现象能再描述具体一点吗？

guo652917087, 2020-06-11:
@WCCWCC ,
   非常感谢！问题解决了。

closed 2020-06-11 by guo652917087