espressif/esp-idf #5412: i2c sometimes change bytes when uart interrupts occures

moggiozzi, 2020-06-08:
## Environment

- Development Kit:      Custom board
- Module or chip used:  ESP32-WROOM-32
- IDF version (run ``git describe --tags`` to find it): v4.1-beta1-266-g56c1646e2
- Build System:        idf.py
- Compiler version (run ``xtensa-esp32-elf-gcc --version`` to find it): xtensa-esp32-elf-gcc (crosstool-NG esp-2019r2) 8.2.0
- Operating System:     Windows
- (Windows only) environment type: ESP Command Prompt
- Using an IDE?: No
- Power Supply:         USB

## Problem Description

In my board mlx90640 connected to ESP32 i2c.
When communicate over uart and i2c at the same time I read incorrect bytes from i2c.

### Expected Behavior

Read correct bytes from i2c.

### Steps to reproduce

Run project from https://github.com/moggiozzi/mlxtest
Related issue https://github.com/melexis/mlx90640-library/issues/68

Alvin1Zhang, 2020-06-09:
Thanks for reporting. We have a fix internally under review, will share updates once the fix is available on GitHub.

pompompom, 2020-11-12:
Hi, 
Does this issue has been solved ? I am noticing the same behaviour, with my board

AxelLin, 2022-12-18:
> > Thanks for reporting. We have a fix internally under review, will share updates once the fix is available on GitHub.
> 
> @Alvin1Zhang Could you share which commit fixes this issue?

@Alvin1Zhang Could you share the commit fix? Or at least update the status?

ginkgm, 2022-12-20:
Hi @moggiozzi ,

We don't have a MLX90640 , is there a simpler example to reproduce the issue? 

Do you have external pull-ups on the I2C pins of your board? what's the value?

Is it possible for you to reproduce the issue at a fixed place? when the issue happens, what's on the I2C bus, what's the expected data and what's read instead?

Hi @pompompom ,

I have the same questions for you as above. And what's your HW and IDF version please?

Michael