espressif/esp-idf #5413: Independent Channel Control for Built-in I2S/DAC1/DAC2

physiii, 2020-06-08:
The [i2s_adc_dac](https://github.com/espressif/esp-idf/tree/master/examples/peripherals/i2s_adc_dac) example just mirrors the left and right channels into the built in DAC1/DAC2. But I want to generate a 1khz tone from DAC1 and 1khz tone with 180 degree phase shift from DAC2.

How do I i2s_write to left/right channels separately?

egnor, 2022-07-27:
[this seems obsolete and should probably be closed?]

This seems to be more of a question than a bug report?

I believe multiple channels work the same way when using the internal ADC/DAC as they are when using digital I2S -- each individual sample includes both left and right channel data (which means the sample size is twice as big when using stereo; e.g. for 16 bit sound, the samples are 32 bit values when stereo is enabled).

physiii, 2022-07-27:
I will test this out on the next project, closed for now.

closed 2022-07-27 by physiii