espressif/esp-idf #5414: Core Panic if RMT ISR executes during Preferences save

rljonesau, 2020-06-08:
I have been experiencing issues with core panics ever since I started using the RMT peripheral and it's standard driver.

Every time I do a stack trace, it has been a Preferences save that was interrupted by the RMT ISR.

A sample stack trace:

[image]

ESP32 IDF Version: v3.2.3-14-gd3e562907

koobest, 2020-06-08:
Hi,@rljonesau
Please try to comment this line: https://github.com/espressif/esp-idf/blob/055943e29346e77c50589c61a8a26101a8b35d7b/components/driver/rmt.c#L593

I think this issue is caused by accessing flash during cache disabling. 

thanks !!

rljonesau, 2020-06-08:
OK sounds quite plausible.
I will need to learn how to use the IDF with Platform IO, it is mainly an Arduino framework app.

But yes I do see that RMT RX BUFFER FULL message appearing at random times, due to noise from the 433MHz receiver I'm using.

Thank you for your reposnse.

rljonesau, 2020-06-28:
Hello, I have finally worked out how to recompile the IDF for Arduino, under PlatformIO.

Yes, commenting out that line solves the cache disabled error, avoiding the core panic.

I note the latest IDF logging has a new method ESP_DRAM_LOGE (et al) which sounds like it would fix the problem, and retain the error reporting. I tried using that, but the IDF release to match up with arduino-esp does not support the function. :-(

Should I push a Pull Request along with a description of the problem so at least the problem is eventually dealt with at some point in the future?

closed 2022-05-21 by espressif-bot