espressif/esp-idf #5415: [ble mesh] server generic level model event

chegewara, 2020-06-08:
I have implemented generic level model on server node. It seems to work because i can see arriving events, but here is problem:
- primary address of this node is 0x0002,
- node has 4 elements: (0x0002-0x0005),
- dst address is different than `param->model->element->element_addr`

```
I (117251) EXAMPLE: event 0x00, opcode 0x8206, src 0x0001, dst 0x0002
I (117251) EXAMPLE: level 0xffff8000, address 0x0005
I (732671) EXAMPLE: event 0x00, opcode 0x8206, src 0x0001, dst 0x0003
I (732671) EXAMPLE: level 0xfffffd70, address 0x0005
```
After adding more logs i can see that wrong model is passed to this event:
```
I (139252) EXAMPLE: event 0x00, opcode 0x8203, src 0x0001, dst 0xc001
I (140962) EXAMPLE: event 0x00, opcode 0x8203, src 0x0001, dst 0xc001
I (142032) EXAMPLE: event 0x00, opcode 0x8207, src 0x0001, dst 0xc001
I (142032) EXAMPLE: level 0x7fff, address 0x0003, group => 0xc001
I (142912) EXAMPLE: event 0x00, opcode 0x8207, src 0x0001, dst 0xc001
I (142912) EXAMPLE: level 0xffff8000, address 0x0005, group => 0x0000
```
ON/OFF model and level model, both are added to the same group. All works fine with `param->ctx.recv_dst`, just problem with `param->model` which seems to be always last element on node.

Thanks

Alvin1Zhang, 2020-06-09:
Thanks for reporting, we will look into.

Campou, 2020-06-23:
Hi @chegewara 

Based on the `onoff_server` example, I added a Generic Level Server model to the first and the third elements, and have a test based on the above description. My log output is like this:
```
    ESP_LOGI(TAG, "event 0x%02x, opcode 0x%04x, src 0x%04x, dst 0x%04x, model_id 0x%04x, element_addr 0x%04x",
        event, param->ctx.recv_op, param->ctx.addr, param->ctx.recv_dst, param->model->model_id, param->model->element->element_addr);
```
And the following is my test log.

[image]

If convenient, could you please share the test example on your side?

Thanks.

Alvin1Zhang, 2020-07-30:
@chegewara Thanks for reporting, will you please help share any updates or details for the issue? Thanks.

closed 2021-02-26 by Alvin1Zhang