espressif/esp-idf #5417: rev 4.0.1: ble gatt congested error

enricop, 2020-06-08:
Hello,
after upgrading our firmware to esp-idf 4.0.1, we can no longer complete some data transactions from our phone to the gatt server on the esp32.

In particular now we have the "congested" event that is triggered regularly when the communication is too busy, and after that, the following error appears, and the connection between the phone and the esp32 blocks.
`E (21442) BT_L2CAP: L2CAP - CID: 0x0004 cannot send, already congested            xmit_hold_q.count: 1 buff_quota: 20`

@Weijian-Espressif  @WCCWCC : we have seen the following commit 6ee978c6aa2c1d54dc24c0a37c02a1d09ae17ea1

Searching the forums and previous issues we found [this](https://www.esp32.com/viewtopic.php?p=52126#p52126) and [this](https://github.com/nkolban/esp32-snippets/issues/482).
We tried setting connection params in the following way, but nothing changes.
```
conn_params.latency = 0;
conn_params.max_int = 0x20;    // max_int = 0x20*1.25ms = 40ms
conn_params.min_int = 0x10;    // min_int = 0x10*1.25ms = 20ms
conn_params.timeout = 400;    // timeout = 400*10ms = 4000ms
esp_ble_gap_update_conn_params(&conn_params);
```
Is there a proper way to handle this. We can't afford now to implement a retry delayed mechanism when the congested event is triggered. We would prefer to instruct the peer device to have a specific data rate. but how?

Thanks for your support.

```
out: cfg_data_chunk {
	cid: 0x03, APPEND_CHUNK (p406)
	f_k: confdoc.json
	seq: 18
	payload: [0xfe 0x9f .. 0xba 0xf4]
}
I (21372) ble_rxtx: answering with a payload of length 19
I (21382) ble_rxtx: sending message with CID 0x05, DATA_FINISH (p406)
out: cfg_data_finish {
	cid: 0x05, DATA_FINISH (p406)
	f_k: confdoc.json
	hmac: 0xf5 0x7e .. 0x6c 0xa9
}
I (21402) ble_rxtx: answering with a payload of length 18
I (21402) BLE_GATT_service: GATT_WRITE_EVT end
I (21422) BLE_GATT_service: unhandled event 20
E (21422) BT_L2CAP: L2CAP - CID: 0x0004 cannot send, already congested            xmit_hold_q.count: 1 buff_quota: 20
E (21432) BT_L2CAP: L2CAP - CID: 0x0004 cannot send, already congested            xmit_hold_q.count: 1 buff_quota: 20
E (21442) BT_L2CAP: L2CAP - CID: 0x0004 cannot send, already congested            xmit_hold_q.count: 1 buff_quota: 20
E (21452) BT_L2CAP: L2CAP - CID: 0x0004 cannot send, already congested            xmit_hold_q.count: 1 buff_quota: 20
I (21462) BLE_GATT_service: unhandled event 20
I (21472) BLE_GATT_service: ESP_GATTS_CONF_EVT, status = 0, attr_handle 54
I (21472) BLE_GATT_service: ESP_GATTS_CONF_EVT, status = 0, attr_handle 54
I (21482) BLE_GATT_service: ESP_GATTS_CONF_EVT, status = 0, attr_handle 54
I (21492) BLE_GATT_service: ESP_GATTS_CONF_EVT, status = 0, attr_handle 54
I (21502) BLE_GATT_service: ESP_GATTS_CONF_EVT, status = 0, attr_handle 54
I (21502) BLE_GATT_service: ESP_GATTS_CONF_EVT, status = 0, attr_handle 54
I (21512) BLE_GATT_service: ESP_GATTS_CONF_EVT, status = 0, attr_handle 54
I (21522) BLE_GATT_service: ESP_GATTS_CONF_EVT, status = 0, attr_handle 54
I (21532) BLE_GATT_service: ESP_GATTS_CONF_EVT, status = 0, attr_handle 54
I (21532) BLE_GATT_service: ESP_GATTS_CONF_EVT, status = 0, attr_handle 54
I (21542) BLE_GATT_service: ESP_GATTS_CONF_EVT, status = 143, attr_handle 54
I (21552) BLE_GATT_service: ESP_GATTS_CONF_EVT, status = 129, attr_handle 54
I (21562) BLE_GATT_service: ESP_GATTS_CONF_EVT, status = 129, attr_handle 54
I (21562) BLE_GATT_service: ESP_GATTS_CONF_EVT, status = 129, attr_handle 54
I (21572) BLE_GATT_service: ESP_GATTS_CONF_EVT, status = 129, attr_handle 54
I (21582) BLE_GATT_service: ESP_GATTS_CONF_EVT, status = 0, attr_handle 54
I (21592) BLE_GATT_service: ESP_GATTS_CONF_EVT, status = 0, attr_handle 54
I (21592) BLE_GATT_service: ESP_GATTS_CONF_EVT, status = 0, attr_handle 54
I (21602) BLE_GATT_service: ESP_GATTS_CONF_EVT, status = 0, attr_handle 54
```
a screenshot of the error:
[congested_error.png](https://user-images.githubusercontent.com/667557/84052201-88a3dd00-a9b0-11ea-9553-d42cd443abfc.png)

our sdkconfig:
[sdkconfig.txt](https://github.com/espressif/esp-idf/files/4747075/sdkconfig.txt)

WCCWCC, 2020-06-09:
Hi @enricop ,

 * When you send data, you should detect congest events. When status is true, it means that the packet cache limit has been reached, and sending the data again will trigger an error log. 
   **E (21442) BT_L2CAP: L2CAP - CID: 0x0004 cannot send, already congested xmit_hold_q.count: 1 buff_quota: 20**

 * When you send a packet, you should check the status of congest, and then send it when congest status is not equal to true.

 * You can also refer to ble_throughput demo, check before sending how many buffers are available, and then send the corresponding number of packets.

cltsang, 2020-10-09:
I am seeing the same error. My code sends sequential notifications in a tight loop.

``` C++
  for (int i = 0; i < numChunks; i++) {
    auto chunk = longString.substring(i * chunkSize, (i + 1) * chunkSize);
    characteristic->setValue(chunk.c_str());
    characteristic->notify(true);
  }
```

It worked in arduino-esp32. But after converting it as a component of an idf project, it shows the same error.
My only relevant modification to get ble running is to add the line

`add_compile_definitions(-DARDUINO_ARCH_ESP32)`

to `CMakeLists.txt` (following https://github.com/espressif/arduino-esp32/issues/3436#issuecomment-700640080).

Is there anything else I missed?

Thank you.

WCCWCC, 2020-10-09:
Hi @cltsang ，

Maybe you can refer to the following code to solve the problem.  This API can query the number of buffers remaining available for the current connection.  

```c
            if (is_connect) {
                int free_buff_num = esp_ble_get_cur_sendable_packets_num(gl_profile_tab[PROFILE_A_APP_ID].conn_id);
                if(free_buff_num > 0) {
                    for( ; free_buff_num > 0; free_buff_num--) {
                        esp_ble_gatts_send_indicate(gl_profile_tab[PROFILE_A_APP_ID].gatts_if, gl_profile_tab[PROFILE_A_APP_ID].conn_id,
                                                    gl_profile_tab[PROFILE_A_APP_ID].char_handle,
                                                    sizeof(indicate_data), indicate_data, false);
                    }
                } else { //Add the vTaskDelay to prevent this task from consuming the CPU all the time, causing low-priority tasks to not be executed at all.
                    vTaskDelay( 10 / portTICK_PERIOD_MS );
                }
            }
```

cltsang, 2020-10-09:
@WCCWCC 
Thanks for the quick reply. I will try 😀 
But still, I couldn't find `esp_ble_get_cur_sendable_packets_num` getting called in the arduino-esp32 repository.
So I'm wondering: what is done in arduino-esp32 such that this check is not needed? Is it some project configuration done at compile time?

pawan-rathore-hsc, 2020-10-30:
@cltsang  Are you able to solve this, I am getting the same issue.

cltsang, 2020-11-01:
@pawan-rathore-hsc I'm sorry to report that I have no luck so far.

I tried calling `esp_ble_get_cur_sendable_packets_num` to make sure the buffer is available before calling `characteristic->notify()`, but even when I add `vTaskDelay` in the loop, the buffer is always available, and notify is send in a batch after the loop.
So I'm not quite what the problem is here. Maybe I should translate all my ble-related code from arduino-esp32 API's to idf API's only.

I am looking into using a more update-to-date version of esp-idf and arduino-esp32, instead of relying on the current supported version of platformio.

cltsang, 2020-11-11:
I got around building my project with esp-idf@release/v4.0, with arduino-esp32@idf-release/v4.0
Still got the same error.

WCCWCC, 2020-11-11:
@cltsang  
   To solve your problem，Can you tell me the commit id of idf， It is best to provide a link.
   Such as: https://github.com/espressif/esp-idf/tree/release/v4.0
 
  You have the function ``esp_ble_get_cur_sendable_packets_num`` in the code?

   The use of the function ``esp_ble_get_cur_sendable_packets_num`` is referred to https://github.com/espressif/esp-idf/tree/release/v4.0/examples/bluetooth/bluedroid/ble/ble_throughput.

cltsang, 2020-11-11:
@WCCWCC 
Thank you for trying to help.
I'm using [this commit of arduino-esp32](https://github.com/espressif/arduino-esp32/commit/d011dd7ef5d26e7ba2c2c093f422c534bacf4661) on [this commit of esp-idf](https://github.com/espressif/esp-idf/commit/a6e65a47a83495ae18d99994c30d5706d7379ca0)

I edited BLECharacteristic.cpp to add `#include "esp_gatt_common_api.h"`, as well as wrapping call to `esp_ble_gatts_send_indicate` [at line 519](https://github.com/espressif/arduino-esp32/blob/d011dd7ef5d26e7ba2c2c093f422c534bacf4661/libraries/BLE/src/BLECharacteristic.cpp#L519) with a loop like this
``` C++
		esp_err_t errRc;
		do {
			auto const free_buff_num = esp_ble_get_cur_sendable_packets_num(myPair.first);
			if (free_buff_num > 0) {
			errRc = ::esp_ble_gatts_send_indicate(
				getService()->getServer()->getGattsIf(),
				myPair.first,
				getHandle(), length, (uint8_t*)m_value.getValue().data(), !is_notification); // The need_confirm = false makes this a notify.
				break;
			} else {
				vTaskDelay(10);
			}
		} while(true);
```

However, `notify()` only goes through 10 times. After that, it's stuck, as if there is no rotation in the buffer, all while the BLE client has received 0 notifications.
So I cannot get `notify()` to send out packets one at a time.

But when I was building on arduino-esp32 only, instead of arduino-esp32 as a component of esp-idf, all seem to be working fine. What am I missing?

me-no-dev, 2020-11-12:
@cltsang I will suggest that you copy the sdkconfig from arduino into your project root, so that you are compiling with the same settings. Start then adjusting from there.

https://github.com/espressif/arduino-esp32/blob/d011dd7ef5d26e7ba2c2c093f422c534bacf4661/tools/sdk/sdkconfig

xiewenxiang, 2021-10-01:
I think it can add some flow control in application code:

https://github.com/espressif/esp-idf/blob/master/examples/bluetooth/bluedroid/ble/ble_throughput/throughput_server/main/example_ble_server_throughput.c

```c
    case ESP_GATTS_CONGEST_EVT:
        if (param->congest.congested) {
            can_send_notify = false;
        } else {
            can_send_notify = true;
            xSemaphoreGive(gatts_semaphore);
        }

        break;
```

```c

while(1){
        if (!can_send_notify) {
            int res = xSemaphoreTake(gatts_semaphore, portMAX_DELAY);
            assert(res == pdTRUE);
        } else {
            if (is_connect) {
                int free_buff_num = esp_ble_get_cur_sendable_packets_num(gl_profile_tab[PROFILE_A_APP_ID].conn_id);
                if(free_buff_num > 0) {
                    for( ; free_buff_num > 0; free_buff_num--) {
                        esp_ble_gatts_send_indicate(gl_profile_tab[PROFILE_A_APP_ID].gatts_if, gl_profile_tab[PROFILE_A_APP_ID].conn_id,
                                                    gl_profile_tab[PROFILE_A_APP_ID].char_handle,
                                                    sizeof(indicate_data), indicate_data, false);
                    }
                } else { //Add the vTaskDelay to prevent this task from consuming the CPU all the time, causing low-priority tasks to not be executed at all.
                    vTaskDelay( 10 / portTICK_PERIOD_MS );
                }
            }
        }
}
```