espressif/esp-idf #5420: freemodbus-master

LiThirteen, 2020-06-09:
请问ESP-IDF  v4.2中  关于freemdbus组件的主机模式，如何便捷的读取多个从机的特征值呢？
现在只知道在这里初始化的时候，能够指定某一个Slave addr， 如果有很多个，岂不是要写跟多个cid了嘛？
const mb_parameter_descriptor_t device_parameters[] = {
    // { CID,         Param Name,       Units,      Slave Addr,       Reg Type, Start, Size, Instance Offset, Data Type, Data Size, Parameter Options, Access Mode}
    { 0, STR("Test"), STR("__"),   MB_DEVICE_ADDR1, MB_PARAM_HOLDING, 0, 1,HOLD_OFFSET(data_buffer), PARAM_TYPE_U8, 1, OPTS( 0, 100, 1 ), PAR_PERMS_READ_WRITE_TRIGGER },
};

alisitsyn, 2020-06-09:
Hi @LiThirteen,
If you need to read parameters of many slaves you still need to specify the cid for every slave in the object dictionary. This is the approach using of modbus controller interface. I would propose you to define slave parameters as structures and read them as whole structure:
#define SLAVE_LENGTH_OF_DATA sizeof(input_reg_params)
#define SLAVE_LENGTH_OF_HOLDING sizeof(holding_reg_params)

const mb_parameter_descriptor_t device_parameters[] = {
    // { CID, Param Name, Units, Modbus Slave Addr, Modbus Reg Type, Reg Start, Reg Size, Instance Offset, Data Type, Data Size, Parameter Options, Access Mode}
    { CID_INP_DATA_SLAVE1, STR("Data_registers_slave1"), STR("Data_object"), MB_DEVICE_ADDR1, MB_PARAM_INPUT, 0, SLAVE_LENGTH_OF_DATA >> 1,
                    INPUT_OFFSET(input_data0), PARAM_TYPE_ASCII, SLAVE_LENGTH_OF_DATA, OPTS( -10, 10, 1 ), PAR_PERMS_READ_WRITE_TRIGGER },
    { CID_INP_DATA_SLAVE2, STR("Data_registers_slave2"), STR("__"), MB_DEVICE_ADDR2, MB_PARAM_INPUT, 0, SLAVE_LENGTH_OF_DATA >> 1,
                    INPUT_OFFSET(input_data0), PARAM_TYPE_ASCII, SLAVE_LENGTH_OF_DATA, OPTS( -10, 10, 1 ), PAR_PERMS_READ_WRITE_TRIGGER },
    { CID_HOLD_DATA_SLAVE1, STR("Holding_registers_slave1"), STR("Hold_object"), MB_DEVICE_ADDR1, MB_PARAM_HOLDING, 0, SLAVE_LENGTH_OF_HOLDING >> 1,
                    HOLD_OFFSET(test_regs), PARAM_TYPE_ASCII, SLAVE_LENGTH_OF_HOLDING, OPTS( -10, 10, 1 ), PAR_PERMS_READ_WRITE_TRIGGER },
    { CID_HOLD_DATA_SLAVE2, STR("Holding_registers_slave2"), STR("__"), MB_DEVICE_ADDR2, MB_PARAM_HOLDING, 0, SLAVE_LENGTH_OF_HOLDING >> 1,
                    HOLD_OFFSET(test_regs), PARAM_TYPE_ASCII, SLAVE_LENGTH_OF_HOLDING, OPTS( -10, 10, 1 ), PAR_PERMS_READ_WRITE_TRIGGER }
    ...
};

Then you can read and read/write them how it does the master_operation_func()  function. 
Note: There is a Modbus protocol limit for maximum number of registers read in one transaction = 120 registers.  
If this approach is not good for your application then you can expose and use the read write functions of freemodbus (eMBMasterReqReadHoldingRegister(), MBMasterReqReadInputRegister() and etc.) for your application how the modbus controller does it in components/freemodbus/serial_master/modbus_controller/mbc_serial_master.c: line 171.
In this case you can specify slave address and registers to read as parameters of the function.

LiThirteen, 2020-06-10:
@alisitsyn 好的

Alvin1Zhang, 2020-06-19:
@LiThirteen Would you please help share if any updates for the issue? Thanks.

LiThirteen, 2020-06-19:
@Alvin1Zhang  有两种方案：1、放弃封装好的modbus控制器接口，根据自己项目的需求来重写。2、使用modbus控制器接口，在定义时，把从机的id都写入进去。 具体实现代码我还没有写，我正在移植esp-idf种modbus-master组件到esp8266种并正在进行测试。

alisitsyn, 2020-06-22:
@LiThirteen,
Let me know if you have futher questions or need help with the implementation.

LiThirteen, 2020-06-28:
@alisitsyn Thank you

closed 2020-06-28 by LiThirteen