espressif/esp-idf #5421: [ble mesh] scene server, not restore scenes after reboot

chegewara, 2020-06-09:
Maybe i dont understand API, but i am assuming that scenes should be stored by low level code and re-loaded on restart to recall immediately.
I have build app with scenes (server and client) which works, store scenes and recall works. When esp32 is restarted i am getting this message:
`W (7068) BLE_MESH: scene_recall, Scene Number 0x0001 not exist`
which basically means i have to store new scenes all over again. Here is my related code:
```
static esp_ble_mesh_scene_register_t scenes[3] = {
    {.scene_value = NET_BUF_SIMPLE(10)},
    {.scene_value = NET_BUF_SIMPLE(10)},
    {.scene_value = NET_BUF_SIMPLE(10)}};

static esp_ble_mesh_scenes_state_t scene_state = {
    .scene_count = 3,
    .scenes = scenes};

ESP_BLE_MESH_MODEL_PUB_DEFINE(scene_pub, 2 + 5, ROLE_NODE);
static esp_ble_mesh_scene_srv_t scene_server = {
    .rsp_ctrl.get_auto_rsp = ESP_BLE_MESH_SERVER_AUTO_RSP,
    .rsp_ctrl.set_auto_rsp = ESP_BLE_MESH_SERVER_AUTO_RSP,
    .state = &scene_state,
};

static esp_ble_mesh_model_t root_models[] = {
    ESP_BLE_MESH_MODEL_CFG_SRV(&config_server),
    ESP_BLE_MESH_MODEL_GEN_ONOFF_SRV(&onoff_pub, &onoff_server0),
    ESP_BLE_MESH_MODEL_GEN_LEVEL_SRV(&level_pub_0, &level_server0),
    ESP_BLE_MESH_MODEL_SCENE_SRV(&scene_pub, &scene_server),
    ESP_BLE_MESH_MODEL_SCENE_SETUP_SRV(&scene_setup_pub, &scene_setup_server),
};
```

Campou, 2020-06-23:
Hi @chegewara 

Sorry for the late reply.

Currently the scene value is not stored in the flash from the mesh stack. We will add this feature later. Currenly recommend to store the corresponding value in the application layer with the nvs functions.

Thanks.

chegewara, 2020-07-07:
Hi @Campou 
thanks, i will probably store and restore scenes in my app, that will works.
Trying to do it now leads me to another issue/bug unless i misunderstood API and doing it wrong. I can open new issue if you want.

From mesh client i am calling `esp_ble_mesh_time_scene_client_set_state` which seems to work with `ESP_BLE_MESH_SERVER_AUTO_RSP` on server, scene is stored in ram and can be recalled later on.
Now i am using `ESP_BLE_MESH_SERVER_RSP_BY_APP` to store scenes and here is the problem. Check out logs:
```
I (138950) EXAMPLE: event 0x02, opcode 0x8247, src 0x000a, dst 0xc001
I (138950) EXAMPLE: Scene time set2 0x0002
I (147820) EXAMPLE: event 0x02, opcode 0x8243, src 0x000a, dst 0xc001
I (147820) EXAMPLE: Scene time set2 0x0000
```
code:
```
    ESP_LOGI(TAG, "event 0x%02x, opcode 0x%04x, src 0x%04x, dst 0x%04x",
             event, param->ctx.recv_op, param->ctx.addr, param->ctx.recv_dst);
```
Event 2 is `ESP_BLE_MESH_TIME_SCENE_SERVER_RECV_SET_MSG_EVT` which is nothing unusual, since i am calling function to set scene state, the problem i have there is no function to change scene state and since its working with AUTO_RSP i thought i am doing it correctly.
At the same time when i inspect opcodes i can see correct opcodes for `ESP_BLE_MESH_MODEL_OP_SCENE_RECALL_UNACK` and `ESP_BLE_MESH_MODEL_OP_SCENE_STORE_UNACK` (0x8243 and 0x8247), but event event that is expected (at least from example provided earlier) is `ESP_BLE_MESH_TIME_SCENE_SERVER_STATE_CHANGE_EVT`.

Thanks

chegewara, 2020-07-07:
With store/recall scene in app is another problem now. Recall event is always reporting recall scene 0:
```
I (17930) EXAMPLE: event 0x02, opcode 0x8247, src 0x0009, dst 0xc000
I (17930) EXAMPLE: Scene time store 0x0001
I (17940) board_store_scene: 1
I (25580) EXAMPLE: event 0x00, opcode 0x8203, src 0x0009, dst 0xc000
I (28580) EXAMPLE: event 0x00, opcode 0x8207, src 0x000a, dst 0xc001
I (28580) EXAMPLE: level 0x00004fad, address 0xc001, group => 0xc001
I (28680) EXAMPLE: event 0x00, opcode 0x8207, src 0x000a, dst 0xc001
I (28680) EXAMPLE: level 0x000052b0, address 0xc001, group => 0xc001
I (31960) EXAMPLE: event 0x02, opcode 0x8247, src 0x000a, dst 0xc001
I (31960) EXAMPLE: Scene time store 0x0002
I (31970) board_store_scene: 2
I (41990) EXAMPLE: event 0x02, opcode 0x8243, src 0x000a, dst 0xc001
I (41990) EXAMPLE: Scene time recall 0x0000
I (41990) : scene cmd => 8192[8192], 
I (42000) board_recall_scene: 0
I (44870) EXAMPLE: event 0x02, opcode 0x8243, src 0x000a, dst 0xc001
I (44870) EXAMPLE: Scene time recall 0x0000
I (44870) board_recall_scene: 0
I (47120) EXAMPLE: event 0x02, opcode 0x8243, src 0x000a, dst 0xc001
I (47120) EXAMPLE: Scene time recall 0x0000
I (47130) board_recall_scene: 0
I (48980) EXAMPLE: event 0x02, opcode 0x8243, src 0x000a, dst 0xc001
I (48980) EXAMPLE: Scene time recall 0x0000
I (48990) board_recall_scene: 0
```
code:
```
        case ESP_BLE_MESH_TIME_SCENE_SERVER_RECV_SET_MSG_EVT:
            switch (param->ctx.recv_op)
            {
                case ESP_BLE_MESH_MODEL_OP_SCENE_RECALL:
                case ESP_BLE_MESH_MODEL_OP_SCENE_RECALL_UNACK:{
                    ESP_LOGI(TAG, "Scene time recall 0x%04x", param->value.state_change.scene_recall.scene_number);
                    xTaskNotify(sceneHandle, (param->value.state_change.scene_recall.scene_number) | 0x2000, eSetValueWithOverwrite);
                    break;
                }
                case ESP_BLE_MESH_MODEL_OP_SCENE_STORE:
                case ESP_BLE_MESH_MODEL_OP_SCENE_STORE_UNACK: {
                    ESP_LOGI(TAG, "Scene time store 0x%04x", param->value.state_change.scene_store.scene_number);
                    xTaskNotify(sceneHandle, (param->value.state_change.scene_store.scene_number) | 0x1000, eSetValueWithOverwrite);
                    break;
                }
                case ESP_BLE_MESH_MODEL_OP_SCENE_DELETE:
                case ESP_BLE_MESH_MODEL_OP_SCENE_DELETE_UNACK:
                    ESP_LOGI(TAG, "Scene time delete 0x%04x", param->value.state_change.scene_delete.scene_number);
                    break;
                case ESP_BLE_MESH_MODEL_OP_SCENE_REGISTER_GET:
                case ESP_BLE_MESH_MODEL_OP_SCENE_STATUS:
                case ESP_BLE_MESH_MODEL_OP_SCENE_REGISTER_STATUS:
                    ESP_LOGI(TAG, "Scene time all 0x%04x", param->ctx.recv_op);
                    break;

                default:
                    break;
            }
            break;
```
ESP-IDF: `I (30) boot: ESP-IDF v4.2-dev-1905-g625bd5eb1-dirty`

chegewara, 2020-07-07:
Here you compare store/recall scene logs with AUTO_RSP:
```
I (21070) EXAMPLE: event 0x00, opcode 0x8247, src 0x0009, dst 0xc000
I (21070) EXAMPLE: Scene time store 0x0001
I (38390) EXAMPLE: event 0x00, opcode 0x8203, src 0x0009, dst 0xc000
I (41050) EXAMPLE: event 0x00, opcode 0x8203, src 0x0009, dst 0xc000
I (43970) EXAMPLE: event 0x00, opcode 0x8207, src 0x000a, dst 0xc001
I (43970) EXAMPLE: level 0x000046a4, address 0xc001, group => 0xc001
I (44070) EXAMPLE: event 0x00, opcode 0x8207, src 0x000a, dst 0xc001
I (44070) EXAMPLE: level 0x000049a7, address 0xc001, group => 0xc001
I (46800) EXAMPLE: event 0x00, opcode 0x8247, src 0x000a, dst 0xc001
I (46800) EXAMPLE: Scene time store 0x0002
I (49030) EXAMPLE: event 0x00, opcode 0x8243, src 0x000a, dst 0xc001
I (49030) EXAMPLE: Scene time recall 0x0001
I (57550) EXAMPLE: event 0x00, opcode 0x8243, src 0x000a, dst 0xc001
I (57550) EXAMPLE: Scene time recall 0x0002
```
Client code is the same, the same device used.

As you can see event is 0x00 instead of 0x02 and scene recall with correct scene number.

chegewara, 2020-07-23:
@Campou Any help with this?
This bug do not allow to progress and client cant send devices to production.

Campou, 2020-07-24:
Hi @chegewara 

Sorry for the late reply.

I will look into this issue and get back to you ASAP.

Thanks.

Campou, 2020-07-24:
Hi @chegewara 

The `esp_ble_mesh_time_scene_server_cb_value_t` (i.e. param->value) contains the following parameters,
```
esp_ble_mesh_time_scene_server_state_change_t state_change; /*!< ESP_BLE_MESH_TIME_SCENE_SERVER_STATE_CHANGE_EVT */
esp_ble_mesh_time_scene_server_recv_get_msg_t get;  /*!< ESP_BLE_MESH_TIME_SCENE_SERVER_RECV_GET_MSG_EVT */
esp_ble_mesh_time_scene_server_recv_set_msg_t set;  /*!< ESP_BLE_MESH_TIME_SCENE_SERVER_RECV_SET_MSG_EVT */
```

If the `ESP_BLE_MESH_SERVER_AUTO_APP` is used, the parameter `state_change` can be accessed, and if `ESP_BLE_MESH_SERVER_RSP_BY_APP` is used, the parameter `get` or `set` should be accessed depends on the event type.

So in the above code, since the event is `ESP_BLE_MESH_TIME_SCENE_SERVER_RECV_SET_MSG_EVT`, the ` param->value.state_change.scene_recall.scene_numbe` should be changed to ` param->value.set.scene_recall.scene_number`, same for the following code.
```
case ESP_BLE_MESH_TIME_SCENE_SERVER_RECV_SET_MSG_EVT:
    switch (param->ctx.recv_op)
    {
        case ESP_BLE_MESH_MODEL_OP_SCENE_RECALL:
        case ESP_BLE_MESH_MODEL_OP_SCENE_RECALL_UNACK:{
            ESP_LOGI(TAG, "Scene time recall 0x%04x", param->value.set.scene_recall.scene_number);
            xTaskNotify(sceneHandle, (param->value.set.set.scene_number) | 0x2000, eSetValueWithOverwrite);
            break;
        }
        case ESP_BLE_MESH_MODEL_OP_SCENE_STORE:
        case ESP_BLE_MESH_MODEL_OP_SCENE_STORE_UNACK: {
            ESP_LOGI(TAG, "Scene time store 0x%04x", param->value.set.scene_store.scene_number);
            xTaskNotify(sceneHandle, (param->value.set.scene_store.scene_number) | 0x1000, eSetValueWithOverwrite);
            break;
        }
        case ESP_BLE_MESH_MODEL_OP_SCENE_DELETE:
        case ESP_BLE_MESH_MODEL_OP_SCENE_DELETE_UNACK:
            ESP_LOGI(TAG, "Scene time delete 0x%04x", param->value.set.scene_delete.scene_number);
            break;
        case ESP_BLE_MESH_MODEL_OP_SCENE_REGISTER_GET:
        case ESP_BLE_MESH_MODEL_OP_SCENE_STATUS:
        case ESP_BLE_MESH_MODEL_OP_SCENE_REGISTER_STATUS:
            ESP_LOGI(TAG, "Scene time all 0x%04x", param->ctx.recv_op);
            break;

        default:
            break;
    }
    break;
```

Thanks.

chegewara, 2020-07-24:
I dont know how did i missed that, even if its a bit strange.

Thanks

chegewara, 2020-07-24:
Thanks @Campou, you were great of help
Now i can work with scenes.

I will close this issue for now, since scenes can be handled from app level.

closed 2020-07-24 by chegewara

ankit-thealchemist, 2021-04-04:
@chegewara  @Campou : Guys I am implementing the scene server and I have hard time in storing the scenes. Do we have the demo code for the scene server. Thanks for help in advance!!

chegewara, 2021-04-04:
@ankit-thealchemist What i had to do was to use NVS to store values in scene like structure when device receive event to set scene and then load values according to scene ID when get event arrive. It is not hard to implement and depends on your device configuration.

ankit-thealchemist, 2021-04-05:
Thanks for speedy reply! Will try to implement. Actually I am using nrf mesh app as a client. In that app, storing the scenes using the scene setup is not working.

ankit-thealchemist, 2021-04-05:
> Hi @chegewara
> 
> Sorry for the late reply.
> 
> Currently the scene value is not stored in the flash from the mesh stack. We will add this feature later. Currenly recommend to store the corresponding value in the application layer with the nvs functions.
> 
> Thanks.

hello, Do we have this functionality now??

hipal31, 2023-09-19:
> Hi @chegewara
> 
> Sorry for the late reply.
> 
> Currently the scene value is not stored in the flash from the mesh stack. We will add this feature later. Currenly recommend to store the corresponding value in the application layer with the nvs functions.
> 
> Thanks.

@Campou Is this functionality implemented in sdk ?

hipal31, 2023-09-21:
Hi @chegewara Can you give me code snippet to store and restore scenes?

chegewara, 2023-09-22:
Hi.
I cant share full code, but this is some good starting point:
```
static void _ble_mesh_scene_server_cb(esp_ble_mesh_time_scene_server_cb_event_t event,
                                             esp_ble_mesh_time_scene_server_cb_param_t *param)
{
    esp_ble_mesh_server_state_value_t state = {0};
    uint16_t primary_addr = esp_ble_mesh_get_primary_element_address();

    ESP_LOGI(TAG, "event 0x%02x, opcode 0x%04x, src 0x%04x, dst 0x%04x",
             event, param->ctx.recv_op, param->ctx.addr, param->ctx.recv_dst);

    switch (event)
    {
        case ESP_BLE_MESH_TIME_SCENE_SERVER_STATE_CHANGE_EVT:
            switch (param->ctx.recv_op)
            {
                case ESP_BLE_MESH_MODEL_OP_SCENE_RECALL:
                case ESP_BLE_MESH_MODEL_OP_SCENE_RECALL_UNACK:{
                    ESP_LOGI(TAG, "[0]Scene time recall 0x%04x", param->value.state_change.scene_recall.scene_number);
                    xTaskNotify(sceneHandle, (param->value.state_change.scene_recall.scene_number - 1) | 0x2000, eSetValueWithOverwrite);
                    ESP_LOG_BUFFER_HEX(__func__, scenes[param->value.state_change.scene_recall.scene_number - 1].scene_value, 10);
                    ESP_LOG_BUFFER_HEX(__func__, &scene_state, sizeof(scene_state));
                    break;
                }
                case ESP_BLE_MESH_MODEL_OP_SCENE_STORE:
                case ESP_BLE_MESH_MODEL_OP_SCENE_STORE_UNACK: {
                    ESP_LOGI(TAG, "[0]Scene time store 0x%04x", param->value.state_change.scene_store.scene_number);
                    ESP_LOG_BUFFER_HEX(__func__, scenes[param->value.state_change.scene_store.scene_number - 1].scene_value, 10);
                    ESP_LOG_BUFFER_HEX(__func__, &scene_state, sizeof(scene_state));
                    xTaskNotify(sceneHandle, (param->value.state_change.scene_store.scene_number - 1) | 0x1000, eSetValueWithOverwrite);
                    break;
                }
                case ESP_BLE_MESH_MODEL_OP_SCENE_DELETE:
                case ESP_BLE_MESH_MODEL_OP_SCENE_DELETE_UNACK:
                    ESP_LOGI(TAG, "[0]Scene time delete 0x%04x", param->value.state_change.scene_delete.scene_number);
                    break;
                case ESP_BLE_MESH_MODEL_OP_SCENE_REGISTER_GET:
                case ESP_BLE_MESH_MODEL_OP_SCENE_STATUS:
                case ESP_BLE_MESH_MODEL_OP_SCENE_REGISTER_STATUS:
                    ESP_LOGI(TAG, "[0]Scene time all 0x%04x", param->ctx.recv_op);
                    break;

                default:
                    break;
            }
            break;
        case ESP_BLE_MESH_TIME_SCENE_SERVER_RECV_GET_MSG_EVT:

            break;
        case ESP_BLE_MESH_TIME_SCENE_SERVER_RECV_SET_MSG_EVT:
            switch (param->ctx.recv_op)
            {
                case ESP_BLE_MESH_MODEL_OP_SCENE_RECALL:
                case ESP_BLE_MESH_MODEL_OP_SCENE_RECALL_UNACK:{
                    ESP_LOGI(TAG, "[1]Scene time recall 0x%04x", param->value.set.scene_recall.scene_number);
                    xTaskNotify(sceneHandle, (param->value.set.scene_recall.scene_number - 1) | 0x2000, eSetValueWithOverwrite);
                    break;
                }
                case ESP_BLE_MESH_MODEL_OP_SCENE_STORE:
                case ESP_BLE_MESH_MODEL_OP_SCENE_STORE_UNACK: {
                    ESP_LOGI(TAG, "[1]Scene time store 0x%04x", param->value.set.scene_store.scene_number);
                    xTaskNotify(sceneHandle, (param->value.set.scene_store.scene_number - 1) | 0x1000, eSetValueWithOverwrite);
                    break;
                }
                case ESP_BLE_MESH_MODEL_OP_SCENE_DELETE:
                case ESP_BLE_MESH_MODEL_OP_SCENE_DELETE_UNACK:
                    ESP_LOGI(TAG, "[1]Scene time delete 0x%04x", param->value.set.scene_delete.scene_number);
                    break;
                case ESP_BLE_MESH_MODEL_OP_SCENE_REGISTER_GET:
                case ESP_BLE_MESH_MODEL_OP_SCENE_STATUS:
                case ESP_BLE_MESH_MODEL_OP_SCENE_REGISTER_STATUS:
                    ESP_LOGI(TAG, "[1]Scene time all 0x%04x", param->ctx.recv_op);
                    break;

                default:
                    break;
            }
            break;
        case ESP_BLE_MESH_TIME_SCENE_SERVER_RECV_STATUS_MSG_EVT:

            break;

        default:
            ESP_LOGE(TAG, "Unknown Scene event 0x%02x", event);
            break;
    }
}
```
Im not sure if espressif ble mesh didnt change since then, but here is logic:
- scene number counts from 1 i believe, since i have -1 in few lines
-  this is my own logic to control store/load scenes and you can use own
```
 xTaskNotify(sceneHandle, (param->value.set.scene_store.scene_number - 1) | 0x1000, eSetValueWithOverwrite);
```
- when you store scene you just have to store all vital values for mesh device; in my case it was to store array of 3 ledc values in NVS
- scene values are loaded from NVS to "scene array" when esp32 restart

hipal31, 2023-09-22:
@chegewara thanks for quick reply. Yesterday, I tried my own logic and it worked as per my requirement.