espressif/esp-idf #5422: menuconfig fail on idf 4.0.1 by path conventions (in component.cmake)

F99, 2020-06-09:
Recently moved from 3.3 to 4.0.1 for integration.
hello_world sample runs as expected.
cloned a project from https://github.com/jeremyjh/ESP32_TFT_library/
`idf.py menuconfig` raises error:

```
CMake Error at C:/Users/F70/000-KL/FarmControlRoom/ESP_IDF_401/tools/cmake/component.cmake:280 (foreach):
  Syntax error in cmake code at

    C:/Users/F70/000-KL/FarmControlRoom/ESP_IDF_401/tools/cmake/component.cmake:280

  when parsing string

    C:/Users/F70/000-KL/FarmControlRoom/ESP32_TFT_library/main;C:\Users\F70\000-KL\FarmControlRoom\ESP_IDF_401/components

  Invalid character escape '\U'.
Call Stack (most recent call first):
  C:/Users/F70/000-KL/FarmControlRoom/ESP_IDF_401/tools/cmake/component.cmake:447 (__component_add_include_dirs)
  main/CMakeLists.txt:2 (idf_component_register)

-- Configuring incomplete, errors occurred!
See also "C:/Users/F70/000-KL/FarmControlRoom/ESP32_TFT_library/build/CMakeFiles/CMakeOutput.log".
cmake failed with exit code 1
```
original report here:
https://github.com/jeremyjh/ESP32_TFT_library/issues/8
similar bug here:
https://github.com/espressif/esp-idf/issues/4658
but the advice, replacing ldgen.mk, could not resolve the problem in cmake.

## Environment

- Development Kit:    ESP32-LyraT v4.3  [ESP32-Wrover-Kit|ESP32-DevKitC|ESP32-PICO-Kit|ESP32-LyraT|ESP32-LyraTD-MSC|none]
- Kit version -  (for WroverKit/PicoKit/DevKitC): [v1|v2|v3|v4]
- Module or chip used: ESP32-WROVER-B  [ESP32-WROOM-32|ESP32-WROOM-32D|ESP32-WROOM-32U|ESP32-WROVER|ESP32-WROVER-I|ESP32-WROVER-B|ESP32-WROVER-IB|ESP32-SOLO-1|ESP32-PICO-D4|ESP32]
- IDF version v4.0.1 (run ``git describe --tags`` to find it):
    // v3.2-dev-1148-g96cd3b75c
- Build System:    idf.py     [Make|CMake|idf.py]
- Compiler version  xtensa-esp32-elf-gcc (crosstool-NG esp-2019r2) 8.2.0  (run ``xtensa-esp32-elf-gcc --version`` to find it):
    // 1.22.0-80-g6c4433a
- Operating System:    Win7  [Windows|Linux|macOS]
- (Windows only) environment type: Plain Command Prompt  [MSYS2 mingw32|ESP Command Prompt|Plain Command Prompt|PowerShell].
- Using an IDE?: No [No|Yes (please give details)]
- Power Supply:       USB  [USB|external 5V|external 3.3V|Battery]

projectgus, 2020-06-23:
Hi @F99,

Thanks for being patient while someone got back to you about this.

Internally CMake always uses "CMake style paths", which use `/` for a delimiter on all platforms, not `\`.

When ESP-IDF reads the `IDF_PATH` variable internally, it uses [file(TO_CMAKE_PATH ...)](https://cmake.org/cmake/help/v3.0/command/file.html) to convert the slash format over.

The problematic path seems to be `C:\Users\F70\000-KL\FarmControlRoom\ESP_IDF_401/components` rather than the path related to the TFT library, which is already in "CMake" path style.

Is it possible for you to please share the top-level project CMakeLists.txt file? In particular, any parts to do with paths or setting `EXTRA_COMPONENT_DIRS` or similar variables?

F99, 2020-06-23:
Sure!
This is 
`C:\Users\F70\000-KL\FarmControlRoom\ESP32_TFT_library\main\CMakeLists.txt`:
[CMakeLists.txt](https://github.com/espressif/esp-idf/files/4819988/CMakeLists.txt)
And in the path above, it's :
[CMakeLists.txt](https://github.com/espressif/esp-idf/files/4820010/CMakeLists.txt)
Nice to hear from you ;)

In response to your post, I tried to replace the suspected variable by correct path convention:
```
set(SOURCES tft_demo.c)
idf_component_register(
        SRCS ${SOURCES}
        INCLUDE_DIRS
          ${CMAKE_CURRENT_LIST_DIR}
          C:/Users/F70/000-KL/FarmControlRoom/ESP_IDF_401/components
        REQUIRES
            tft
            spiffs
)
```
And, after initial fail and second try, it seems to work. I get the config menu. While for me this could be a workaround, it should be fixed, imho.

projectgus, 2020-06-24:
Hi @F99,

Thanks for that. I missed this line in the ESP32-TFT-Library.

I can explain the bug in the CMakeLists.txt file and how to fix it, will post on the linked issue.

Angus

projectgus, 2020-06-24:
Have commented on the linked issue and submitted a PR to fix it in that project.

Unfortunately there isn't anything we do to make this simpler from the ESP-IDF side. (We could add a step to convert all INCLUDE_DIRS entries to CMake paths automatically but this means that it's not possible to use backslash escapes in any of those parameters, and this be necessary sometimes e.g. unicode in paths.)

closed 2020-06-24 by projectgus