espressif/esp-idf #5424: Occasional crash on NVS operations

itayperl, 2020-06-09:
## Environment

- Development Kit:      **none**
- Module or chip used: **ESP32-WROVER-B**
- IDF version: **v4.2-dev-1660-g7d7521367**
- Build System:      **idf.py (ninja)**
- Compiler version: **(crosstool-NG esp-2020r1) 8.2.0**
- Operating System: **Linux**
- Using an IDE?: **No**
- Power Supply:       **external**

## Problem Description

Occasionally when performing NVS operations (read or write), the ESP32 panics with the following message:

```
Guru Meditation Error: Core  1 panic'ed (Cache disabled but cached memory region accessed)

Setting breakpoint at 0x40089db5 and returning...
0x40089db5: spi_flash_op_block_func at /dev/esp-idf/components/spi_flash/cache_utils.c:103 (discriminator 1)
```

This happens every ~10 NVS operations in my application.

From the stack trace, it looks like some WiFi ISR is causing the crash. I have disabled WiFi IRAM optimizations (both CONFIG_ESP32_WIFI_IRAM_OPT and CONFIG_ESP32_WIFI_RX_IRAM_OPT) to avoid overflowing the iram segment. Could this be related?

## Debug Logs

Stack trace of crashing NVS operation:

```
#0  0x4009865d in spi_flash_hal_poll_cmd_done (host=0x3ffbf0f8 <esp_flash_default_host_drv>)
    at /dev/esp-idf/components/soc/src/esp32/include/hal/spi_flash_ll.h:86
#1  0x400983f8 in spi_flash_hal_read (host=<optimized out>, buffer=0x3ffef7f0, address=<optimized out>, read_len=32)
    at /dev/esp-idf/components/soc/src/hal/spi_flash_hal_common.inc:121
#2  0x40098ed6 in spi_flash_chip_generic_read (chip=0x3ffbf0dc <default_chip>, buffer=<optimized out>, address=42336,
    length=<optimized out>) at /dev/esp-idf/components/spi_flash/spi_flash_chip_generic.c:155
#3  0x4008ab8a in esp_flash_read (chip=0x3ffbf0dc <default_chip>, buffer=0x3ffef7f0, address=42336, length=32)
    at /dev/esp-idf/components/spi_flash/esp_flash_api.c:555
#4  0x4008ace4 in spi_flash_read (src=42336, dstv=0x3ffef7f0, size=32)
    at /dev/esp-idf/components/spi_flash/esp_flash_api.c:760
#5  0x40124a80 in nvs::nvs_flash_read (srcAddr=42336, destAddr=0x3ffef7f0, size=32)
    at /dev/esp-idf/components/nvs_flash/src/nvs_ops.cpp:76
#6  0x401235d8 in nvs::Page::readEntry (this=0x3ffbd79c, index=<optimized out>, dst=...)
    at /dev/esp-idf/components/nvs_flash/src/nvs_page.hpp:206
#7  0x40123932 in nvs::Page::findItem (this=<optimized out>, nsIndex=6 '\006', datatype=-1, key=0x3ffef890 "item.b8",
    itemIndex=@0x3ffef810: 0, item=..., chunkIdx=chunkIdx@entry=255 '\377', chunkStart=chunkStart@entry=-1)
    at /dev/esp-idf/components/nvs_flash/src/nvs_page.cpp:835
#8  0x401228f5 in nvs::Storage::findItem (chunkStart=-1, chunkIdx=255 '\377', item=..., page=<synthetic pointer>: <optimized out>,
    key=0x3ffef890 "item.b8", datatype=-1, nsIndex=6 '\006', this=0x3ffbd6cc)
    at /dev/esp-idf/components/nvs_flash/src/nvs_storage.cpp:154
#9  nvs::Storage::eraseItem (this=0x3ffbd6cc, nsIndex=6 '\006', datatype=<optimized out>, key=0x3ffef890 "item.b8")
    at /dev/esp-idf/components/nvs_flash/src/nvs_storage.cpp:604
#10 0x401249ee in nvs::Storage::eraseItem (key=0x3ffef890 "item.b8", nsIndex=<optimized out>, this=<optimized out>)
    at /dev/esp-idf/components/nvs_flash/src/nvs_storage.hpp:96
#11 nvs::NVSHandleSimple::erase_item (this=<optimized out>, key=0x3ffef890 "item.b8")
    at /dev/esp-idf/components/nvs_flash/src/nvs_handle_simple.cpp:81
#12 0x40121325 in nvs_erase_key (c_handle=<optimized out>, key=0x3ffef890 "item.b8")
    at /dev/esp-idf/components/nvs_flash/src/nvs_api.cpp:277
```

Stack trace of the crashing ISR:
```
#0  panic_handler (frame=<optimized out>, pseudo_excause=true)
    at /dev/esp-idf/components/esp_system/port/panic_handler.c:495
#1  0x400925c0 in panicHandler (frame=0x3ffbe2c0)
    at /dev/esp-idf/components/esp_system/port/panic_handler.c:546
#2  0x400814dd in xt_highint4 ()
    at /dev/esp-idf/components/esp_system/port/esp32/dport_panic_highint_hdl.S:117
#3  0x40040025 in ?? ()
#4  0x4008b8b9 in lmacProcessRxSucData ()
#5  0x40084168 in _xt_lowint1 () at /dev/esp-idf/components/freertos/xtensa/xtensa_vectors.S:1105
```

More info from the panic frame:
```
(gdb) p/x *frame
$23 = {exit = 0xb33fffff, pc = 0x4008b8b9, ps = 0x60034, a0 = 0x8008beb2, a1 = 0x3ffbe380, a2 = 0x3ffc34c8, a3 = 0x3ffbf21c, 
  a4 = 0x80089f4a, a5 = 0x3ffef640, a6 = 0xfffffff7, a7 = 0x3ffc334c, a8 = 0xad, a9 = 0x3ffbe360, a10 = 0x3ffc29fc, 
  a11 = 0x3ffc2a04, a12 = 0x3ffc29fc, a13 = 0x60021, a14 = 0xfe, a15 = 0x1, sar = 0x1c, exccause = 0x7, excvaddr = 0x0, 
  lbeg = 0x4008e778, lend = 0x4008e794, lcount = 0x0, tmp0 = 0x3ffc29fc, tmp1 = 0x60021, tmp2 = 0x4008149c}

(gdb) x/i frame->pc
   0x4008b8b9 <lmacProcessRxSucData+9>: beqz.n  a8, 0x4008b8dd <lmacProcessRxSucData+45>
```

devanlai, 2020-06-09:
Hi @itayperl, the issue you're seeing sounds similar to what I'm seeing in #5354. In the minimal test case to reproduce it, it doesn't actually generate a useful panic trace, but when I've seen it crash sporadically in my actual application, I've seen similar traces where there is an SPI flash operation concurrent with `lmacProcessRxSucData()`, just like your trace.

Can you confirm whether you've also set `CONFIG_SPIRAM_ALLOW_BSS_SEG_EXTERNAL_MEMORY` vs if you have some other configuration that can also trigger this kind of error?

liuzfesp, 2020-06-11:
This issue should have been fixed by: 2c599137, please help to check it.

AxelLin, 2020-06-11:
@liuzfesp 
I cannot find commit 2c599137 from current master tree (v4.2-dev-1660-g7d75213674b4).

closed 2020-06-30 by Alvin1Zhang