espressif/esp-idf #5425: Issue initializing LAN8720A in 4.0.1

alexreinert, 2020-06-09:
I have a custom designed board with an ESP32-WROOM-32D and a LAN8720A PHY. As clock I use the clock output of the ESP on GPIO17.

Most times I call esp_eth_driver_install I will get this error:

```
E (389) lan8720: lan8720_pwrctl(333): power up failed
E (399) lan8720: lan8720_init(369): power control failed
E (399) esp_eth: esp_eth_driver_install(193): init phy failed
```

As the board worked without any issue with the Arduino core, I did some debugging an found the issue inside lan8720_pwrctl: On powerup the power_down register is written and directly read out to check, if the the chip is correctly powered up. Unfortunately, it takes about 10msec for the register to update (like the reset register in lan8720_reset).

After finding that, I modified the lan8720_pwrctl to have a timeout for the powerup in the same way, as in case of reset:

```
static esp_err_t lan8720_pwrctl(esp_eth_phy_t *phy, bool enable) {
    ESP_LOGI(TAG, "test");
    phy_lan8720_t *lan8720 = __containerof(phy, phy_lan8720_t, parent);
    esp_eth_mediator_t *eth = lan8720->eth;
    bmcr_reg_t bmcr;
    PHY_CHECK(eth->phy_reg_read(eth, lan8720->addr, ETH_PHY_BMCR_REG_ADDR, &(bmcr.val)) == ESP_OK,
              "read BMCR failed", err);
    if (!enable) {
        /* General Power Down Mode */
        bmcr.power_down = 1;
    } else {
        /* Normal operation Mode */
        bmcr.power_down = 0;
    }
    PHY_CHECK(eth->phy_reg_write(eth, lan8720->addr, ETH_PHY_BMCR_REG_ADDR, bmcr.val) == ESP_OK,
              "write BMCR failed", err);
    if (!enable) {
        PHY_CHECK(eth->phy_reg_read(eth, lan8720->addr, ETH_PHY_BMCR_REG_ADDR, &(bmcr.val)) == ESP_OK,
                    "read BMCR failed", err);
        PHY_CHECK(bmcr.power_down == 1, "power down failed", err);
    } else {
        /* wait for reset complete */
        uint32_t to = 0;
        for (to = 0; to < lan8720->reset_timeout_ms / 10; to++) {
            vTaskDelay(pdMS_TO_TICKS(10));
            PHY_CHECK(eth->phy_reg_read(eth, lan8720->addr, ETH_PHY_BMCR_REG_ADDR, &(bmcr.val)) == ESP_OK,
                    "read BMCR failed", err);
            if (bmcr.power_down == 0) {
                break;
            }
        }
        PHY_CHECK(to < lan8720->reset_timeout_ms / 10, "power up failed", err);
    }
    return ESP_OK;
err:
    return ESP_FAIL;
}
```

With this modified method, there is no issue anymore on esp_eth_driver_install.

Because I'm not complettly aware, how such bugfixes can be integreated in all affected release branches, I opened this issue and I hope, that you can include that fix in all releases beginning with 4.0.

Thanks,
Alex

suda-morris, 2020-06-10:
Thank your for your rich debug information. We will take a look soon @alexreinert

suda-morris, 2020-07-30:
Yes, we have added the fix in one of recent commit, but it's still not get merged and updated to Github, sorry for the inconvenience. :cry:

alexreinert, 2020-07-30:
Knowing, that it will be fixed in the next release is absolutly fine to me. Thanks!

alexreinert, 2020-08-26:
Very sad, that the fix is not included in the new 4.1 release. :-(

suda-morris, 2020-08-31:
So sorry, the 4.1 release test has just started before we merged that fix. But we have merged the fix to release/v4.1 in our internal channel, and will be pushed to github soon.

adam-esp, 2020-09-11:
please provide a link to the revised library.

adam-esp, 2020-09-11:
Where can i find "lan8720_pwrctl" functions?

adam-esp, 2020-09-11:
> 
> 
> Where can i find "lan8720_pwrctl" functions?

\esp-idf\components\esp_eth\src\esp_eth_phy_lan8720.c

closed 2020-09-16 by espressif-bot

Arcitec, 2025-04-02:
@suda-morris Hi!

I'm trying to find out in which version of ESP-IDF v3 SDK the bug fix was included. Or if this fix is only available in v4 and newer SDK?

I was digging around. All the source code files have changed names since this ticket was made, but I found the newest 87xx driver in ESP-IDF and looked for the "pdMS_TO_TICKS" delay command and found this:

https://github.com/espressif/esp-idf/blob/5c51472e82a58098dda8d40a1c4f250c374fc900/components/esp_eth/src/phy/esp_eth_phy_lan87xx.c#L320-L331

The comment talks about fixing a timing bug: Waiting a few milliseconds after setting the speed, before starting the driver?

But then I looked more and this code was added in 2022, two years after this ticket was made:

https://github.com/espressif/esp-idf/commit/78f791d4d5072f084196d015ad8ddd682d6fd853#diff-4e502998b078beb7b0b2192adfd2941804e573cbc42bd454f174820e04ad7ea2

So now I'm confused.

I just need to know if there's a version of ESP-IDF v3 that has the fix too.