espressif/esp-idf #5430: esp_modem_post_attach_start does not returning esp_modem_start_ppp result code

kapitan-u, 2020-06-11:
esp_modem_post_attach_start does not returning esp_modem_start_ppp result code

subsequently esp_netif_attach will not return appropriate error code
please consider replacing

return ESP_OK;

with

return esp_modem_start_ppp(dte);

static esp_err_t esp_modem_post_attach_start(esp_netif_t * esp_netif, void * args)
{
    esp_modem_netif_driver_t *driver = args;
    modem_dte_t *dte = driver->dte;
    const esp_netif_driver_ifconfig_t driver_ifconfig = {
            .driver_free_rx_buffer = NULL,
            .transmit = esp_modem_dte_transmit,
            .handle = dte
    };
    driver->base.netif = esp_netif;
    ESP_ERROR_CHECK(esp_netif_set_driver_config(esp_netif, &driver_ifconfig));
    return esp_modem_start_ppp(dte);
//    return ESP_OK;
}

Alvin1Zhang, 2020-06-11:
@kapitan-u Thanks for reporting. Would you please help provide more details as suggested in the [issue template](https://github.com/espressif/esp-idf/issues/new?assignees=&labels=&template=bug_report.md&title=)? Information like elf, sdk configuration, backtrace, log outputs, commit ID, hardware and etc. would help us debug further. Thanks.

kapitan-u, 2020-06-11:
ESP-IDF:          v4.2-dev-1660-g7d7521367

file

https://github.com/espressif/esp-idf/blob/master/examples/protocols/pppos_client/components/modem/src/esp_modem_netif.c

modem error not propagated to application and may not be detected and handled properly

------------ original code ----------
static esp_err_t esp_modem_post_attach_start(esp_netif_t * esp_netif, void * args)
{
esp_modem_netif_driver_t *driver = args;
modem_dte_t *dte = driver->dte;
const esp_netif_driver_ifconfig_t driver_ifconfig = {
.driver_free_rx_buffer = NULL,
.transmit = esp_modem_dte_transmit,
.handle = dte
};
driver->base.netif = esp_netif;
ESP_ERROR_CHECK(esp_netif_set_driver_config(esp_netif, &driver_ifconfig));
esp_modem_start_ppp(dte);
return ESP_OK;
}
-----------------------------------------------------

+++++++++++proposed solution++++++++++++++++
static esp_err_t esp_modem_post_attach_start(esp_netif_t * esp_netif, void * args)
{
esp_modem_netif_driver_t *driver = args;
modem_dte_t *dte = driver->dte;
const esp_netif_driver_ifconfig_t driver_ifconfig = {
.driver_free_rx_buffer = NULL,
.transmit = esp_modem_dte_transmit,
.handle = dte
};
driver->base.netif = esp_netif;
ESP_ERROR_CHECK(esp_netif_set_driver_config(esp_netif, &driver_ifconfig));
return esp_modem_start_ppp(dte);
}
++++++++++++++++++++

Alvin1Zhang, 2020-06-23:
@kapitan-u Thanks for reporting, we have a fix internally and under reviewing, hope the fix will be available on Github soon. Thanks.

kapitan-u, 2020-06-23:
Another suggestion for modem driver:

sim800.c

in function 

modem_dce_t *sim800_init(modem_dte_t *dte)

closed 2020-07-10 by espressif-bot