espressif/esp-idf #5433: [4.0.1] MCPWM fault not work

Wielebny666, 2020-06-11:
## Environment
- IDF version 4.0.1
- Compiler version xtensa-esp32-elf-gcc (crosstool-NG esp-2019r2) 8.2.0
- Operating System:     [Windows]
- (Windows only) environment type: [ESP Command Prompt]
- Using an IDE?: [Eclipse]
- Power Supply:         [USB]

## Problem Description
Changing the fault signal has no effect on outputs A and B.

### Code to reproduce this issue
```cpp
	mcpwm_pin_config_t pin_config =
		{
			.mcpwm0a_out_num = output_a,
			.mcpwm0b_out_num = output_b,
			.mcpwm_fault0_in_num = GPIO_NUM_14,
		};

	ESP_ERROR_CHECK(mcpwm_set_pin(MCPWM_UNIT_0, &pin_config));

	mcpwm_config_t mcpwm_config =
		{
			.frequency = MOSFET_PWM_INIT_FREQ,
			.cmpr_a = 50.0,
			.cmpr_b = 50.0,
			.counter_mode = MCPWM_UP_DOWN_COUNTER,
		};

	ESP_ERROR_CHECK(mcpwm_init(MCPWM_UNIT_0, MCPWM_TIMER_0, &mcpwm_config));
	ESP_ERROR_CHECK(mcpwm_stop(MCPWM_UNIT_0, MCPWM_TIMER_0));
	ESP_ERROR_CHECK(mcpwm_set_duty_type(MCPWM_UNIT_0, MCPWM_TIMER_0, MCPWM_OPR_A, MCPWM_DUTY_MODE_0));
	ESP_ERROR_CHECK(mcpwm_set_duty_type(MCPWM_UNIT_0, MCPWM_TIMER_0, MCPWM_OPR_B, MCPWM_DUTY_MODE_1));

	ESP_LOGI(TAG, "MCPWM Fault init...");
	gpio_pulldown_en(GPIO_NUM_14);
	ESP_ERROR_CHECK(mcpwm_fault_init(MCPWM_UNIT_0, MCPWM_HIGH_LEVEL_TGR, MCPWM_SELECT_F0));
	ESP_ERROR_CHECK(mcpwm_fault_set_cyc_mode(MCPWM_UNIT_0, MCPWM_TIMER_0, MCPWM_SELECT_F0, MCPWM_FORCE_MCPWMXA_LOW, MCPWM_FORCE_MCPWMXB_LOW));

	ESP_ERROR_CHECK(mcpwm_start(MCPWM_UNIT_0, MCPWM_TIMER_0));

```

I tried with mcpwm_basic_config_example.c and also not work.

Alvin1Zhang, 2020-06-12:
Thanks for reporting, we will look into.

Wielebny666, 2020-06-14:
Some conclusions?

koobest, 2020-06-15:
Hi, @Wielebny666 
I can not reproduce your issue with your test code.

[image]

I would suggest that you configure IO in this way:

```
    mcpwm_gpio_init(MCPWM_UNIT_0, MCPWM0A, 18);
    mcpwm_gpio_init(MCPWM_UNIT_0, MCPWM0A, 19);
    mcpwm_gpio_init(MCPWM_UNIT_0, MCPWM_FAULT_0, 5);
```

Wielebny666, 2020-06-15:
This is strange.

#doesn't work
```
		mcpwm_pin_config_t pin_config =
			{
				.mcpwm0a_out_num = output_a,
				.mcpwm0b_out_num = output_b,
				.mcpwm1a_out_num = GPIO_NUM_NC,
				.mcpwm1b_out_num = GPIO_NUM_NC,
				.mcpwm2a_out_num = GPIO_NUM_NC,
				.mcpwm2b_out_num = GPIO_NUM_NC,
				.mcpwm_sync0_in_num = GPIO_NUM_NC,
				.mcpwm_sync1_in_num = GPIO_NUM_NC,
				.mcpwm_sync2_in_num = GPIO_NUM_NC,
				.mcpwm_fault0_in_num = input_shutdown,
				.mcpwm_fault1_in_num =  GPIO_NUM_NC,
				.mcpwm_fault2_in_num =  GPIO_NUM_NC,
				.mcpwm_cap0_in_num =  GPIO_NUM_NC,
				.mcpwm_cap1_in_num =  GPIO_NUM_NC,
				.mcpwm_cap2_in_num =  GPIO_NUM_NC,
			};

		ESP_ERROR_CHECK(mcpwm_set_pin(MCPWM_UNIT_0, &pin_config));
```
#work
```
	mcpwm_gpio_init(MCPWM_UNIT_0, MCPWM0A, output_a);
	mcpwm_gpio_init(MCPWM_UNIT_0, MCPWM0B, output_b);
	mcpwm_gpio_init(MCPWM_UNIT_0, MCPWM_FAULT_0, input_shutdown);
```

koobest, 2020-06-15:
Hi, @Wielebny666 

`mcpwm_pin_config_t`  contains other pins' configuration, if not used, you should set it to -1.

Wielebny666, 2020-06-15:
I know.
gpio.h:
```
    GPIO_NUM_NC = -1,    /*!< Use to signal not connected to S/W */
```

wilhelmw201, 2020-06-18:
hi,
are you sure this doesn't work?
		mcpwm_pin_config_t pin_config =
			{
				.mcpwm0a_out_num = output_a,
				.mcpwm0b_out_num = output_b,
				.mcpwm1a_out_num = GPIO_NUM_NC,
				.mcpwm1b_out_num = GPIO_NUM_NC,
				.mcpwm2a_out_num = GPIO_NUM_NC,
				.mcpwm2b_out_num = GPIO_NUM_NC,
				.mcpwm_sync0_in_num = GPIO_NUM_NC,
				.mcpwm_sync1_in_num = GPIO_NUM_NC,
				.mcpwm_sync2_in_num = GPIO_NUM_NC,
				.mcpwm_fault0_in_num = input_shutdown,
				.mcpwm_fault1_in_num =  GPIO_NUM_NC,
				.mcpwm_fault2_in_num =  GPIO_NUM_NC,
				.mcpwm_cap0_in_num =  GPIO_NUM_NC,
				.mcpwm_cap1_in_num =  GPIO_NUM_NC,
				.mcpwm_cap2_in_num =  GPIO_NUM_NC,
			};

		ESP_ERROR_CHECK(mcpwm_set_pin(MCPWM_UNIT_0, &pin_config));

Here is a program that i used to test your code. It seems to run correctly. I suspect you didn't call mcpwm_fault_set_cyc_mode to specify the bahavior when a fault is detected, in which the default is MCPWM_ACTION_NO_CHANGE
`    printf("Hello world!\n");

    mcpwm_config_t pwm_config;
    pwm_config.frequency = 300;                        
    pwm_config.cmpr_a = 25.0;                               
    pwm_config.cmpr_b = 35.0;                               
    pwm_config.counter_mode = MCPWM_UP_DOWN_COUNTER;       
    pwm_config.duty_mode = MCPWM_DUTY_MODE_0;              
    mcpwm_init(MCPWM_UNIT_0, MCPWM_TIMER_0, &pwm_config);  
    
    int output_a = 4;
    int output_b = 2;
    int input_shutdown = 25;

    mcpwm_pin_config_t pin_config =
    {
        .mcpwm0a_out_num = output_a,
        .mcpwm0b_out_num = output_b,
        .mcpwm1a_out_num = GPIO_NUM_NC,
        .mcpwm1b_out_num = GPIO_NUM_NC,
        .mcpwm2a_out_num = GPIO_NUM_NC,
        .mcpwm2b_out_num = GPIO_NUM_NC,
        .mcpwm_sync0_in_num = GPIO_NUM_NC,
        .mcpwm_sync1_in_num = GPIO_NUM_NC,
        .mcpwm_sync2_in_num = GPIO_NUM_NC,
        .mcpwm_fault0_in_num = input_shutdown,
        .mcpwm_fault1_in_num =  GPIO_NUM_NC,
        .mcpwm_fault2_in_num =  GPIO_NUM_NC,
        .mcpwm_cap0_in_num =  GPIO_NUM_NC,
        .mcpwm_cap1_in_num =  GPIO_NUM_NC,
        .mcpwm_cap2_in_num =  GPIO_NUM_NC,
    };
    ESP_ERROR_CHECK(mcpwm_set_pin(MCPWM_UNIT_0, &pin_config));

    mcpwm_fault_set_cyc_mode(MCPWM_UNIT_0, MCPWM_TIMER_0, MCPWM_SELECT_F0, MCPWM_ACTION_FORCE_LOW, MCPWM_ACTION_FORCE_LOW);
    ESP_ERROR_CHECK(mcpwm_fault_init(MCPWM_UNIT_0, MCPWM_HIGH_LEVEL_TGR, MCPWM_SELECT_F0));

    for(;;)
    {
        printf("Fault pin: %d\n", gpio_get_level(input_shutdown));
        vTaskDelay(pdMS_TO_TICKS(1000));
    }`

Alvin1Zhang, 2020-06-23:
@Wielebny666 Would you please help share if any updates for this issue? Thanks.

Wielebny666, 2020-06-23:
Hi,

> [...] I suspect you didn't call mcpwm_fault_set_cyc_mode to specify the bahavior when a fault is detected, in which the default is MCPWM_ACTION_NO_CHANGE.
> 
Seriously?

my code:
```
[...]
ESP_ERROR_CHECK(mcpwm_fault_init(MCPWM_UNIT_0, MCPWM_HIGH_LEVEL_TGR, MCPWM_SELECT_F0)); 
ESP_ERROR_CHECK(mcpwm_fault_set_cyc_mode(MCPWM_UNIT_0, MCPWM_TIMER_0, MCPWM_SELECT_F0, MCPWM_FORCE_MCPWMXA_LOW, MCPWM_FORCE_MCPWMXB_LOW));
[...]
```

I changed nothing in the code except gpio initiation and it started working.

Difference between my and your code is in the order of calls mcpwm_fault_init and mcpwm_fault_set_cyc_mode.

I will check it again today...

Wielebny666, 2020-06-23:
OK, i admit mea culpa.

This work
```
	mcpwm_pin_config_t pin_config =
		{
			.mcpwm0a_out_num = output_a,
			.mcpwm0b_out_num = output_b,
			.mcpwm1a_out_num = GPIO_NUM_NC,
			.mcpwm1b_out_num = GPIO_NUM_NC,
			.mcpwm2a_out_num = GPIO_NUM_NC,
			.mcpwm2b_out_num = GPIO_NUM_NC,
			.mcpwm_sync0_in_num = GPIO_NUM_NC,
			.mcpwm_sync1_in_num = GPIO_NUM_NC,
			.mcpwm_sync2_in_num = GPIO_NUM_NC,
			.mcpwm_fault0_in_num = input_pwm,
			.mcpwm_fault1_in_num =  GPIO_NUM_NC,
			.mcpwm_fault2_in_num =  GPIO_NUM_NC,
			.mcpwm_cap0_in_num =  GPIO_NUM_NC,
			.mcpwm_cap1_in_num =  GPIO_NUM_NC,
			.mcpwm_cap2_in_num =  GPIO_NUM_NC,
		};

	ESP_ERROR_CHECK(mcpwm_set_pin(MCPWM_UNIT_0, &pin_config));

	mcpwm_config_t mcpwm_config =
		{
			.frequency = MOSFET_PWM_INIT_FREQ,
			.cmpr_a = 50.0,
			.cmpr_b = 50.0,
			.counter_mode = MCPWM_UP_DOWN_COUNTER, };

	ESP_ERROR_CHECK(mcpwm_init(MCPWM_UNIT_0, MCPWM_TIMER_0, &mcpwm_config));

	ESP_ERROR_CHECK(mcpwm_stop(MCPWM_UNIT_0, MCPWM_TIMER_0));
	ESP_ERROR_CHECK(mcpwm_set_duty_type(MCPWM_UNIT_0, MCPWM_TIMER_0, MCPWM_OPR_A, MCPWM_DUTY_MODE_0));
	ESP_ERROR_CHECK(mcpwm_set_duty_type(MCPWM_UNIT_0, MCPWM_TIMER_0, MCPWM_OPR_B, MCPWM_DUTY_MODE_1));

	ESP_ERROR_CHECK(mcpwm_fault_init(MCPWM_UNIT_0, MCPWM_LOW_LEVEL_TGR, MCPWM_SELECT_F0));
	ESP_ERROR_CHECK(mcpwm_fault_set_cyc_mode(MCPWM_UNIT_0, MCPWM_TIMER_0, MCPWM_SELECT_F0, MCPWM_FORCE_MCPWMXA_LOW, MCPWM_FORCE_MCPWMXB_LOW));

	ESP_ERROR_CHECK(mcpwm_start(MCPWM_UNIT_0, MCPWM_TIMER_0));
```
but, not work

```
	mcpwm_pin_config_t pin_config =
		{
			.mcpwm0a_out_num = output_a,
			.mcpwm0b_out_num = output_b,
			.mcpwm1a_out_num = GPIO_NUM_NC,
			.mcpwm1b_out_num = GPIO_NUM_NC,
			.mcpwm2a_out_num = GPIO_NUM_NC,
			.mcpwm2b_out_num = GPIO_NUM_NC,
			.mcpwm_sync0_in_num = GPIO_NUM_NC,
			.mcpwm_sync1_in_num = GPIO_NUM_NC,
			.mcpwm_sync2_in_num = GPIO_NUM_NC,
			.mcpwm_fault0_in_num =  GPIO_NUM_NC,
			.mcpwm_fault1_in_num = input_shutdown,
			.mcpwm_fault2_in_num =  GPIO_NUM_NC,
			.mcpwm_cap0_in_num =  GPIO_NUM_NC,
			.mcpwm_cap1_in_num =  GPIO_NUM_NC,
			.mcpwm_cap2_in_num =  GPIO_NUM_NC,
		};

	ESP_ERROR_CHECK(mcpwm_set_pin(MCPWM_UNIT_0, &pin_config));

	mcpwm_config_t mcpwm_config =
		{
			.frequency = MOSFET_PWM_INIT_FREQ,
			.cmpr_a = 50.0,
			.cmpr_b = 50.0,
			.counter_mode = MCPWM_UP_DOWN_COUNTER, };

	ESP_ERROR_CHECK(mcpwm_init(MCPWM_UNIT_0, MCPWM_TIMER_0, &mcpwm_config));

	ESP_ERROR_CHECK(mcpwm_stop(MCPWM_UNIT_0, MCPWM_TIMER_0));
	ESP_ERROR_CHECK(mcpwm_set_duty_type(MCPWM_UNIT_0, MCPWM_TIMER_0, MCPWM_OPR_A, MCPWM_DUTY_MODE_0));
	ESP_ERROR_CHECK(mcpwm_set_duty_type(MCPWM_UNIT_0, MCPWM_TIMER_0, MCPWM_OPR_B, MCPWM_DUTY_MODE_1));

	ESP_ERROR_CHECK(mcpwm_fault_init(MCPWM_UNIT_0, MCPWM_LOW_LEVEL_TGR, MCPWM_SELECT_F1));
	ESP_ERROR_CHECK(mcpwm_fault_set_cyc_mode(MCPWM_UNIT_0, MCPWM_TIMER_0, MCPWM_SELECT_F1, MCPWM_FORCE_MCPWMXA_LOW, MCPWM_FORCE_MCPWMXB_LOW));

	ESP_ERROR_CHECK(mcpwm_start(MCPWM_UNIT_0, MCPWM_TIMER_0));
```
and also not work:
```
	mcpwm_pin_config_t pin_config =
		{
			.mcpwm0a_out_num = output_a,
			.mcpwm0b_out_num = output_b,
			.mcpwm1a_out_num = GPIO_NUM_NC,
			.mcpwm1b_out_num = GPIO_NUM_NC,
			.mcpwm2a_out_num = GPIO_NUM_NC,
			.mcpwm2b_out_num = GPIO_NUM_NC,
			.mcpwm_sync0_in_num = GPIO_NUM_NC,
			.mcpwm_sync1_in_num = GPIO_NUM_NC,
			.mcpwm_sync2_in_num = GPIO_NUM_NC,
			.mcpwm_fault0_in_num = input_pwm,
			.mcpwm_fault1_in_num = input_shutdown,
			.mcpwm_fault2_in_num =  GPIO_NUM_NC,
			.mcpwm_cap0_in_num =  GPIO_NUM_NC,
			.mcpwm_cap1_in_num =  GPIO_NUM_NC,
			.mcpwm_cap2_in_num =  GPIO_NUM_NC,
		};

	ESP_ERROR_CHECK(mcpwm_set_pin(MCPWM_UNIT_0, &pin_config));

	mcpwm_config_t mcpwm_config =
		{
			.frequency = MOSFET_PWM_INIT_FREQ,
			.cmpr_a = 50.0,
			.cmpr_b = 50.0,
			.counter_mode = MCPWM_UP_DOWN_COUNTER, };

	ESP_ERROR_CHECK(mcpwm_init(MCPWM_UNIT_0, MCPWM_TIMER_0, &mcpwm_config));

	ESP_ERROR_CHECK(mcpwm_stop(MCPWM_UNIT_0, MCPWM_TIMER_0));
	ESP_ERROR_CHECK(mcpwm_set_duty_type(MCPWM_UNIT_0, MCPWM_TIMER_0, MCPWM_OPR_A, MCPWM_DUTY_MODE_0));
	ESP_ERROR_CHECK(mcpwm_set_duty_type(MCPWM_UNIT_0, MCPWM_TIMER_0, MCPWM_OPR_B, MCPWM_DUTY_MODE_1));

	ESP_ERROR_CHECK(mcpwm_fault_init(MCPWM_UNIT_0, MCPWM_LOW_LEVEL_TGR, MCPWM_SELECT_F0));
	ESP_ERROR_CHECK(mcpwm_fault_set_cyc_mode(MCPWM_UNIT_0, MCPWM_TIMER_0, MCPWM_SELECT_F0, MCPWM_FORCE_MCPWMXA_LOW, MCPWM_FORCE_MCPWMXB_LOW));

	ESP_ERROR_CHECK(mcpwm_fault_init(MCPWM_UNIT_0, MCPWM_LOW_LEVEL_TGR, MCPWM_SELECT_F1));
	ESP_ERROR_CHECK(mcpwm_fault_set_cyc_mode(MCPWM_UNIT_0, MCPWM_TIMER_0, MCPWM_SELECT_F1, MCPWM_FORCE_MCPWMXA_LOW, MCPWM_FORCE_MCPWMXB_LOW));

	ESP_ERROR_CHECK(mcpwm_start(MCPWM_UNIT_0, MCPWM_TIMER_0));
```

when init via **mcpwm_gpio_init**
all configuration work correctly.

Wielebny666, 2020-06-25:
This problem is in all ver from 2.1 to master

Wielebny666, 2020-07-09:
Why nobody approved pull requests?
Is 4.0 branch still alive?

costaud, 2020-07-09:
fixed in master branch, will be backported to 4.0 soon.

Wielebny666, 2020-07-09:
> fixed in master branch, will be backported to 4.0 soon.

are you sure?

Wielebny666, 2020-07-29:
> fixed in master branch, will be backported to 4.0 soon.
20 days have passed and there is still no patch in branch 4.0

Wielebny666, 2020-07-31:
Please reopen.

Alvin1Zhang, 2020-08-03:
Thanks for reporting, and reopened the ticket, will update once the fix on release/4.0 is available on GitHub. Thanks.

Alvin1Zhang, 2020-08-05:
Thanks for reporting and sorry for slow turnaround. Fix on release/4.0 is available at https://github.com/espressif/esp-idf/commit/fcb0206c59e0859c71e3c3013540a9779a322c0f, feel free to reopen if the issue still happens.

Wielebny666, 2020-08-05:
> This problem is in all ver from 2.1 to master
Reopen, Please check other versions.

Wielebny666, 2020-09-07:
Relase v4.1 stable have this same problem.
REOPEN

Ethan-CYQ, 2020-09-08:
Hi, @Wielebny666 
Release/v4.1 have solved this problem in our internal gitlab, 
but have no synchronize to github yet. We will do that as soon as possible.

Alvin1Zhang, 2020-12-16:
Fix on release/4.0 is https://github.com/espressif/esp-idf/commit/fcb0206c59e0859c71e3c3013540a9779a322c0f, feel free to reopen. Thanks.

closed 2020-12-16 by Alvin1Zhang