espressif/esp-idf #5435: Base64 bad newline after 72 chars feature

SinanAkkoyun, 2020-06-11:
Implement a base64_encode function with add_pad = 0 and the base_64 char set, or at least reveal the option to set add_pad in base64.h

igrr, 2020-06-12:
@SinanAkkoyun do you mean you are calling `base64_encode` defined by the wpa_supplicant component? Functions defined by wpa_supplicant are not considered to be part of the public API, and can be changed/renamed/removed. Please consider using base64 functions from mbedtls instead:
https://tls.mbed.org/api/base64_8h.html#a87c42fcb843b964e68015a46aad5bffd.

SinanAkkoyun, 2020-06-12:
Oh thanks, I thought it was the other way around! And I thought the mbed base64 function is slower than the wpa one, is that wrong?

sagb2015, 2020-08-05:
@SinanAkkoyun For base64 execution it should take around same time in both cases. For other crypto operations, mbedtls functions may use hardware accelerators and are more optimised.

closed 2020-08-05 by Alvin1Zhang

SinanAkkoyun, 2020-08-05:
@sagb2015 ohh thank you so much! But isn't it so, that mbedtls https post and get have a way smaller buffer size limit than wolfssl, thus way slower in big GET and POST?