espressif/esp-idf #5436: SD_MMC works but after a random time fails until I power it off

hepa94meca, 2020-06-11:
**Hardware**

- ESP32_Wroover B
- uSD 32GB
- LAN8720
- IDE Arduino
- Upload Speed  921600
- Flash 80Mhz

**Description**
I follow the instructions of the connection on the next link.. https://github.com/espressif/esp-idf/tree/master/examples/storage/sd_card#note-about-gpio12
 
- D2   - GPIO 12 (PULL_UP by code => pinMode(12, INPUT_PULLUP)
- D3   - GPIO 13 (10Kh pull-up)
- CMD -GPIO 15 (10Kh pull-up)
- VSS   -GND
- VDD  -3.3V
- CLK   -GPIO 14 (10Kh pull-up)
- VSS   -GND 
- D0    -GPIO2  (10Kh pull-up)
- D1    -GPIO4  (10Kh pull-up)

When I try just the SD_MMC it works great without problem, I tasted 7 days continuously, but when I add Ethernet (LAN8720) after 2 or 3 days the SD stops working, I have to restart the ESP by  powering off other ways such as restarting by code doesn’t initialize the SD and it fails from the beginning .

I try also with SPI_SD protocol and happens more less the same, after some days the SD fails.

**Steps to reproduce.**
Use SD_MMC with Ethernet Connection, append some data on a file and wait until it fails 

The LAN8720 connection is the follow...
https://sautter.com/blog/ethernet-on-esp32-using-lan8720/ 

**Debug Messages**
E (30975) sdmmc_cmd: sdmmc_read_sectors_dma: sdmmc_send_cmd returned 0x107
E (30975) diskio_sdmmc: sdmmc_read_blocks failed (263)
E (32012) sdmmc_req: sdmmc_host_wait_for_event returned 0x107
E (32012) sdmmc_cmd: sdmmc_read_sectors_dma: sdmmc_send_cmd returned 0x107
E (32023) diskio_sdmmc: sdmmc_read_blocks failed (263)
E (33082) sdmmc_req: sdmmc_host_wait_for_event returned 0x107
E (33082) sdmmc_cmd: sdmmc_read_sectors_dma: sdmmc_send_cmd returned 0x107
E (33093) diskio_sdmmc: sdmmc_read_blocks failed (263)
[E][vfs_api.cpp:265] VFSFileImpl(): fopen(/sdcard/dbin.csv) failed

**Code**
`
void setup(void)
{
	Serial.begin(9600);
	

    pinMode(12, INPUT_PULLUP);
    if(!SD_MMC.begin()){
        Serial.println("Card Mount Failed");
        return;
    }

  WiFi.onEvent(WiFiEvent);
  ETH.begin(ETH_ADDR, ETH_POWER_PIN, ETH_MDC_PIN, ETH_MDIO_PIN, ETH_TYPE, ETH_CLK_MODE);
	InitServer();
	InitWebSockets();
}
long contador=0;
long previousMillisSD=0;
#define Minuto 60000
void loop(void)
{

  unsigned long currentMillis = millis(); 
if (currentMillis - previousMillisSD >=Minuto*10) {
   previousMillisSD = currentMillis;
   appendFile(SD_MMC, "/hello.txt", "World!\r\n");
  }
    
    
}

void appendFile(fs::FS &fs, const char * path, const char * message){
    Serial.printf("Appending to file: %s\n", path);

    File file = fs.open(path, FILE_APPEND);
    if(!file){
        Serial.println("Failed to open file for appending");
        return;
    }
    if(file.print(message)){
        Serial.println("Message appended");
    } else {
        Serial.println("Append failed");
    }
}
`

igrr, 2020-06-11:
Hi @hepa94meca, if restarting the ESP doesn't help, and you need to toggle the power, then it is likely that the SD card has entered a state where it doesn't respond to commands anymore. Do you have sufficient power supply and decoupling for all the components in the circuit (SD, Ethernet, ESP)?

When the issue happens, can you capture the exchange between the ESP and the SD card (using a logic analyzer) after resetting the ESP?

hepa94meca, 2020-06-11:
@igrr thanks for answer soon, I was doing some tests and I  realize that if I put a 
potentiometer to measure it with a GPIO and then disconnect the pot , the SD fails, I think that as you say is the power supply. Do you reccomend me to put some extra filters or something related in order to make more reliable the signal apart form increase the power supply.
Unfortunately I don´t have a logic analyzer I will buy one, and upload the pictures.

Alvin1Zhang, 2021-01-12:
@hepa94meca Thanks for reporting, would you please share if any updates for the issue? Thanks.

hepa94meca, 2021-01-12:
The problem was solved by increasing the power supply. Thank you so much.

tom-borcin, 2021-02-25:
Hello @hepa94meca,

thank you for reporting this issue. At Espressif we really appreciate it when our users report issues as it helps us make our products better. We see that your issue was resolved by increasing the power supply and because of that we are closing this issue to keep our backlog manageable.

If you don't agree with this decision, please feel free to reopen the issue or contact me directly.

Regards,

Tomas

closed 2021-02-25 by tom-borcin