espressif/esp-idf #5437: ESP-IDF Tools Installer v2.2 does not work

hamedshamkhali, 2020-06-11:
----------------------------- Delete below -----------------------------

**Reminder: If your issue is a general question, starts similar to "How do I..", or is related to 3rd party development kits/libs, please discuss this on our community forum at https://esp32.com instead.**

INSTRUCTIONS
============

Before submitting a new issue, please follow the checklist and try to find the answer.

- [ ] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [ ] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [ ] I have searched the issue tracker for a similar issue and not found a similar issue.

If the issue cannot be solved after the steps before, please follow these instructions so we can get the needed information to help you in a quick and effective fashion.

1. Fill in all the fields under **Environment** marked with [ ] by picking the correct option for you in each case and deleting the others.
2. Describe your problem.
3. Include [debug logs from the "monitor" tool](https://docs.espressif.com/projects/esp-idf/en/latest/get-started/idf-monitor.html#automatically-decoding-addresses), or [coredumps](https://docs.espressif.com/projects/esp-idf/en/latest/api-guides/core_dump.html).
4. Providing as much information as possible under **Other items if possible** will help us locate and fix the problem.
5. Use [Markdown](https://guides.github.com/features/mastering-markdown/) (see formatting buttons above) and the Preview tab to check what the issue will look like.
6. Delete these instructions from the above to the below marker lines before submitting this issue.

**IMPORTANT: If you do not follow these instructions and provide the necessary details, your issue may not be resolved.**

----------------------------- Delete above -----------------------------

## Environment
Windows 10 with latest updates applied.

## Problem Description
I cannot install ESP-IDF Tools and ESP-IDF using the official installer.
//Detailed problem description goes here.
The installation fails where v4.6.0.0-idf-20190628 is renaming to v4.6.0.0-idf-20190628.tmp 
[Setup Log 2020-06-11 #002.txt](https://github.com/espressif/esp-idf/files/4766182/Setup.Log.2020-06-11.002.txt)

### Expected Behavior

### Actual Behavior
WARNING: Rename d:\Users\hsham\.espressif\tools\mconf\v4.6.0.0-idf-20190628 to d:\Users\hsham\.espressif\tools\mconf\v4.6.0.0-idf-20190628.tmp failed, retrying...

### Steps to reproduce

1. install ESP-IDF-tool

// If possible, attach a picture of your setup/wiring here.

### Code to reproduce this issue

## Debug Logs

32.zip is already downloaded
                          WARNING: destination path already exists, removing
                          Extracting d:\Users\hsham\.espressif\dist\mconf-v4.6.0.0-idf-20190628-win32.zip to d:\Users\hsham\.espressif\tools\mconf\v4.6.0.0-idf-20190628
                          WARNING: Rename d:\Users\hsham\.espressif\tools\mconf\v4.6.0.0-idf-20190628 to d:\Users\hsham\.espressif\tools\mconf\v4.6.0.0-idf-20190628.tmp failed, retrying...
                          WARNING: Rename d:\Users\hsham\.espressif\tools\mconf\v4.6.0.0-idf-20190628 to d:\Users\hsham\.espressif\tools\mconf\v4.6.0.0-idf-20190628.tmp failed, retrying...
                          WARNING: Rename d:\Users\hsham\.espressif\tools\mconf\v4.6.0.0-idf-20190628 to d:\Users\hsham\.espressif\tools\mconf\v4.6.0.0-idf-20190628.tmp failed, retrying...
                          WARNING: Rename d:\Users\hsham\.espressif\tools\mconf\v4.6.0.0-idf-20190628 to d:\Users\hsham\.espressif\tools\mconf\v4.6.0.0-idf-20190628.tmp failed, retrying...
                          WARNING: Rename d:\Users\hsham\.espressif\tools\mconf\v4.6.0.0-idf-20190628 to d:\Users\hsham\.espressif\tools\mconf\v4.6.0.0-idf-20190628.tmp failed, retrying...
                          WARNING: Rename d:\Users\hsham\.espressif\tools\mconf\v4.6.0.0-idf-20190628 to d:\Users\hsham\.espressif\tools\mconf\v4.6.0.0-idf-20190628.tmp failed, retrying...
                          WARNING: Rename d:\Users\hsham\.espressif\tools\mconf\v4.6.0.0-idf-20190628 to d:\Users\hsham\.espressif\tools\mconf\v4.6.0.0-idf-20190628.tmp failed, retrying...
                          WARNING: Rename d:\Users\hsham\.espressif\tools\mconf\v4.6.0.0-idf-20190628 to d:\Users\hsham\.espressif\tools\mconf\v4.6.0.0-idf-20190628.tmp failed, retrying...
                          WARNING: Rename d:\Users\hsham\.espressif\tools\mconf\v4.6.0.0-idf-20190628 to d:\Users\hsham\.espressif\tools\mconf\v4.6.0.0-idf-20190628.tmp failed, retrying...
                          WARNING: Rename d:\Users\hsham\.espressif\tools\mconf\v4.6.0.0-idf-20190628 to d:\Users\hsham\.espressif\tools\mconf\v4.6.0.0-idf-20190628.tmp failed, retrying...
                          WARNING: Rename d:\Users\hsham\.espressif\tools\mconf\v4.6.0.0-idf-20190628 to d:\Users\hsham\.espressif\tools\mconf\v4.6.0.0-idf-20190628.tmp failed, retrying...
                          WARNING: Rename d:\Users\hsham\.espressif\tools\mconf\v4.6.0.0-idf-20190628 to d:\Users\hsham\.espressif\tools\mconf\v4.6.0.0-idf-20190628.tmp failed, retrying...
                          WARNING: Rename d:\Users\hsham\.espressif\tools\mconf\v4.6.0.0-idf-20190628 to d:\Users\hsham\.espressif\tools\mconf\v4.6.0.0-idf-20190628.tmp failed, retrying...
                          WARNING: Rename d:\Users\hsham\.espressif\tools\mconf\v4.6.0.0-idf-20190628 to d:\Users\hsham\.espressif\tools\mconf\v4.6.0.0-idf-20190628.tmp failed, retrying...
                          WARNING: Rename d:\Users\hsham\.espressif\tools\mconf\v4.6.0.0-idf-20190628 to d:\Users\hsham\.espressif\tools\mconf\v4.6.0.0-idf-20190628.tmp failed, retrying...
                          WARNING: Rename d:\Users\hsham\.espressif\tools\mconf\v4.6.0.0-idf-20190628 to d:\Users\hsham\.espressif\tools\mconf\v4.6.0.0-idf-20190628.tmp failed, retrying...
                          WARNING: Rename d:\Users\hsham\.espressif\tools\mconf\v4.6.0.0-idf-20190628 to d:\Users\hsham\.espressif\tools\mconf\v4.6.0.0-idf-20190628.tmp failed, retrying...
[... 36 lines cut ...]
                          WARNING: Rename d:\Users\hsham\.espressif\tools\mconf\v4.6.0.0-idf-20190628 to d:\Users\hsham\.espressif\tools\mconf\v4.6.0.0-idf-20190628.tmp failed, retrying...
                          WARNING: Rename d:\Users\hsham\.espressif\tools\mconf\v4.6.0.0-idf-20190628 to d:\Users\hsham\.espressif\tools\mconf\v4.6.0.0-idf-20190628.tmp failed, retrying...
                          WARNING: Rename d:\Users\hsham\.espressif\tools\mconf\v4.6.0.0-idf-20190628 to d:\Users\hsham\.espressif\tools\mconf\v4.6.0.0-idf-20190628.tmp failed, retrying...
                          WARNING: Rename d:\Users\hsham\.espressif\tools\mconf\v4.6.0.0-idf-20190628 to d:\Users\hsham\.espressif\tools\mconf\v4.6.0.0-idf-20190628.tmp failed, retrying...
                          WARNING: Rename d:\Users\hsham\.espressif\tools\mconf\v4.6.0.0-idf-20190628 to d:\Users\hsham\.espressif\tools\mconf\v4.6.0.0-idf-20190628.tmp failed, retrying...
                          WARNING: Rename d:\Users\hsham\.espressif\tools\mconf\v4.6.0.0-idf-20190628 to d:\Users\hsham\.espressif\tools\mconf\v4.6.0.0-idf-20190628.tmp failed, retrying...
                          WARNING: Rename d:\Users\hsham\.espressif\tools\mconf\v4.6.0.0-idf-20190628 to d:\Users\hsham\.espressif\tools\mconf\v4.6.0.0-idf-20190628.tmp failed, retrying...
                          WARNING: Rename d:\Users\hsham\.espressif\tools\mconf\v4.6.0.0-idf-20190628 to d:\Users\hsham\.espressif\tools\mconf\v4.6.0.0-idf-20190628.tmp failed, retrying...
                          WARNING: Rename d:\Users\hsham\.espressif\tools\mconf\v4.6.0.0-idf-20190628 to d:\Users\hsham\.espressif\tools\mconf\v4.6.0.0-idf-20190628.tmp failed, retrying...
                          WARNING: Rename d:\Users\hsham\.espressif\tools\mconf\v4.6.0.0-idf-20190628 to d:\Users\hsham\.espressif\tools\mconf\v4.6.0.0-idf-20190628.tmp failed, retrying...
                          WARNING: Rename d:\Users\hsham\.espressif\tools\mconf\v4.6.0.0-idf-20190628 to d:\Users\hsham\.espressif\tools\mconf\v4.6.0.0-idf-20190628.tmp failed, retrying...
                          WARNING: Rename d:\Users\hsham\.espressif\tools\mconf\v4.6.0.0-idf-20190628 to d:\Users\hsham\.espressif\tools\mconf\v4.6.0.0-idf-20190628.tmp failed, retrying...
[... cut ...]
2020-06-11 11:54:16.297   Message box (OK):
                          Installation has failed with exit code 1
2020-06-11 11:56:22.547   User chose OK.
2020-06-11 11:56:22.557   Message box (Yes/No):
                          Installation log has been created, it may contain more information about the problem.
                          Display the installation log now?
2020-06-11 11:56:23.632   User chose Yes.
2020-06-11 11:56:23.668   CurStepChanged raised an exception.

## Other items if possible

igrr, 2020-06-11:
Could you please try removing both d:\Users\hsham\.espressif\tools\mconf\v4.6.0.0-idf-20190628 and d:\Users\hsham\.espressif\tools\mconf\v4.6.0.0-idf-20190628.tmp, and then try running the installer again?

hamedshamkhali, 2020-06-11:
I did delete from both mconf and cmake and it worked. Although I had to install some packages on Anaconda manually. Thanks

Alvin1Zhang, 2020-06-12:
@hamedshamkhali Thanks for reporting and glad to hear it worked. Has your issue been resolved? Thanks.

Alvin1Zhang, 2020-06-16:
@hamedshamkhali Glad to hear it worked. Feel free to reopen if the issue still happens. Thanks.

closed 2020-06-16 by Alvin1Zhang