espressif/esp-idf #5440: Unable to start Mesh network

dshil, 2020-06-12:
## Environment

- Module or chip used:  ESP32-WROVER-B
- IDF version: v4.0.1
- Build System: idf.py (cmake - the same)
- Compiler version: xtensa-esp32-elf-gcc (crosstool-NG esp-2019r2) 8.2.0
- Operating System: Linux
- Using an IDE: No
- Power Supply: USB

## Problem Description

I have a ESP-32 board that communicate with remote server using MQTT. Recently I successfully migrated to v4.0.1 from 3.3.1. Thus, now I am able to use new features. I was especially interested in using new event loop library with IP, WiFi and Mesh events. I have the following workflow: 2 ESP-32 boards are connected to the remote server and communicate with this server using MQTT. I want to build private mesh network between these 2 devices and being able to continue using WiFi.

I setup my mesh as:

```cpp
    ESP_ERROR_CHECK(esp_mesh_init());
    ESP_ERROR_CHECK(esp_mesh_set_max_layer(CONFIG_MESH_MAX_LAYER));
    ESP_ERROR_CHECK(esp_mesh_set_vote_percentage(1));
    ESP_ERROR_CHECK(esp_mesh_set_ap_assoc_expire(10));
    
    // I don't want my nodes to connect to the internet on the mesh layer.
    esp_mesh_fix_root(true);

    mesh_cfg_t cfg;
    memset(&cfg, 0, sizeof(cfg));
    cfg.crypto_funcs = &g_wifi_default_mesh_crypto_funcs;
    cfg.channel = CONFIG_MESH_CHANNEL;
    cfg.mesh_ap.max_connection = CONFIG_MESH_AP_CONNECTIONS;

    memcpy((uint8_t*)&cfg.mesh_id, MESH_ID, 6);

    ESP_ERROR_CHECK(esp_mesh_set_config(&cfg));

    ESP_ERROR_CHECK(
        esp_event_handler_register(MESH_EVENT, ESP_EVENT_ANY_ID, &handle_event_, this));
```

And when I flash and then run the program it is crashed each time with the following:

```plain
Error! Should enable WiFi modem sleep when both WiFi and Bluetooth are enabled!!!!!!
```

I read about `ESP32 Wi-Fi Power-saving Mode` and try to play with `esp_wifi_set_ps()` but the result is the same.

Since, mesh implementation is hidden (@igrr any chance you will make it public? :) ) I can't see source code and figure out what to do. It would be nice if somebody from the mesh team can help me to resolve the issue (probably @dongdong004).

ESP-iPENCIL, 2020-06-15:
will fix this issue and let you know after the merge. @dshil @dongdong004

dshil, 2020-06-15:
Hi @ESP-iPENCIL and thanks! Feel free to ping me if you will need some additional information.

Alvin1Zhang, 2020-07-01:
Thanks for reporting and late reply. We have a fix under internal reviewing, the issue will be updated and closed once the fix is available on GitHub.

dshil, 2020-07-02:
Hi @Alvin1Zhang, We are really looking forward for this fix. Without it we can't use mesh.

closed 2020-07-10 by espressif-bot

dshil, 2020-07-29:
@ydesp @Alvin1Zhang in which release could I expect to see this patch?

Alvin1Zhang, 2020-07-30:
Thanks for reporting, and sorry for slow turnaround. The fix has been available on master branch, we are back porting this fix to release/4.0 now, thanks for your patience in waiting for the fix on release/4.0. Thanks.

dshil, 2020-08-04:
Hi @Alvin1Zhang , any updates?

You told only about release/4.0, but what about 4.2, 4.1?

dshil, 2020-08-04:
The required cherry-pick was found, thanks all involved!

dshil, 2020-08-18:
@ydesp I was finally able to check the fix and it seems it didn't work. I still have the mentioned panic:

```cpp
I (2401) coexist: [1569027] Error! Should enable WiFi modem sleep when both WiFi and Bluetooth a
abort() was called at PC 0x4012f49a on core 0
```

The only difference is that we recently migrated from bluedroid BLE to NimBLE stack, probably it can be an issue. I double checked the used version of esp-idf and we have the fix you did previously:

```
commit e5fed6d67d8bb8c447881cc9fad459cc86d39a7e
Merge: 655772ec2 334052194
Author: Jiang Jiang Jian <jack@espressif.com>
Date:   Wed Jul 1 13:51:30 2020 +0800

    Merge branch 'mesh/bugfix_coexist_warning_4.2' into 'release/v4.2'
    
    mesh: remove coexist warnings in ESP_WIFI_MESH (backport 4.2)
    
    See merge request espressif/esp-idf!9412

commit 334052194e9d596e986950d452287c3bbefa3d56
Author: chenyudong <chenyudong@espressif.com>
Date:   Tue Jun 30 16:33:08 2020 +0800

    mesh: remove coexist warnings in ESP_WIFI_MESH
    
    1. remove warning log when mesh is started:
    Error! Should use default passive scan time parameter for WiFi scan when Bluetooth is enabled!!!!!!
    2. remove abort when mesh is started:
    Error! Should enable WiFi modem sleep when both WiFi and Bluetooth are enabled!!!!!!
```

@ydesp , @ESP-iPENCIL , @Alvin1Zhang

dshil, 2020-08-19:
I will open another issue.

iscapla, 2020-09-21:
Hello,
Any idea about this issue?
I'm developing under v4.1 with Wifi Mesh and BLE at the same time.

dshil, 2020-09-21:
Hi @iscapla, the issue was fixed and fix was cherry-picked to all branches mentioned above. What kind of problem do you have?

iscapla, 2020-09-21:
My issue is the same as you describe above but mixing into the same project the examples of GATT server and Mesh internal_communication.
In addition, I'm trying to reduce power consumption as mush as possible. For this, I followed some recommendation from this post #947 .
Could It be any problem with Mesh/BLE cooperation?

If you need the example project I can provide it tomorrow.

Thanks.

iscapla, 2020-09-22:
Hi,
Here is my example project: 
[mesh_wifi_ble.zip](https://github.com/espressif/esp-idf/files/5259549/mesh_wifi_ble.zip)

I'm using esp-idf release v4.1

Thanks so much

dshil, 2020-09-22:
Hi @iscapla, make sure you properly setup sleep mode. 

You need:

```
ESP_ERROR_CHECK(esp_wifi_set_ps(WIFI_PS_MIN_MODEM));
```

Probably you use `WIFI_PS_NONE`.

iscapla, 2020-09-22:
Hi @dshil, my initialization steps are:
1. Flash init
2. PM configuration (MAX=160, MIN=40)
3. Netif init
4. Wifi init
5. esp_wifi_set_ps(WIFI_PS_MIN_MODEM). In my case, I didn't check the return code (let me check it)
6. BLE initialization as the GATT server example use + bt_sleep_enable after controller initialization
7. More configuration about wifi (example)
8. All mesh configuration.

In menuconfig only BLE modem sleep is enable because I couldn't find any near this option related to wifi. Is there any?

iscapla, 2020-09-23:
I just solved my problem using master branch.
Thanks.