espressif/esp-idf #5441: Ability to change the name IDF_PATH

howroyd, 2020-06-12:
I've been developing on ESP32 for a long time now, on Windows using the normal setup described in the documentation.

Now I need to do some work on the ESP8266 which I am doing on release/v3.3

The only issue is that each time I want to change which device I am compiling for I need to change the PATH variable IDF_PATH to point to the respective folder for that chip.

It would be very handy if we could change the name/word "IDF_PATH" within the toolchain to allow for both to be used at the same time, without all the issues with Windows changing paths, restarting software etc.

e.g. IDF32_PATH and IDF8266_PATH would be my choice, which I could then edit the respective toolchains to use that new path name.

igrr, 2020-06-12:
If you are using the CMake build system, `idf.py` should be able to infer IDF_PATH automatically. So you might not actually need to set this environment variable at the system level.

For Make based projects you can do something like this in your project makefile, before including project.mk:
```make
ifndef IDF_PATH
ifdef ESP32_IDF_PATH
export IDF_PATH := $(ESP32_IDF_PATH)
endif
endif
```

For CMake based projects where you don't use idf.py, but use plain CMake or some IDE which integrates with CMake, like Eclipse or CLion, you can do the same thing in the project CMakeLists file: check if `IDF_PATH` is not set and `ESP32_IDF_PATH` is set, export IDF_PATH into the environment.

Final note, since ESP-IDF 4.0, it is no longer required to set any variables (including IDF_PATH) at system level. IDF Tools Installer for Windows creates a shortcut to a CMD prompt, which sets up all required variables (IDF_PATH, PATH, Python virtual environment) when the command prompt is launched.

Alvin1Zhang, 2020-06-23:
@howroyd Thanks for reporting. Would you please help if any updates for this feature request? Thanks.

Alvin1Zhang, 2020-07-14:
Thanks for reporting, will close now due to short of feedback. Feel free to reopen with more updates. Thanks.

closed 2020-07-14 by Alvin1Zhang