espressif/esp-idf #5442: CAN-Bus not working in ESP-IDF Pre-release v4.1-beta2

ISOMAX, 2020-06-12:
Hello, 

ESP-IDF Pre-release v4.1-beta2

running Example 
https://github.com/espressif/esp-idf/tree/master/examples/peripherals/can/can_network/can_network_master

with 
CONFIG_ESP32_REV_MIN=2 
or
CONFIG_ESP32_REV_MIN=3

on a V3 Chip (!!)

the Can-Bus is not working.
the Baudrate is set to 1/2 of the real Baudrate. 

with 
CONFIG_ESP32_REV_MIN=0 
or
CONFIG_ESP32_REV_MIN=1

on any Chip Rev 1 or V3 all runs fine.

ISOMAX, 2020-06-12:
https://esp32.com/viewtopic.php?f=14&t=15581&p=59670&hilit=CAN+BUS#p59670

ISOMAX, 2020-06-12:
it seems this Commit: https://github.com/espressif/esp-idf/commit/03d5742e110d2d5a8fbdf60ad9fcf894d3f98eb5 was not merged into ESP-IDF Pre-release v4.1-beta2

ISOMAX, 2020-06-12:
Just as a Information: in v4.0.1 the Behavoiur is like Expected. So the Fix https://github.com/espressif/esp-idf/commit/03d5742e110d2d5a8fbdf60ad9fcf894d3f98eb5 must be applied to V4.1 and V4.2 etc.. as well to get same Behavoiur for V3 Chips. 
Thanks, 
Franz

Dazza0, 2020-06-15:
The fix is already in v4.1 ([see link](https://github.com/espressif/esp-idf/blob/84b51781c80740fda92784dafcfc96c13b0d8b66/components/soc/esp32/include/soc/can_caps.h#L21)), and v4.2 ([see link](https://github.com/espressif/esp-idf/blob/release/v4.2/components/soc/soc/esp32/include/soc/can_caps.h))

Dazza0, 2020-06-15:
@ISOMAX found the issue. Seems like brp_div bit is automatically set to 1 on hardware power-on (will check with the digital engineers if that's the case). Therefore, the `can_ll_set_bus_timing()` is missing a clause to set the bit to 0 if brp_div isn't used.

For now, you can change the [set bus timing function](https://github.com/espressif/esp-idf/blob/cd4fa46f5372a552a67452313d246d4d09d2b39f/components/soc/src/esp32/include/hal/can_ll.h#L361) from
```c
static inline void can_ll_set_bus_timing(can_dev_t *hw, uint32_t brp, uint32_t sjw, uint32_t tseg1, uint32_t tseg2, bool triple_sampling)
{
#ifdef CAN_BRP_DIV_SUPPORTED
    if (brp > CAN_BRP_DIV_THRESH) {
        //Need to set brp_div bit
        hw->interrupt_enable_reg.brp_div = 1;
        brp /= 2;
    }
#endif
    hw->bus_timing_0_reg.brp = (brp / 2) - 1;
    hw->bus_timing_0_reg.sjw = sjw - 1;
    hw->bus_timing_1_reg.tseg1 = tseg1 - 1;
    hw->bus_timing_1_reg.tseg2 = tseg2 - 1;
    hw->bus_timing_1_reg.sam = triple_sampling;
}
```

to 

```c
static inline void can_ll_set_bus_timing(can_dev_t *hw, uint32_t brp, uint32_t sjw, uint32_t tseg1, uint32_t tseg2, bool triple_sampling)
{
#ifdef CAN_BRP_DIV_SUPPORTED
    if (brp > CAN_BRP_DIV_THRESH) {
        //Need to set brp_div bit
        hw->interrupt_enable_reg.brp_div = 1;
        brp /= 2;
    } else {
        hw->interrupt_enable_reg.brp_div = 0;
    }
#endif
    hw->bus_timing_0_reg.brp = (brp / 2) - 1;
    hw->bus_timing_0_reg.sjw = sjw - 1;
    hw->bus_timing_1_reg.tseg1 = tseg1 - 1;
    hw->bus_timing_1_reg.tseg2 = tseg2 - 1;
    hw->bus_timing_1_reg.sam = triple_sampling;
}
```
I'll push a commit to fix this master and backport it to back to v4.1

ISOMAX, 2020-06-21:
i can confirm this Fix works.

franzhoepfinger, 2020-08-27:
issue still Present in v4.1 Release.

franz-ISOMAX, 2020-09-24:
any PROGRESS on this topic ?

franz-ms-muc, 2020-10-13:
seems to be still in V4.2beta1

franz-ms-muc, 2020-10-13:
```c
/* ------------------------ Bus Timing Registers --------------------------- */

/**
 * @brief   Set bus timing
 *
 * @param hw Start address of the TWAI registers
 * @param brp Baud Rate Prescaler
 * @param sjw Synchronization Jump Width
 * @param tseg1 Timing Segment 1
 * @param tseg2 Timing Segment 2
 * @param triple_sampling Triple Sampling enable/disable
 *
 * @note Must be called in reset mode
 * @note ESP32 rev 2 or later can support a x2 brp by setting a brp_div bit,
 *       allowing the brp to go from a maximum of 128 to 256.
 */
static inline void twai_ll_set_bus_timing(twai_dev_t *hw, uint32_t brp, uint32_t sjw, uint32_t tseg1, uint32_t tseg2, bool triple_sampling)
{
#ifdef TWAI_BRP_DIV_SUPPORTED
    if (brp > TWAI_BRP_DIV_THRESH) {
        //Need to set brp_div bit
        hw->interrupt_enable_reg.brp_div = 1;
        brp /= 2;
    } else {
        hw->interrupt_enable_reg.brp_div = 0;
    }
#endif
    hw->bus_timing_0_reg.brp = (brp / 2) - 1;
    hw->bus_timing_0_reg.sjw = sjw - 1;
    hw->bus_timing_1_reg.tseg1 = tseg1 - 1;
    hw->bus_timing_1_reg.tseg2 = tseg2 - 1;
    hw->bus_timing_1_reg.sam = triple_sampling;
}
```

devUsama, 2021-02-02:
CAN Bus still not working in **ESP-IDF v4.2** release. I tried to run the examples **twai_network_master** and **twai_network_slave** on two different ESP32 Kits connecting their rx and tx lines at **TWAI_TIMING_CONFIG_25KBITS**, but no communication occurred between them.
Any Updates on this issue?

Dazza0, 2021-02-02:
@devUsama This issue should have been fixed on v4.2 as of commit fa7c5fcd4299f753151403813c62c74ac828b59c. Are you sure it's related to this issue?

devUsama, 2021-02-02:
@Dazza0 I'm getting issue when i run the twai_network master and slave examples with **TWAI_TIMING_CONFIG_25KBITS**, **TWAI_MODE_NORMAL** configurations on two ESP-WROOM-32 Kits. I simply connected the IO 21 (Tx) of master to IO 22 (Rx) of slave node and similarly slave's TX to master's Rx but there is no coomunication happening between both nodes.

If you need any further info I can provide,

Dazza0, 2021-02-02:
@devUsama You need to connect connect the master and slave via TWAI/CAN transceivers (e.g., a VP230). Refer to the diagram in the [example README](https://github.com/espressif/esp-idf/blob/release/v4.2/examples/peripherals/twai/twai_network/README.md) for more details.

devUsama, 2021-02-02:
I went through this diagram, but i thought that like **twai self mode** example, I can also communicate master and slave nodes by just connecting there tx and rx lines as voltage levels are same. 
So, is it necessary to use transceivers for this configuration?

Dazza0, 2021-02-02:
@devUsama Yes. Normal TWAI/CAN buses require a transceiver. The self test example essentially a debugging feature.

devUsama, 2021-02-02:
I see, thanks for the help!

franz-ISOMAX, 2021-02-02:
Here you see a possible Design: 

 

https://github.com/Meisterschulen-am-Ostbahnhof-Munchen/ISOBUS_Hardware

franz-ms-muc, 2021-03-02:
so, the Downloadable Version of v4.2 has not the Fix in ...

Dazza0, 2021-07-15:
@franz-ms-muc  The fix was included in the v4.2.1 release as of commit https://github.com/espressif/esp-idf/commit/fa7c5fcd4299f753151403813c62c74ac828b59c. Thank you for your patience.

closed 2021-07-15 by Dazza0