espressif/esp-idf #5443: ESP32 IDF ERASE API

vikrantdeveloper, 2020-06-12:
I was looking at the documentation of :- 
   1. esp_err_t esp_partition_read(const esp_partition_t* partition, size_t src_offset, void* dst, size_t size);
do we need to provide offset in the range of 4K byte it should be divisable by 4K ya API handles it internallly.

Thanks!

igrr, 2020-06-12:
No limitations for offset and size in `esp_partition_read`.

However the issue title is "ESP32 IDF erase API", so i'm not sure if you meant to ask about the read or about the erase function?

The limitations about offset and size for `esp_partition_erase_range' are documented in the API reference: offset and size must be 4k aligned.

vikrantdeveloper, 2020-06-15:
so the erase API must have the size and the offset in the range of 4K

projectgus, 2020-06-15:
Hi @vikrantdeveloper,

That's right, this is a limitation of NOR SPI flash chips (such as the ones used with ESP32).  4KB sectors must be erased to all-0xFF before being written with new data. The reads & writes don't need to cover entire 4KB sectors, but the erase operations do.

vikrantdeveloper, 2020-06-15:
But that creates issues suppose we have data in the range of 4096 erase bytes when we erase data through (esp_partition API) then minimum 4096 size is required. What if some data is stored in that region. The only way that remain is to read the data before write and then write the data again

projectgus, 2020-06-17:
Hi @vikrantdeveloper,

That's correct, but this is how the flash chip works in hardware - all flash chips have sector restrictions of this kind.

Suggest looking into the NVS (non volatile storage) library and also the wear levelling API - these both provide ways to store data in more conventional forms (like a FAT filesystem) inside the flash chip, by handling the sector erase/write issues for you. NVS is suitable for storing key/value data and wear levelling plus FAT allows you to have a full read/write filesystem in flash.

* https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-reference/storage/nvs_flash.html
* https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-reference/storage/wear-levelling.html

You could also look into attaching an SD card or similar to the SD interface of the ESP32. The SD card has its own embedded  flash controller that handles the flash sectors, so you don't need to manage it in the ESP-IDF firmware.

As this is not an issue with ESP-IDF itself I'm going to close this now. If you have further questions about how to store data in flash on ESP32 then please post on our discussion forum at https://esp32.com/ .

closed 2020-06-17 by projectgus