espressif/esp-idf #5444: I2S ADC glitches

marina936, 2020-06-12:
I am trying to read from ADC with I2S. I configure I2S frequency to 120 KHz, 16 bit and only channel right, like this:

static const i2s_config_t i2s_config = {
.mode = I2S_MODE_MASTER | I2S_MODE_RX | I2S_MODE_ADC_BUILT_IN,
.sample_rate = 120000,
.bits_per_sample = I2S_BITS_PER_SAMPLE_16BIT,
.channel_format = I2S_CHANNEL_FMT_ONLY_RIGHT,
.communication_format = 0,
.intr_alloc_flags = 0, // default interrupt priority
.dma_buf_count = 3,
.dma_buf_len = NUM_SAMPLES,
.use_apll = true,  
.fixed_mclk = 120000
};
I add continuos square signals of 1KHz and amplitude 1V to the ADC.

This is what I receive reading ADC with I2S.

[image]

In the signals I receive I show some kind of peaks (transitories) in the edge of the signals.

Can anyone explain why this happens? And how I can delete these peaks?

koobest, 2020-06-13:
Hi, @marina936
The relationship between data storage order and sampling is：

[image]

 [ sample1, sample0, sample3, sample2 ...];

marina936, 2020-06-15:
I don't understand the relation of that with the strange glitches in every signal

koobest, 2020-06-15:
Hi,@marina936
This is not sample glitches. Can you swap the sample data by software and then draw the waveform?
```
uint16_t *sample_buf;// the buffer userd to store the sample data.
...
...
uint16_t tmp;
...

for (int i = 0; i < sample_cnt; i+=2) {
    tmp = sample_buf[i];
    sample_buf[i] = sample_buf[i+1];
    sample_buf[i+1] = tmp;
}

draw(sample_buf, sample_cnt);
```

thanks

marina936, 2020-06-15:
Why is necessary to swap sample data?

koobest, 2020-06-15:
See the reply here: https://github.com/espressif/esp-idf/issues/5444#issuecomment-643563107

closed 2020-06-15 by marina936