espressif/esp-idf #5445: Eclipse: unresolved references all over the place

SolidStateLEDLighting, 2020-06-12:
Eclipse has a long track record (several years worth) of problems with unresolved references....  I just installed the Windows 2020-03 version -- all new install of IDF and tools.... I installed the newest ESP32 plug-in.

There must be an article somewhere dealing with this issue that is current?

My code still compiles and runs --- I just have compiler warnings all over the place.

e.g. Type 'xSemaphoreHandle' could not be resolved

e.g. Function 'ESP_ERROR_CHECK' could not be resolved

e.g. Type 'nvs_handle_t' could not be resolved

e.g. Type 'esp_timer_handle_t' could not be resolved

My entire project is currently built in another tool -- so I know the code was solid before I imported it.   And the project still builds but has all these dumb complaints.

Can someone point me in the right direction, please?

kolipakakondal, 2020-06-16:
Hi, We can use this https://github.com/espressif/idf-eclipse-plugin/issues/100#issuecomment-644537779 for tracing the issue.

Alvin1Zhang, 2020-06-23:
Thanks for reporting, since the issue has been tracked in https://github.com/espressif/idf-eclipse-plugin/issues/100#issuecomment-644537779, will close this ticket. Thanks.

closed 2020-06-23 by Alvin1Zhang