espressif/esp-idf #5446: Move this to documentation - Event typo

hassance3, 2020-06-12:
I didn't find where to create a request to fix a event name typo, so please remove this or moved to the right place.

**Document link:**
[docs.espressif.com - esp_err_t esp_ble_gatts_create_service()](https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-reference/bluetooth/esp_gatts.html#_CPPv428esp_ble_gatts_create_service13esp_gatt_if_tP18esp_gatt_srvc_id_t8uint16_t)

**Typo:**
`Create a service. When service creation is done, a callback event BTA_GATTS_CREATE_SRVC_EVT is called to...`

**Fix:**
`Create a service. When service creation is done, a callback event ESP_GATTS_CREATE_EVT is called to...`

Rhank you.

Alvin1Zhang, 2020-06-15:
@hassance3 Thanks for reporting and pointing out, we will look into.

WCCWCC, 2020-06-23:
@hassance3 
We will fix this problem as soon as possible.

Alvin1Zhang, 2020-07-30:
Thanks for reporting and sorry for slow turnaround, the fix is available at https://github.com/espressif/esp-idf/commit/f73cd1d4f1128c8a66b9f386bde409d08f2ac8fa. Feel free to reopen if the issue still happens.

closed 2020-07-30 by Alvin1Zhang