espressif/esp-idf #5447: fatal error: nimble/connection.h: No such file or directory

tagtraeumer, 2020-06-12:
## Environment

- Development Kit:      ESP32-DevKitC
- Kit version: v4
- Module or chip used:  ESP32-WROOM-32
- IDF version: v4.0.1
- Build System: CMake and/or idf.py
- Compiler version: xtensa-esp32-elf-gcc (crosstool-NG esp-2019r2) 8.2.0
- Operating System:     Windows 10
- (Windows only) environment type: MSYS2 mingw32 for CMake, tested on others with idf.py
- Using an IDE?: Yes CLion
- Power Supply:    USB

## Problem Description

After activating Bluetooth in menuconfig choosing either NumBLE or Bluedroid my software doesn't compile anymore. I haven't added any BT specific code yet. Prior to activating BT in menuconfig, the program compiles successfully as expected

### Expected Behavior

After activating BT in menuconfig the software should compile successfully

### Actual Behavior

Build Errors.
**NimBLE**:
xxx\esp-idf\components\bt\host\nimble\nimble\nimble\host\src\ble_sm_alg.c:28:10: fatal error: nimble/connection.h: No such file or directory
 #include "nimble/connection.h"
          ^~~~~~~~~~~~~~~~~~~~~
compilation terminated.

**Bluedroid**:

xxx/esp-idf/components/bt/host/bluedroid/stack/btm/btm_ble_addr.c: In function 'btm_ble_init_pseudo_addr':
xxx/esp-idf/components/bt/host/bluedroid/stack/btm/btm_ble_addr.c:299:25: error: 'tBTM_SEC_DEV_REC' {aka 'struct <anonymous>'} has no member named 'connection'
     if (memcmp(p_dev_rec->connection.pseudo_addr, dummy_bda, BD_ADDR_LEN) == 0) {
                         ^~
xxx/esp-idf/components/bt/host/bluedroid/stack/btm/btm_ble_addr.c:300:25: error: 'tBTM_SEC_DEV_REC' {aka 'struct <anonymous>'} has no member named 'connection'
         memcpy(p_dev_rec->connection.pseudo_addr, new_pseudo_addr, BD_ADDR_LEN);
                         ^~
xxx/esp-idf/components/bt/host/bluedroid/stack/btm/btm_ble_addr.c: In function 'btm_ble_addr_resolvable':
xxx/esp-idf/components/bt/host/bluedroid/stack/btm/btm_ble_addr.c:327:23: error: 'tBTM_SEC_DEV_REC' {aka 'struct <anonymous>'} has no member named 'connection'
             (p_dev_rec->connection.key_type & BTM_LE_KEY_PID)) {
                       ^~
xxx/esp-idf/components/bt/host/bluedroid/stack/btm/btm_ble_addr.c:335:30: error: 'tBTM_SEC_DEV_REC' {aka 'struct <anonymous>'} has no member named 'connection'
         SMP_Encrypt(p_dev_rec->connection.keys.irk, BT_OCTET16_LEN,
                              ^~
xxx/esp-idf/components/bt/host/bluedroid/stack/btm/btm_ble_addr.c: In function 'btm_ble_match_random_bda':
xxx/esp-idf/components/bt/host/bluedroid/stack/btm/btm_ble_addr.c:384:27: error: 'tBTM_SEC_DEV_REC' {aka 'struct <anonymous>'} has no member named 'connection'
                 (p_dev_rec->connection.key_type & BTM_LE_KEY_PID)) {
                           ^~
xxx/esp-idf/components/bt/host/bluedroid/stack/btm/btm_ble_addr.c:386:34: error: 'tBTM_SEC_DEV_REC' {aka 'struct <anonymous>'} has no member named 'connection'
             SMP_Encrypt(p_dev_rec->connection.keys.irk, BT_OCTET16_LEN,
                                  ^~
xxx/esp-idf/components/bt/host/bluedroid/stack/btm/btm_ble_addr.c: In function 'btm_find_dev_by_identity_addr':
xxx/esp-idf/components/bt/host/bluedroid/stack/btm/btm_ble_addr.c:457:33: error: 'tBTM_SEC_DEV_REC' {aka 'struct <anonymous>'} has no member named 'connection'
                 memcmp(p_dev_rec->connection.static_addr, bd_addr, BD_ADDR_LEN) == 0) {
                                 ^~
xxx/esp-idf/components/bt/host/bluedroid/stack/btm/btm_ble_addr.c:458:27: error: 'tBTM_SEC_DEV_REC' {aka 'struct <anonymous>'} has no member named 'connection'
             if ((p_dev_rec->connection.static_addr_type & (~BLE_ADDR_TYPE_ID_BIT)) !=
                           ^~
In file included from xxx/esp-idf/components/bt/host/bluedroid/stack/include/stack/bt_types.h:24,
                 from xxx/esp-idf/components/bt/host/bluedroid/stack/btm/btm_ble_addr.c:27:
xxx/esp-idf/components/bt/host/bluedroid/stack/btm/btm_ble_addr.c:461:54: error: 'tBTM_SEC_DEV_REC' {aka 'struct <anonymous>'} has no member named 'connection'
                                   __func__, p_dev_rec->connection.static_addr_type, addr_type);
                                                      ^~
xxx/esp-idf/components/bt/common/include/bt_common.h:85:128: note: in definition of macro 'BT_PRINT_W'
 #define BT_PRINT_W(tag, format, ...)   {esp_log_write(ESP_LOG_WARN,    tag, LOG_FORMAT(W, format), esp_log_timestamp(), tag, ##__VA_ARGS__); }
                                                                                                                                ^~~~~~~~~~~
xxx/esp-idf/components/bt/host/bluedroid/stack/btm/btm_ble_addr.c:460:17: note: in expansion of macro 'BTM_TRACE_WARNING'
                 BTM_TRACE_WARNING("%s find pseudo->random match with diff addr type: %d vs %d",
                 ^~~~~~~~~~~~~~~~~
xxx/esp-idf/components/bt/host/bluedroid/stack/btm/btm_ble_addr.c: In function 'btm_identity_addr_to_random_pseudo':
xxx/esp-idf/components/bt/host/bluedroid/stack/btm/btm_ble_addr.c:496:38: error: 'tBTM_SEC_DEV_REC' {aka 'struct <anonymous>'} has no member named 'connection'
             memcpy(bd_addr, p_dev_rec->connection.pseudo_addr, BD_ADDR_LEN);
                                      ^~

... and many more ...

### Steps to reproduce
Not sure anymore what of these are default values for menuconfig, I'm writing down the ones that I think I might have changed and/or are critical to this problem:

**General**:
In menuconfig -> Component config -> Bluetooth 
- Bluetooth: Active

-> Bluetooth controller:
- Bluetooth controller mode: 'BLE Only"
- CPU core: Core 0
- HCI mode: VHCI

**NimBLE**:
Bluetooth Host: 'NimBLE - BLE only'
-> NimBLE Options
- Memory allocation strategy: Internal memory
- CPU Core: Core 0

**Bluedroid**:
Bluetooth Host: 'Bluedroid - Dual-Mode'
-> Bluedroid Options
- Classic Bluetooth: Inactive
- Bluetooth Low Energy: Active
- Include GATT server module: Active
- Include GATT client module: Inactive
- Include BLE security module: Active
- Slave enable connection parameters update during pairing: Inactive
etc ...

Alvin1Zhang, 2020-06-15:
@tagtraeumer Thanks for reporting, we will look into.

prasad-alatkar, 2020-06-15:
Hi @tagtraeumer The issue seems to be environment specific as keyword "ble" is getting replaced with "connection" during compilation. Can you please try to reproduce the issue on plain MSYS2 command line?

tagtraeumer, 2020-06-18:
Hi, I'm sorry I didn't answer for a while. Currently were having some problems within the family and I couldn't find time to take care of that matter. I'll come back to you as soon as I can .

tagtraeumer, 2020-08-04:
Hi thank you for your patience, it has been quite some difficult weeks. 

@prasad-alatkar I wasn't really able to build the project on the commandline. maybe my description in my initial post was not quite specific, because i mentioned MSYS2, but i mean that i configured mingw64 as a toolchain for CLion and this toolchain is used for CMake. I installed the commandline but couldn't do any relevant building, since my project is based on cmake (I use idf.py) and all documentation I found describe MSYS2 in combination with make. (see pictures attached) 
maybe this could be a reason for this error?

[image]

[image]

Some background info:
Generally it worked for me until I wanted to activate BT/BLE. My objective was changing from platformio to plain esp-idf in the hope to save some memory and have more control over the components and settings. BLE using platformio worked.

mws-rmain, 2021-07-17:
Possibly the problem is due to an incomplete patch of the build system.
I'm experiencing the same issue ('esp_bt.h not found') when attempting a build.
The build system has an argument ('-Ixxx') to look for includes here:

`/home/xxxx/esp/esp-idf/components/bt/include
`
however, the 'esp_bt.h' file is located two layers further down, under one of the directories below (based on the target silicon):

```
/home/xxxx/esp/esp-idf/components/bt/include/esp32/include
/home/xxxx/esp/esp-idf/components/bt/include/esp32c3/include
/home/xxxx/esp/esp-idf/components/bt/include/esp32s3/include
```
In my case, the target is esp32.
When the esp-idf components are compiling, the 'build.ninja' file contains the correct include path (first option shown above).
later in the 'build.ninja' file (when my code is compiling), the include path is the truncated path (not patched for the target silicon).

Kiznoh, 2021-08-03:
Hi, I've been unable to reproduce this issue using IDF 4.3 and CLion 2021.1.3.

These are the settings used:
The selected C and C++ compilers are located inside the `.espressif/tools` folder.

[image]

The only required CMake settings are the environment variables `IDF_PATH`, `IDF_TARGET`, and `PATH`.

[image]

`PATH` is set to IDF's python env, compilers' bin folder, and git.

[image]

Please note, building IDF projects with MSYS or CLion are not officially supported by ESP-IDF.

closed 2021-08-03 by dobairoland