espressif/esp-idf #5448: Unable to Connect to MQTT Broker via TLS

mntolia, 2020-06-12:
## Environment

Development Kit: ESP32-Wrover-Kit
Kit version (for WroverKit/PicoKit/DevKitC): v4
Module or chip used: ESP32-WROOM-32
IDF version : 4
Build System: CMake
Operating System: Linux
Using an IDE?: PlatformIO on Visual Studio Code
Power Supply: USB

I am using Arduino as an ESP-IDF Component

## Problem Description

The ESP32 is not connecting to my MQTT broker via TLS. This is the error message I am getting:
`(569328) MQTT_CLIENT: mqtt_message_receive: transport_read() error: errno=128␛[0m`

I am not sure what error no 128 means. I checked the error codes references but it doesn't have number 128. 

### Steps to reproduce

1. I generated a self-signed certificate with the Common Name as the hostname. 
2. I saved the certificate authority file as a const char * in the esp32 code.

### Code to reproduce this issue
Note: I have hidden the key content as ...

```cpp
   const char *cert_content =  "-----BEGIN CERTIFICATE-----\n"
                                               "....\n"
                                               "-----END CERTIFICATE-----";
    Serial.println("Connecting to MQTT Broker");
    esp_mqtt_client_config_t mqtt_config = {0};
    mqtt_config.transport = MQTT_TRANSPORT_OVER_SSL;
    mqtt_config.uri = "mqtt://broker.aliv.com.sg";
    mqtt_config.username = "facescanner";
    mqtt_config.password = "=3jSbtjRiE8R";
    mqtt_config.port = 8883;
    mqtt_config.client_id = "esp32";
    mqtt_config.cert_pem = cert_content;
    mqtt_config.event_handle = mqtt_event_handler_cb;
    Serial.println("Initializing Client");
    esp_mqtt_client_handle_t client = esp_mqtt_client_init(&mqtt_config);
    Serial.println("Registering Callback");
    Serial.println("Connecting");
    esp_mqtt_client_start(client);
```
## Debug Logs

```
␛[0;32mI (776948) MQTT_CLIENT: Sending MQTT CONNECT message, type: 1, id: 0000␛[0m
␛[0;31mE (776988) MQTT_CLIENT: mqtt_message_receive: transport_read() error: errno=128␛[0m
␛[0;31mE (776988) MQTT_CLIENT: esp_mqtt_connect: mqtt_message_receive() returned -1␛[0m
␛[0;32mI (776988) MQTT_CLIENT: Error MQTT Connected␛[0m
␛[0;32mI (776998) MQTT_EXAMPLE: MQTT_EVENT_DISCONNECTED␛[0m
```

## Other items if possible

https://gist.github.com/mntolia/2cf2576b3c9ae3c6bd82b189c82dcd54

ESP-Marius, 2020-06-15:
Hi @mntolia 

I think your issue here is probably that your URI is overwriting your SSL settings. The URI will always take precedence over the other configs (domain, port, transport), and since your says `mqtt://` you will end up with a plain TCP connection. You can try with: 

`mqtt_config.uri = "mqtts://broker_url:8883";`

mntolia, 2020-06-15:
> Hi @mntolia
>
> I think your issue here is probably that your URI is overwriting your SSL settings. The URI will always take precedence over the other configs (domain, port, transport), and since your [...]

It worked. Thanks a lot. I don't know how I missed that.

closed 2020-06-15 by mntolia