espressif/esp-idf #5450: Baud rates <460800 partially gives garbage

riegaz, 2020-06-13:
## Environment

- Development Kit:      ESP32-WROOM-32D
- IDF version (run ``git describe --tags`` to find it): v4.2
- Build System:         idf.py
- Compiler version (run ``xtensa-esp32-elf-gcc --version`` to find it): 1.22.0-80-g6c4433a
- Operating System:     Windows
- (Windows only) environment type: MSYS2 mingw32
- Using an IDE?: No
- Power Supply:         USB

## Problem Description
Changing the UART console baud rate to values >460800 leads to the following output. The garbage after the readable output are random ESP_LOGE or printf messages. However, for some reason they are not readable. Baud rate is set to 921600. Any baud rate >460800 leads to this error.

> <\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><27>[0;32mI (181) cpu_start: Pro cpu up.<27>[0m<\r><\n>
> <27>[0;32mI (181) cpu_start: Application information:<27>[0m<\r><\n>
> <27>[0;32mI (182) cpu_start: App version: 1<27>[0m<\r><\n>
> <27>[0;32mI (182) cpu_start: Compile time: Jun 13 2020 00:52:34<27>[0m<\r><\n>
> <27>[0;32mI (182) cpu_start: ELF file SHA256: 002ffd6307be9048...<27>[0m<\r><\n>
> <27>[0;32mI (182) cpu_start: ESP-IDF: v4.2-dev-1660-g7d7521367-dirty<27>[0m<\r><\n>
> <27>[0;32mI (183) cpu_start: Starting app cpu, entry point is 0x40081804<27>[0m<\r><\n>
> <27>[0;32mI (0) cpu_start: App cpu up.<27>[0m<\r><\n>
> <27>[0;32mI (183) heap_init: Initializing. RAM available for dynamic allocation:<27>[0m<\r><\n>
> <27>[0;32mI (184) heap_init: At 3FFAFF10 len 000000F0 (0 KiB): DRAM<27>[0m<\r><\n>
> <27>[0;32mI (184) heap_init: At 3FFB6388 len 00001C78 (7 KiB): DRAM<27>[0m<\r><\n>
> <27>[0;32mI (184) heap_init: At 3FFB9A20 len 00004108 (16 KiB): DRAM<27>[0m<\r><\n>
> <27>[0;32mI (185) heap_init: At 3FFC4468 len 0001BB98 (110 KiB): DRAM<27>[0m<\r><\n>
> <27>[0;32mI (185) heap_init: At 3FFE0440 len 00003AE0 (14 KiB): D/IRAM<27>[0m<\r><\n>
> <27>[0;32mI (185) heap_init: At 3FFE4350 len 0001BCB0 (111 KiB): D/IRAM<27>[0m<\r><\n>
> <27>[0;32mI (186) heap_init: At 4009516C len 0000AE94 (43 KiB): IRAM<27>[0m<\r><\n>
> <27>[0;32mI (186) cpu_start: Pro cpu start user code<27>[0m<\r><\n>
> ž`ž<6>˜~<30><24>xæž†˜<\0><24>†<6>˜ø˜æ˜˜<24>˜ø<6>fžþfxàž<6>æø<6>ž˜<24><24>xf<6>~fæà<6>~f<6>x<\0>˜à<6>ž†ž<\0>†~<\0>˜xf†žf†øfž<30>æ<30>žf˜ž†<6>ž`ž<6>˜~<30><24>xæž†˜<\0><24>†<6>˜ø˜æ˜˜<24>˜ø<6>fžþfxàž<6>æø<6>ž˜<24><24>xàž<6>æø<6>ž<\0>˜˜æž˜<24>˜àfžþžž`ž<6>˜ž†<6>ž`ž<6>˜~<30><24>xæž†˜<\0><24>†<6>˜ø˜æ˜˜<24>˜à<6>f~þfø<6>~<6>†ø<6>~˜<24>˜˜<6>~<6>†ø<6>~†žøž~<6>˜øæà<6>žf<6>xf<6>žf†ø<6>˜žøž<\0><24>f˜æ˜<\0>˜€<6>ff††ž`ž<6>˜ž†<6>ž`ž<6>˜~<30><24>xæž†˜<\0><24>†<\0>˜˜<24>˜à<6>f~þfø<6>~<6>†ø<6>~˜<24>˜˜<6>~<6>†ø<6>~†žøž~<6>˜øæà<6>žf<6>xf<6>žf†ø<6>˜žøž<\0>˜<\0><6>ff˜€<6>ff††ž`ž<6>˜žf€˜€ž`ž<6>˜~<30><24>xæž†˜<\0><24>†<24>˜ø˜<30>††<\0><24>€<6>f`f˜þf<24>†˜†˜`f~<\0><24>€<6>f<\0>˜àæžøž`†øæžàžàžf†ø<6>˜àæžæž<\0>fžàžf<6><24>~f†øæøfžþžøž<\0>˜žf˜†fx<6>˜æ˜x<\0>˜~æ<30>žf˜ž†<6>ž`ž<6>˜~<30><24>xæž†˜<\0><24>†<24>˜ø˜<30>††<\0>˜øfæø<6>~ffžþf€<6>fž˜<24><24>€fæøfx<\0>˜<30>fæ€<6>˜€<6>x`†øfx<30>æø<6>˜˜æø<6><24>žþž`<6>˜øfx`æ<30>žf˜ž†<6>ž`ž<6>˜~<30><24>xæž†˜<\0><24>†<24>˜ø˜<30>††<\0>˜øfæø<6>~ffžþf€<6>fž˜<24><24>øfx<6><6>x<\0><24>xf†x<6>f~àž`<6><24>àfæøfx<\0>˜<30>fæ€<6>˜€<6>x`†øfx<30>æø<6><24>x<24>æžæž<\0>˜<6>†`ff˜f`ž`ž<6>˜ž†<6>ž`ž<6>˜~<30><24>xæž†˜<\0><24>†<30><24>˜<30>††<\0><24><6>ž††~<\0><24><6>ž†æf~f†øæøfžþžøž˜<24><24><30><6><24>˜à<24>†<\0>˜àf˜€<6>˜xf†xà†<\0>˜<30>f†ø<6>˜à<24><24><24>x<\0><24>x<\0><24>†<\0>˜à<24>x˜<24><30>f<24>~<\0><24><30>à†<\0><24><24>†<\0><24>˜<30>žf˜žf€˜€ž`ž<6>˜~<30><24>xæž†˜<\0><24>†f˜æ˜à˜˜<24><24>fžþf†æø<6>æx<24><24>~<\0><24>`<24>fx<6>f~f†ž<30>f<6>˜øæ~†ž`æà<6>ž†žøž~<6>˜àæžøžxfž~†~<\0>˜€<6>ffæf<30>f<6>f~<\0>˜à<24><30><\0><24>†<\0>˜<\0><6>f€æf<30>f<6>f~<\0>˜à<24><30><\0><24>†<\0>˜<\0><6>f€æf<30>f˜ø˜˜<24>˜à<24><30><\0><24>†<\0><24>˜†ž~<6>ž`<6>˜ø<6>žffx<\0>†~<\0><24>`†˜<30>f<\0>†€<6>˜f<6>`ž`ž<6>˜ž†<6>ž`ž<6>˜~<30>˜xæž~f<24>†f˜æ˜à˜˜<24>˜†<6>fžøž<30>fx<24>†àæž€ž<24><24>~<\0><24>x<24>fxf<6>˜žfxæžþž<24>f†~<\0>˜à˜<30>~<24>˜æ†àæ<30>žf˜ž†<6><6>æ˜<6>`˜€~f<6>f<24>†øæž<30>fx`f~<24>fx<\0>˜˜†ž†ž`fž<6><6>~f<6>x˜<24>˜€<6>xxfx<24><6>~†ž<30>fxž<24><24>x†ž<30>æàæfžæž`fxæ<24>x<\0>˜€<6>xxæfà<6>ž<6>†žøžf<6>žþfžf<6>f<24><24>˜ž~†žþf€<6>x`†øæf~†<6>fxæ<24><24>˜€<6>xxæfx†žàž`fx<24>æfæžàž†ž<30>ff<24><24>˜€<6>xxæf˜<6>~x<6>žþfžfžøžæ<24><24>˜€<6>xxæf˜<6>~x<6>žþfžf<6>f<24><24><6>~f<6>f<24>†øæž<30>fx`f~<24>fx<\0>˜˜†ž†ž`fž<6><6>~f<6>x˜<24>˜ø<6>~þž<\0><6>˜€<6>xxfx<24><6>~†ž<30>fžøž~††˜€f†ø†øæž<24><6>˜††ž<6>†à<6>ž†žøž~<6>˜€<6>xxfx<24><6>~†ž<30>fxæžf†ž`æ~<\0><24>øæàf˜x<\0><24><6>

### Expected Behavior

The output should be alright even after the cpu_start/heap_init messages.

### Actual Behavior
The output after the cpu_start is garbage.

### Steps to reproduce

1. set the UART console baud rate to >460800
2. add some printfs
3. see the uart output

Alvin1Zhang, 2020-06-15:
Thanks for reporting, we will look into.

riegaz, 2020-06-16:
I found the cause. Setting this one leads to the error.

Steps to reproduce:
Create the hello_world example.
Set the idf.py and UART baud rate to 2000000.
Set the option below in the menuconfig.
See the garbage output.

`#
# Power Management
#
CONFIG_PM_ENABLE=y
# CONFIG_PM_DFS_INIT_AUTO is not set
# CONFIG_PM_PROFILING is not set
# CONFIG_PM_TRACE is not set
# end of Power Management
`

closed 2024-12-26 by Alvin1Zhang