espressif/esp-idf #5451: cannot flash or erase_flash in v4.0.1

korstiaanS, 2020-06-14:
Hi,

I just started to migrate from 3.3 to 4.0.1.
I followed the installation guide and I want to flash the ESP32 but it won't work.
Using the same hardware I can flash, erase_flash without any problem in version 3.3.
I only can use `idf.py -p /dev/cu.usbserial-14Y0DC69 monitor`

The erase_flash option gives the following output:

```
idf.py -p /dev/cu.usbserial-14Y0DC69 erase_flash
Checking Python dependencies...
Python requirements from /Users/skor/esp/esp-idf/requirements.txt are satisfied.
Executing action: erase_flash
Running esptool.py in directory /Users/skor/esp/hello_world/build
Executing "/Users/skor/.espressif/python_env/idf4.0_py2.7_env/bin/python /Users/skor/esp/esp-idf/components/esptool_py/esptool/esptool.py -p /dev/cu.usbserial-14Y0DC69 -b 460800 --before default_reset --after hard_reset erase_flash"...
esptool.py v2.8
Serial port /dev/cu.usbserial-14Y0DC69
Connecting........_
Detecting chip type... ESP32

A fatal error occurred: Failed to read register address 6001a00c (result was 01050000)
esptool.py failed with exit code 2

```
Any idea why it works for years now with make and NOT with idf.py?

projectgus, 2020-06-15:
Hi @korstiaanS,

Thanks for the detailed report.

The failure is actually coming from `esptool.py` not `idf.py`. Both the GNU Make build system and `idf.py` execute esptool to perform the flashing steps.

ESP-IDF v3.3 used esptool v2.8-dev, and ESP-IDF v3.3.1 and v4.0.1 both use esptool v2.8. Do you happen to know if the same esptool version was being used successfully before (esptool will print the version to the console each time it runs.)

The error message looks like a error occurred while the serial connection was reading the information from the chip. Usually I'd suggest this was a hardware problem (signal integrity or power stability), or a driver problem, rather than a software problem.

Is there any difference in the serial port device path or the baud rate configured with ESP-IDF v3.3 vs v4.0? (You can set the baud rate by passing the `-b RATE` option to `idf.py`.)

korstiaanS, 2020-06-15:
Hi,

The v3.3 is on my MacBook Pro and the v4.0.1 is running on a virtual machine on Parallels.
Maybe that's the reason why it won't work properly?

projectgus, 2020-06-17:
Hi @korstiaanS ,

Is MacOS the OS in both cases?

Are you passing through the serial port to the VM guest as a USB device or a serial device? If serial, then suggest trying USB instead - but it looks like it's already USB.

Otherwise, would suggest trying the things mentioned in the previous comment.

projectgus, 2020-06-17:
One other though, if it's a different OS version in the Parallels VM then make sure the USB to serial driver is updated to the latest version. Some issues have been reported with MacOS Catalina drivers in particular.

korstiaanS, 2020-06-17:
Hi,

Yes same macOS in both cases.
I'm passing through as a USB device.
My physical mac use pytool 2.8-dev and the virtual one v2.8
I installed FTDI driver 2.4.2 but I'm not sure if the OS is using this driver since it is coming in via parallels?
Changing baud rates won't help either.

projectgus, 2020-06-17:
Can you try updating the FTDI driver in the VM guest as well?

korstiaanS, 2020-06-17:
I did it inside the VM.
I installed the VCP driver. Is that the good one.
[https://www.ftdichip.com/Drivers/VCP/MacOSX/FTDIUSBSerialDriver_v2_4_2.dmg](https://www.ftdichip.com/Drivers/VCP/MacOSX/FTDIUSBSerialDriver_v2_4_2.dmg)
I'm just wondering if macOS really uses this driver?

korstiaanS, 2020-06-17:
I also see that my fysical MBP (the working one) has no FTDI driver installed/running and that my VM has now the FTDI software installed BUT it shows 'No' in the loaded column.

korstiaanS, 2020-06-17:
It works:

I found a driver v2.4.4 on the net and if I install this driver inside my VM it works. 
This is the post:
[http://www.ftdicommunity.com/index.php?topic=505.0](http://www.ftdicommunity.com/index.php?topic=505.0)

So now it works on a VM in parallels!

projectgus, 2020-06-18:
Hi @korstiaanS,

Thanks for providing the link and glad you found a configuration that works in the VM.

If the issue is now resolved, can this issue be closed?

korstiaanS, 2020-06-18:
Hi,

Thank you!
I will close the case.

closed 2020-06-18 by korstiaanS