espressif/esp-idf #5453: Possible off by one error in esp_ble_gattc_get_char_by_uuid?

howroyd, 2020-06-15:
I think there may be an off by one error in the function `esp_ble_gattc_get_char_by_uuid`, or the internals of what it calls.

After searching for and finding a Heartrate Service (using NRF on my phone), `esp_ble_gattc_get_attr_count` correctly reports **three** characteristics in this service:

1. 0x2A37
2. 0x2A38
3. 0x2A39

I print these out to the log, just to make sure of the order, which is correct, meaning 0x2A37 (the actual heartrate value, u16) is at the first index (i.e. 0) in the array`char_elem_result`.

Then using  `esp_ble_gattc_get_char_by_uuid` I search for `remote_filter_char_uuid` which is defined as:
```
const esp_bt_uuid_t Pixel::remote_filter_char_uuid = 
{
    .len = ESP_UUID_LEN_16,
    .uuid = {.uuid16 = 0x2A37,},
};
```

Printing out the value of the last argument to the function  `esp_ble_gattc_get_char_by_uuid` shows pre-call the value is three (correct, the  number of characteristics in the service).  Then post-call I would expect this to be zero.  The documentation states; `it will output the number of characteristic has been found in the gattc cache with the given service.`

Is this intentional (we're now counting from one rather than zero), have I done something wrong or is this a bug?

The relevant debug:
```
I (00:00:45.158) PIXEL: Device open event [if:6, conn:0]: ESP_OK
D (00:00:45.743) PIXEL: Callback; event:46
I (00:00:45.744) PIXEL: Discover service complete conn_id 0
D (00:00:45.758) PIXEL: Callback; event:18
D (00:00:45.758) PIXEL: Config MTU set: 500
D (00:00:46.009) PIXEL: Callback; event:7
I (00:00:46.010) PIXEL: SEARCH RES: conn_id = 0 is primary service 1
I (00:00:46.010) PIXEL: start handle 46 end handle 53 current handle value 46
I (00:00:46.017) PIXEL: service found
I (00:00:46.022) PIXEL: UUID16: 180d
D (00:00:46.026) PIXEL: Callback; event:6
I (00:00:46.030) PIXEL: Get service information from remote device
D (00:00:47.009) PIXEL: Found 3 characteristics for service 0x180D:
D (00:00:47.010) PIXEL: 0. UUID: 0x2A37
D (00:00:47.011) PIXEL: 1. UUID: 0x2A38
D (00:00:47.013) PIXEL: 2. UUID: 0x2A39
D (00:00:47.017) PIXEL: Before: 3
D (00:00:47.020) PIXEL: After: 1
D (00:00:47.023) PIXEL: Heart Rate UUID: 0x2A37, handle:46-53

```

The relevant code, sorry can't share the whole lot, and yes this is cpp:
```
		case states_e::SEARCH_COMPLETE:
        {
            uint16_t char_elem_result_len = 0;

            esp_gatt_status_t status = esp_ble_gattc_get_attr_count(services.list[SERVICE_ID_HR].server->gatt_if,
                                                                    services.list[SERVICE_ID_HR].server->conn_id,
                                                                    ESP_GATT_DB_CHARACTERISTIC,
                                                                    services.list[SERVICE_ID_HR].start_handle,
                                                                    services.list[SERVICE_ID_HR].end_handle,
                                                                    0,
                                                                    &char_elem_result_len);

            if (ESP_GATT_OK != status)
			{
                ESP_LOGE(LOG_TAG, "esp_ble_gattc_get_attr_count error");
            }

            if (char_elem_result_len > 0)
			{
                esp_gattc_char_elem_t* char_elem_result = nullptr;

                char_elem_result = new esp_gattc_char_elem_t[char_elem_result_len];
                ESP_LOGD(LOG_TAG, "Found %u characteristics for service %s:", char_elem_result_len, uuid_to_str(uuid_heartrate_svc));

                esp_ble_gattc_get_all_char( services.list[SERVICE_ID_HR].server->gatt_if,
                                            services.list[SERVICE_ID_HR].server->conn_id,
                                            services.list[SERVICE_ID_HR].start_handle,
                                            services.list[SERVICE_ID_HR].end_handle,
                                            char_elem_result,
                                            &char_elem_result_len, 0);

                for (size_t i = 0 ; i < char_elem_result_len ; ++i)
                {
                    ESP_LOGD(LOG_TAG, "%u. UUID: %s", i, uuid_to_str(char_elem_result[i].uuid));
                }

                if (char_elem_result)
				{
                    uint16_t char_elem_result_idx = char_elem_result_len;
                    ESP_LOGD(LOG_TAG, "Before: %u", char_elem_result_idx);
                    status = esp_ble_gattc_get_char_by_uuid(services.list[SERVICE_ID_HR].server->gatt_if,
                                                            services.list[SERVICE_ID_HR].server->conn_id,
                                                            services.list[SERVICE_ID_HR].start_handle,
                                                            services.list[SERVICE_ID_HR].end_handle,
                                                            remote_filter_char_uuid,
                                                            char_elem_result,
                                                            &char_elem_result_idx);
                    ESP_LOGD(LOG_TAG, "After: %u", char_elem_result_idx);

                    if (status != ESP_GATT_OK)
					{
                        ESP_LOGE(LOG_TAG, "esp_ble_gattc_get_char_by_uuid error");
                    }

					ESP_LOGD(LOG_TAG, "Heart Rate UUID: %s, handle:%u-%u", uuid_to_str(char_elem_result[0].uuid), 
                                                                            services.list[SERVICE_ID_HR].start_handle, 
                                                                            services.list[SERVICE_ID_HR].end_handle);

                    if (char_elem_result_len > 0 && (char_elem_result[0].properties & ESP_GATT_CHAR_PROP_BIT_NOTIFY))
					{
                        char_handle = char_elem_result[0].char_handle;
                        if (ESP_OK == esp_ble_gattc_register_for_notify(services.list[SERVICE_ID_HR].server->gatt_if, 
                                                                        services.list[SERVICE_ID_HR].server->bda, 
                                                                        char_elem_result[0].char_handle))
                        {
                            state = states_e::REGISTERING_FOR_NOTIFICATION;
                        }
                        else
                        {
                            ESP_LOGE(LOG_TAG, "esp_ble_gattc_register_for_notify error");
                        }
                    }

                    delete[] char_elem_result;
                }
				else
				{
                    ESP_LOGE(LOG_TAG, "gattc no mem");
                }
            }
			else
			{
                ESP_LOGE(LOG_TAG, "no char found");
            }
            break;
        }
```

WCCWCC, 2020-06-18:
Hi， @howroyd 

* Maybe I don’t understand it, can you describe it more clearly?

 * esp_ble_gattc_get_all_char->uint16 * count, This API is to find all characteristics under the specified server.

 * esp_ble_gattc_get_char_by_uuid -> uint16 * count，This API is to find the specified uuid of the specified server.

* API return parameter count I did not see the exception

howroyd, 2020-06-18:
Ok cool so this seems deliberate, no problem.

Typically functions would take an array `index` (starting at zero) or `length`, I think I got confused by the word `count`.

I think in this case it is confusing because we are passing an array of structs and then referencing which item in that array we want by `count` which is `index + 1`.

Will close this now, thank you for your comments.

closed 2020-06-18 by howroyd