espressif/esp-idf #5455: ADC Calibration feature in ESP32s2

NOA-Joe, 2020-06-16:
Ver: Master branch 
https://docs.espressif.com/projects/esp-idf/en/latest/esp32s2/api-reference/peripherals/adc.html
above docs mention ADC Calibration feature in esp_adc_cal/include/esp_adc_cal.h, but the in IDF:
https://github.com/espressif/esp-idf/blob/master/components/esp_adc_cal/CMakeLists.txt
The driver code is not supported ESP32s2 yet. The Build log is showing it can't find this .h file.
Is this ADC Calibration feature is work for ESP32s2 or not?

I try include both "esp_adc_cal/include/esp_adc_cal.h" and  "esp_adc_cal.h" but fail.
## Debug Logs

../main/App_IO.c:15:10: fatal error: esp_adc_cal.h: No such file or directory
 #include "esp_adc_cal.h"

fuAlex, 2020-06-16:
@NOA-Joe This feature for ESP32-S2 is still in internal testing, please wait for the release.

hobiseven, 2020-06-24:
Is the s2 adc as bad as the old esp32d???

fuAlex, 2020-06-29:
@hobiseven Compared to ESP32:  
- The input offset is controlled within 10mV.
- The factory-made chip is programmed with a two-point calibration scheme in eFuse. After calibration, the error can be controlled within +/-1%

NOA-Joe, 2020-06-30:
#define DEFAULT_VREF    1200 
#define DEFAULT_MAX     8191
static const adc_channel_t channel = ADC_CHANNEL_6;     
static const adc_bits_width_t width = ADC_WIDTH_BIT_13;
static const adc_atten_t atten = ADC_ATTEN_DB_2_5;

I set the ADC as above,  the - 2.5dB attenuation (ADC_ATTEN_DB_2_5) gives full-scale voltage 1.5V, but in my test, it should be 1.2V that why I set the DEFAULT_VREF as 1200 here.
I am not sure if this because the driver for ESP32 only not for ESP32S2.

closed 2020-08-20 by projectgus

william-ferguson-au, 2020-09-19:
@projectgus any chance this get backported to 4.1?
Would be really useful to be able to use ADC in 4.1 for esp32s2