espressif/esp-idf #5456: Testing secure boot and flash encryption, bootloader cannot set CRYPT_CONFIG efuse

bill-iot-hw, 2020-06-16:
## Environment

- Development Kit:      none
- Kit version n/a
- Module or chip used:  ESP32-WROOM-32U
- IDF version v3.3.2-256-g332e243f1
- Build System:         Make
- Compiler version xtensa-esp32-elf-gcc.exe (crosstool-NG crosstool-ng-1.22.0-80-g6c4433a5) 5.2.0
- Operating System:     Windows
- (Windows only) environment type: MSYS2 mingw32
- Using an IDE?: No
- Power Supply:         external 5V

## Problem Description
Trying to enable secure boot and flash encryption but bootloader tries to set CRYPT_CONFIG efuse and ESP32 crashes with: 
Fatal exception (0): IllegalInstruction
epc1=0x40081038, epc2=0x00000000, epc3=0x00000000, excvaddr=0x00000000, depc=0x00000000

### Expected Behavior
Secure boot and flash encryption are enabled, bootloader encrypts the files and ESP32 boots the factory partition app
### Actual Behavior
The device crashes, reboots, cycle repeats.
### Steps to reproduce

1. Following ESP-IDF docs V3.3 for reflashable bootloader and pregenerated flash encryption key
2. make bootloader
3. follow make bootloader output to burn key and flash bootloader
4. burn pregenerated flash key to efuses
5. make flash monitor
6. Can erase flash and repeat the flashing bootloader and make flash monitor steps to get same result.

## Debug Logs
MONITOR
--- idf_monitor on COM4 115200 ---
--- Quit: Ctrl+] | Menu: Ctrl+T | Help: Ctrl+T followed by Ctrl+H ---
paddr=0x0000102c vadets Jun  8 2016 00:22:57

rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0018,len:4
load:0x3fff001c,len:11336
load:0x40078000,len:21124
load:0x40080400,len:6836
entry 0x400807ec
I (90) boot: Chip Revision: 1
I (90) boot_comm: chip revision: 1, min. bootloader chip revision: 0
I (43) boot: ESP-IDF v3.3.2-256-g332e243f1-dirty 2nd stage bootloader
I (43) boot: compile time 18:20:59
I (44) boot: Enabling RNG early entropy source...
I (49) boot: SPI Speed      : 40MHz
I (54) boot: SPI Mode       : DIO
I (58) boot: SPI Flash Size : 4MB
I (62) boot: Partition Table:
I (65) boot: ## Label            Usage          Type ST Offset   Length
I (73) boot:  0 nvs              WiFi data        01 02 0000d000 00004000
I (80) boot:  1 otadata          OTA data         01 00 00011000 00002000
I (87) boot:  2 phy_init         RF data          01 01 00013000 00001000
I (95) boot:  3 factory          factory app      00 00 00020000 00100000
I (102) boot:  4 ota_0            OTA app          00 10 00120000 00100000
I (110) boot:  5 ota_1            OTA app          00 11 00220000 00100000
I (118) boot:  6 nvs_dev_info     WiFi data        01 02 00320000 00006000
I (125) boot: End of partition table
I (129) boot: Defaulting to factory image
I (134) boot_comm: chip revision: 1, min. application chip revision: 0
I (141) esp_image: segment 0: paddr=0x00020020 vaddr=0x3f400020 size=0x31194 (201108) map
I (221) esp_image: segment 1: paddr=0x000511bc vaddr=0x3ffb0000 size=0x03c94 ( 15508) load
I (227) esp_image: segment 2: paddr=0x00054e58 vaddr=0x40080000 size=0x00400 (  1024) load
0x40080000: _WindowOverflow4 at C:/msys32/home/mcgra/esp/esp-idf/components/freertos/xtensa_vectors.S:1779

I (228) esp_image: segment 3: paddr=0x00055260 vaddr=0x40080400 size=0x0adb0 ( 44464) load
I (255) esp_image: segment 4: paddr=0x00060018 vaddr=0x400d0018 size=0xa0768 (657256) map
0x400d0018: _flash_cache_start at ??:?

I (486) esp_image: segment 5: paddr=0x00100788 vaddr=0x4008b1b0 size=0x0a474 ( 42100) load
0x4008b1b0: xthal_window_spill_nw at /Users/igrokhotkov/e/esp32/hal/hal/windowspill_asm.S:227

I (504) esp_image: segment 6: paddr=0x0010ac04 vaddr=0x00000000 size=0x0537c ( 21372)
I (512) esp_image: Verifying image signature...
I (915) boot: Loaded app from partition at offset 0x20000
I (915) boot_comm: chip revision: 1, min. application chip revision: 0
I (917) esp_image: segment 0: paddr=0x00001020 vaddr=0x3fff0018 size=0x00004 (     4)
I (925) esp_image: segment 1: paddr=0x0000102c vaddr=0x3fff001c size=0x02c48 ( 11336)
I (938) esp_image: segment 2: paddr=0x00003c7c vaddr=0x40078000 size=0x05284 ( 21124)
I (950) esp_image: segment 3: paddr=0x00008f08 vaddr=0x40080400 size=0x01ab4 (  6836)
W (954) secure_boot: Using pre-loaded secure boot key in EFUSE block 2
I (958) secure_boot: Generating secure boot digest...
I (1010) secure_boot: Digest generation complete.
I (1010) boot: Checking flash encryption...
W (1010) flash_encrypt: Using pre-loaded flash encryption key in EFUSE block 1
I (1016) flash_encrypt: Setting CRYPT_CONFIG efuse to 0xF
Fatal exception (0): IllegalInstruction
epc1=0x40081038, epc2=0x00000000, epc3=0x00000000, excvaddr=0x00000000, depc=0x00000000

### efuse summary
EFUSE_NAME             Description = [Meaningful Value] [Readable/Writeable] (Hex Value)
----------------------------------------------------------------------------------------
Security fuses:
FLASH_CRYPT_CNT        Flash encryption mode counter                     = 0 R/W (0x0)
FLASH_CRYPT_CONFIG     Flash encryption config (key tweak bits)          = 0 R/W (0x0)
CONSOLE_DEBUG_DISABLE  Disable ROM BASIC interpreter fallback            = 1 R/W (0x1)
ABS_DONE_0             secure boot enabled for bootloader                = 0 R/W (0x0)
ABS_DONE_1             secure boot abstract 1 locked                     = 0 R/W (0x0)
JTAG_DISABLE           Disable JTAG                                      = 0 R/W (0x0)
DISABLE_DL_ENCRYPT     Disable flash encryption in UART bootloader       = 0 R/W (0x0)
DISABLE_DL_DECRYPT     Disable flash decryption in UART bootloader       = 0 R/W (0x0)
DISABLE_DL_CACHE       Disable flash cache in UART bootloader            = 0 R/W (0x0)
BLK1                   Flash encryption key
  = ?? ?? ?? ?? ?? ?? ?? ?? ?? ?? ?? ?? ?? ?? ?? ?? ?? ?? ?? ?? ?? ?? ?? ?? ?? ?? ?? ?? ?? ?? ?? ?? -/-
BLK2                   Secure boot key
  = ?? ?? ?? ?? ?? ?? ?? ?? ?? ?? ?? ?? ?? ?? ?? ?? ?? ?? ?? ?? ?? ?? ?? ?? ?? ?? ?? ?? ?? ?? ?? ?? -/-
BLK3                   Variable Block 3
  = 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 R/W

Efuse fuses:
WR_DIS                 Efuse write disable mask                          = 384 R/W (0x180)
RD_DIS                 Efuse read disablemask                            = 3 R/W (0x3)
CODING_SCHEME          Efuse variable block length scheme                = 0 R/W (0x0)
KEY_STATUS             Usage of efuse block 3 (reserved)                 = 0 R/W (0x0)

Config fuses:
XPD_SDIO_FORCE         Ignore MTDI pin (GPIO12) for VDD_SDIO on reset    = 0 R/W (0x0)
XPD_SDIO_REG           If XPD_SDIO_FORCE, enable VDD_SDIO reg on reset   = 0 R/W (0x0)
XPD_SDIO_TIEH          If XPD_SDIO_FORCE & XPD_SDIO_REG, 1=3.3V 0=1.8V   = 0 R/W (0x0)
CLK8M_FREQ             8MHz clock freq override                          = 49 R/W (0x31)
SPI_PAD_CONFIG_CLK     Override SD_CLK pad (GPIO6/SPICLK)                = 0 R/W (0x0)
SPI_PAD_CONFIG_Q       Override SD_DATA_0 pad (GPIO7/SPIQ)               = 0 R/W (0x0)
SPI_PAD_CONFIG_D       Override SD_DATA_1 pad (GPIO8/SPID)               = 0 R/W (0x0)
SPI_PAD_CONFIG_HD      Override SD_DATA_2 pad (GPIO9/SPIHD)              = 0 R/W (0x0)
SPI_PAD_CONFIG_CS0     Override SD_CMD pad (GPIO11/SPICS0)               = 0 R/W (0x0)
DISABLE_SDIO_HOST      Disable SDIO host                                 = 0 R/W (0x0)

Identity fuses:
MAC                    Factory MAC Address
  = 98:f4:ab:1a:9e:6c (CRC 0x6d OK) R/W
CHIP_VER_REV1          Silicon Revision 1                                = 1 R/W (0x1)
CHIP_VER_REV2          Silicon Revision 2                                = 0 R/W (0x0)
CHIP_VERSION           Reserved for future chip versions                 = 2 R/W (0x2)
CHIP_PACKAGE           Chip package identifier                           = 1 R/W (0x1)

Calibration fuses:
BLK3_PART_RESERVE      BLOCK3 partially served for ADC calibration data  = 0 R/W (0x0)
ADC_VREF               Voltage reference calibration                     = 1128 R/W (0x4)

Flash voltage (VDD_SDIO) determined by GPIO12 on reset (High for 1.8V, Low/NC for 3.3V).

## Other items if possible

#
### Security features
#
CONFIG_SECURE_SIGNED_ON_BOOT=y
CONFIG_SECURE_SIGNED_ON_UPDATE=y
CONFIG_SECURE_SIGNED_APPS=y
CONFIG_SECURE_BOOT_ENABLED=y
CONFIG_SECURE_BOOTLOADER_ONE_TIME_FLASH=
CONFIG_SECURE_BOOTLOADER_REFLASHABLE=y
CONFIG_SECURE_BOOT_BUILD_SIGNED_BINARIES=y
CONFIG_SECURE_BOOT_SIGNING_KEY="dhp_iot_secure_boot_signing_key.pem"
CONFIG_SECURE_BOOTLOADER_KEY_ENCODING_256BIT=y
CONFIG_SECURE_BOOTLOADER_KEY_ENCODING_192BIT=
CONFIG_SECURE_BOOT_INSECURE=
CONFIG_FLASH_ENCRYPTION_ENABLED=y
CONFIG_FLASH_ENCRYPTION_INSECURE=
CONFIG_FLASH_ENCRYPTION_DISABLE_PLAINTEXT=

hansn57, 2020-06-22:
Using ESP-IDF v4.1-dev-1543-g431066f16 i get to have a similar error while setting flash encryption in debug mode. Example compiles and works as intended if flash encryption bit is not set.

I (782) flash_encrypt: Disable JTAG...
I (786) flash_encrypt: Disable ROM BASIC interpreter fallback...
Fatal exception (0): IllegalInstruction
epc1=0x40006843, epc2=0x00000000, epc3=0x00000000, excvaddr=0x00000000, depc=0x00000000

Switching to ESP-IDF v4.2-dev-1905-g625bd5eb1 everything works as intended.

bill-iot-hw, 2020-06-22:
I hope someone can give a fix/work-around for V3.3 because switching to V4.2 is not an option for my project.

hansn57, 2020-06-23:
Some more information on my debugging process for esp-idf v4.1. I have made a clean new copy of the release/v4.1 branch from the git repo and reinstalled the tool chain from scratch with install.sh

Version now serving is: ESP-IDF v4.1-beta2-141-g84b51781c
with
Executing /home/dev/.espressif/python_env/idf4.1_py2.7_env/bin/python /home/dev/Projects/esp/esp-idf/tools/idf_tools.py list
* xtensa-esp32-elf: Toolchain for Xtensa (ESP32) based on GCC
  - esp-2019r2-8.2.0 (recommended, installed)
* xtensa-esp32s2-elf: Toolchain for Xtensa (ESP32-S2) based on GCC
  - esp-2019r2-8.2.0 (recommended, installed)
* esp32ulp-elf: Toolchain for ESP32 ULP coprocessor
  - 2.28.51-esp-20191205 (recommended, installed)
* esp32s2ulp-elf: Toolchain for ESP32-S2 ULP coprocessor
  - 2.28.51-esp-20191205 (recommended, installed)
* cmake: CMake build system (optional)
  - 3.13.4 (recommended)
* openocd-esp32: OpenOCD for ESP32
  - v0.10.0-esp32-20191114 (recommended, installed)
* ninja: Ninja build system (optional)
  - 1.9.0 (recommended)

issue is gone for `idf.py encrypted-flash` command but still present with `idf.py encrypted-app-flash`

projectgus, 2020-06-23:
Hi @bill-iot-hw @hansn57,

Thanks for being patient while someone got back to you.

Bill, judging from the boot log messages your bootloader.bin is around 40KB in size. Are you using the default partition table offset in the config? With the default offset, the bootloader.bin is limited to 0x7000 (28672) bytes in length. [More details can be found here](https://docs.espressif.com/projects/esp-idf/en/latest/esp32/security/secure-boot-v1.html#secure-boot-bootloader-size).

The ROM code verifies a checksum but it is only an 8-bit checksum so collisions are unfortunately possible. When this happens the ROM code will run until it accesses an address which has been overridden by the partition table data, and then crash with an IllegalInstructionException.

There is a flashing-time check if the bootloader is flashed at the same time as the partition table, but unfortunately these are usually flashed separately when Secure Boot is on. A comprehensive check for this in the build system is planned but not implemented yet.

If the partition table offset is correct for the bootloader size, is it possible for you to share the `bootloader.elf` and `sdkconfig` files corresponding to the failing log output? If they contain sensitive information then I can provide an email address to send them to privately.

@hansn57 Your crash looks similar but may not be the same root cause, but please check the details above as well. If bootloader size doesn't explain the problem then is it possible for you to provide the same files as well, please?

bill-iot-hw, 2020-06-27:
@projectgus Thanks for getting back to me. I did have partition table offset at 0xC000 ( about 49kB ?) and my bootloader bin and relash digest bin are 40kB and 44kB respectively. But I tried increasing the offset to 0xF000 just in case, but still getting the same errors.

I can share the bootloader.elf and sdkconfig but would prefer to do so privately.

projectgus, 2020-06-29:
Hi @bill-iot-hw ,

Thanks for clarifying. Agree that 0xC000 should be enough for those sizes. Could you email the bootloader binary and sdkconfig to angus at espressif dot com, please? If the binary has been rebuilt since the logs above were captured, could you also capture a new log please (will be cross-referencing the crash address shown in the log).

Angus

projectgus, 2020-07-06:
Hi @bill-iot-hw ,

Thanks for sending the files. I was able to reproduce the problem based on the bootloader ELF and found a bug in v3.3 that was fixed in v4.0 but the fix was accidentally not backported.

A fix is pending and will be merged soon, this issue will be closed once it merges. I've attached a patch here that you can apply to your ESP-IDF install (`cd $IDF_PATH; git am FILENAME`) in the meantime, and to confirm the fix works for you.

[0001-bootloader-Fix-regression-burning-efuses-late-in-boo.patch.txt](https://github.com/espressif/esp-idf/files/4877262/0001-bootloader-Fix-regression-burning-efuses-late-in-boo.patch.txt)

@hansn57 This particular bug doesn't exist in ESP-IDF v4.x, so it looks like the issue you're seeing has a different root cause. If you're still seeing crashes depending on how you flash the device, and you can confirm that the bootloader binary is not overlapping the partition table offset (see two comments up from this one), then please open a new issue with all the details and we'll investigate it separately.

hansn57, 2020-07-06:
Hi @projectgus and thanks for the detailed debugging!
We have closely followed the provided how-to and have used also several offsets (0x9000, 0xa000, 0xb000) for our partition just as @bill-iot-hw described in his notes. We did not observe different behavior.

I will provide the files and send it to the provided address.
Have a good day!

bill-iot-hw, 2020-07-08:
@projectgus Thanks for your help, the patch seems to have worked!
I did see a few questionable messages during the first boot as encryption was being applied, though ultimately the image seems to have come up and is working. 
Here's an excerpt and attached are full logs from the first boot with secure boot and flash encryption enabled:

I (2597) flash_encrypt: Encrypting partition 3 at offset 0x20000...
E (15841) esp_image: image at 0x120000 has invalid magic byte
E (15841) boot_comm: mismatch chip ID, expected 0, found 65535
E (15843) boot_comm: can't run on lower chip revision, expected 1, found 255
W (15850) esp_image: image at 0x120000 has invalid SPI mode 255
W (15857) esp_image: image at 0x120000 has invalid SPI size 15
E (15864) esp_image: image at 0x220000 has invalid magic byte
E (15870) boot_comm: mismatch chip ID, expected 0, found 65535
E (15876) boot_comm: can't run on lower chip revision, expected 1, found 255
W (15884) esp_image: image at 0x220000 has invalid SPI mode 255
W (15891) esp_image: image at 0x220000 has invalid SPI size 15
W (15897) flash_encrypt: Not disabling FLASH_CRYPT_CNT - plaintext flashing is still possible
I (15918) flash_encrypt: Flash encryption completed
I (15918) boot: Checking secure boot...
I (15918) secure_boot: blowing secure boot efuse...
I (15924) secure_boot: Disable JTAG...
I (15929) secure_boot: Disable ROM BASIC interpreter fallback...
I (15947) secure_boot: secure boot is now enabled for bootloader image
I (15947) boot: Resetting with flash encryption enabled...
ets Jun  8 2016 00:22:57

[secure-boot-applied.log](https://github.com/espressif/esp-idf/files/4891467/secure-boot-applied.log)

Just wanted to check if this is normal to see these invalid xxx messages during first encryption or if there's something else I need to look in to. Thanks again!

projectgus, 2020-07-09:
Hi @bill-iot-hw ,

Thanks for confirming. The warning log lines from `esp_image` and `boot_comm` are most likely because the two app partitions at these offsets in the flash have been erased and not written (ie the values are all 0xFFFF, 0xFF or 0xF). We should fix the bootloader behaviour so it logs something more meaningful and less alarming in this case, thanks for pointing it out.

> W (15897) flash_encrypt: Not disabling FLASH_CRYPT_CNT - plaintext flashing is still possible

Is this the behaviour you expected? If so then I don't think there is anything else to worry about.

Angus

bill-iot-hw, 2020-07-09:
@projectgus Thanks for confirming, and yes I left plaintext flashing intact for this test. Since you mentioned closing after the fix is merged I'll leave the issue open for now, but I consider my problem resolved : )

fromeijn, 2020-07-27:
@projectgus Just also saw this problem and updated my v3.3 branche, now not seeing this issue anymore. I guess this issue can be closed.

Alvin1Zhang, 2020-07-30:
Thanks for reporting, the fix is available at https://github.com/espressif/esp-idf/commit/9d6c3632e85dc71dc0944da793623e8bbf2d6d8f. Feel free to reopen if the issue still happens. Thanks.

closed 2020-07-30 by Alvin1Zhang