espressif/esp-idf #5457: 在ble安全蓝牙配对中 怎么使用4字节的pin码

fzboffice, 2020-06-16:
尝试过 目前是只能6位

Alvin1Zhang, 2020-06-16:
Thanks for reporting. Would you please help provide more details as suggested in the [issue template](https://github.com/espressif/esp-idf/issues/new?assignees=&labels=&template=bug_report.md&title=)? Information like elf, sdk configuration, backtrace, log outputs, commit ID, hardware and etc. would help us debug further. And for "how" questions, suggest you could raise a post on our forum at https://esp32.com. Thanks.

xiewenxiang, 2020-06-16:
从协议上讲， ble 加密的 key 就是6位十进制数：
The user is shown a six digit number (from "000000" to "999999")
999999 = 0xF423F
换成字节也是4字节之内啊
如果你需要弄的更小，也可以设置成静态的key

fzboffice, 2020-06-17:
> 从协议上讲， ble 加密的 key 就是6位十进制数：
> The user is shown a six digit number (from "000000" to "999999")
> 999999 = 0xF423F
> 换成字节也是4字节之内啊
> 如果你需要弄的更小，也可以设置成静态的key

@xiewenxiang  是我没表达清楚，我注意到手机去配对一个蓝牙设备的时候pin可以是四个数字 现在esp32的配对需要我输入6个数字，我的意思是能否变成输入4个数字即可

xiewenxiang, 2020-06-17:
那是2.1之前的经典蓝牙的配对界面，在那之前还没有BLE

你看到的应该是经典蓝牙的配对界面，如果你使用的是经典蓝牙，那么可以的，可以在menuconfig里面把 simple pairing 关闭

fzboffice, 2020-06-17:
> 那是2.1之前的经典蓝牙的配对界面，在那之前还没有BLE
> 
> 你看到的应该是经典蓝牙的配对界面，如果你使用的是经典蓝牙，那么可以的，可以在menuconfig里面把 simple pairing 关闭

感谢你的解答，ble强制输入6个数字，我明白了！

Alvin1Zhang, 2020-06-17:
Glad to hear your issue gets resolved. Will close, thanks.

closed 2020-06-17 by Alvin1Zhang