espressif/esp-idf #5458: failing to set duty resolution of ledc timer

michox, 2020-06-16:
----------------------------- Delete below -----------------------------

**Reminder: If your issue is a general question, starts similar to "How do I..", or is related to 3rd party development kits/libs, please discuss this on our community forum at https://esp32.com instead.**

INSTRUCTIONS
============

Before submitting a new issue, please follow the checklist and try to find the answer.

- [ x] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [x ] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [x ] I have searched the issue tracker for a similar issue and not found a similar issue.

If the issue cannot be solved after the steps before, please follow these instructions so we can get the needed information to help you in a quick and effective fashion.

1. Fill in all the fields under **Environment** marked with [ ] by picking the correct option for you in each case and deleting the others.
2. Describe your problem.
3. Include [debug logs from the "monitor" tool](https://docs.espressif.com/projects/esp-idf/en/latest/get-started/idf-monitor.html#automatically-decoding-addresses), or [coredumps](https://docs.espressif.com/projects/esp-idf/en/latest/api-guides/core_dump.html).
4. Providing as much information as possible under **Other items if possible** will help us locate and fix the problem.
5. Use [Markdown](https://guides.github.com/features/mastering-markdown/) (see formatting buttons above) and the Preview tab to check what the issue will look like.
6. Delete these instructions from the above to the below marker lines before submitting this issue.

**IMPORTANT: If you do not follow these instructions and provide the necessary details, your issue may not be resolved.**

----------------------------- Delete above -----------------------------

## Environment

- Development Kit:      [ESP32-DevKitC]
- Kit version (v1]
- Module or chip used:  [ESP32-WROOM-32]
- IDF version latest PlatformIO version
- Build System:         PlatformIO

- Operating System:     [Windows]
- (Windows only) environment type: PlatformIO
- Using an IDE?: [No|PlatformIO]

## Problem Description

I am trying to set the duty resolution like in https://github.com/espressif/esp-idf/blob/7d75213674b4572f90c68162ad6fe9b16dae65ad/examples/peripherals/ledc/main/ledc_example_main.c

### Expected Behavior
shozld be able to set duty resolution
### Actual Behavior
IDE warning: a designator for an anonymous union member can only appear within braces corresponding to that anonymous union. Also, LEDC_AUTO_CLK is undefined even though I implemented all headers from the example
### Steps to reproduce
include header and in some function paste the code

### Code to reproduce this issue

```ledc_timer_config_t ledc_timer = {
        .duty_resolution=LEDC_TIMER_13_BIT,               // always 50%
        .freq_hz = 0,                       // frequency of PWM signal
        .speed_mode = LEDC_HIGH_SPEED_MODE, // timer mode
        .timer_num = LEDC_TIMER_1,          // timer index
        .clk_cfg = LEDC_AUTO_CLK,           // Auto select the source clock
    };

```
// If your code is longer than 30 lines, [GIST](https://gist.github.com) is preferred.

## Debug Logs
before deleting .clk_cfg = LEDC_AUTO_CLK, 
```
error: 'ledc_timer_config_t' has no non-static data member named 'clk_cfg'
```
after deleting it
```
sorry, unimplemented: non-trivial designated initializers not supported
```

Alvin1Zhang, 2020-06-17:
Thanks for reporting and letting us know, we will look into.

wilhelmw-97, 2020-06-18:
a compile time error is ..erm.. probably because of a grammatical error in your code. did you get the struct initialization right?

michox, 2020-06-18:
should be. after editing the source code and replacing the union by only duty_resolution, it works fine. Unfortunately, I was not able to find how to initialize an anonymous union like this but also the example should be updated to a working one so I thought I will let you know

Alvin1Zhang, 2020-08-14:
@michox Thanks for reporting, would you please help share if any updates for this issue? Thanks.

michox, 2020-08-14:
I have given up on this and are using the arduino component instead which works fine

rowifi, 2021-01-23:
Instead of using 
   .duty_resolution=LEDC_TIMER_13_BIT,  
try taking it out of the structure construct and initialising separately

 structname. duty_resolution=LEDC_TIMER_13_BIT;

rowifi, 2021-01-23:
Just like ..

```
ledc_timer_config_t ledc_timer;
  ledc_channel_config_t ledc_channel;
 
  ledc_timer.speed_mode   = LEDC_HIGH_SPEED_MODE;
  ledc_timer.timer_num    = LEDC_TIMER_0;
  ledc_timer.duty_resolution       = LEDC_TIMER_8_BIT;// (ledc_timer_bit_t) 1;
  ledc_timer.freq_hz      = 500;
 
  ledc_channel.channel    = LEDC_CHANNEL_0;
  ledc_channel.gpio_num   = LEDA_PIN;
  ledc_channel.speed_mode = LEDC_HIGH_SPEED_MODE;
  ledc_channel.timer_sel  = LEDC_TIMER_0;
  ledc_channel.duty       = 80;
 
  ledc_timer_config(&ledc_timer);
  ledc_channel_config(&ledc_channel);
```

ginkgm, 2023-07-26:
This issue has been long. If you still find issues, feel free to reopen.

closed 2023-07-26 by ginkgm