espressif/esp-idf #5459: tinyusb component does not compile with c++ source files

sAlexander, 2020-06-16:
# Summary

The example provided in esp-idf/examples/peripherals/usb/tusb_sample_descriptor is great for getting started with tinyusb on the esp32s2. However, the example does not compile with C++ source files. 

# How to reproduce

Specifically, if you rename tusb_sample_descriptor.c to tusb_sample_descriptor.cc (and update CMakeList.txt), the linking stage will fail due to duplicate symbols:

```
ld: esp-idf/main/libmain.a(tusb_sample_descriptor.cc.obj): multiple definition of `descriptor_str_kconfig';
ld: esp-idf/main/libmain.a(tusb_sample_descriptor.cc.obj): multiple definition of `descriptor_kconfig';
...
collect2: error: ld returned 1 exit status
```
These variables are all declared in `components/tinyusb/port/common/include/usb_descriptors.h`.

# The potential root cause

Running nm on the c vs cc generated objects shines some light onto the issue:

*c*:
```
xtensa-esp32s2-elf-nm tusb_sample_descriptor.c.obj 
00000000 T app_main
00000012 C descriptor_kconfig
0000001c C descriptor_str_kconfig
0000001c C descriptor_str_tinyusb
00000012 C descriptor_tinyusb
         U _esp_error_check_failed
         U esp_log_timestamp
         U esp_log_write
00000000 r __func__$6692
         U tinyusb_driver_install
         U tud_task
00000001 C tusb_desc_set
00000000 t usb_device_task
         U xTaskCreatePinnedToCore
```

*cc*:
```
xtensa-esp32s2-elf-nm tusb_sample_descriptor.cc.obj 
00000000 T app_main
00000000 B descriptor_kconfig
00000000 B descriptor_str_kconfig
00000000 B descriptor_str_tinyusb
00000000 B descriptor_tinyusb
         U _esp_error_check_failed
         U esp_log_timestamp
         U esp_log_write
         U tinyusb_driver_install
         U tud_task
00000000 B tusb_desc_set
         U xTaskCreatePinnedToCore
00000000 t _ZL15usb_device_taskPv
00000000 r _ZZ8app_mainE19__PRETTY_FUNCTION__

```

For the c code, the compiler puts the usb related variables from `usb_descriptors.h` into the common block ("C" label above). From [the gcc documentation for -fcommon](https://gcc.gnu.org/onlinedocs/gcc/Code-Gen-Options.html):

> The -fcommon places uninitialized global variables in a common block. This allows the linker to resolve all tentative definitions of the same variable in different compilation units to the same object, or to a non-tentative definition. This behavior is inconsistent with C++, and on many targets implies a speed and code size penalty on global variable references. It is mainly useful to enable legacy code to link without errors.

For the cc code, the compiler puts the usb related variables from `usb_descriptors.h` into the BSS block. This is also inline with the documentation:

> The default is -fno-common, which specifies that the compiler places uninitialized global variables in the BSS section of the object file. This inhibits the merging of tentative definitions by the linker so you get a multiple-definition error if the same variable is accidentally defined in more than one compilation unit.

# A potential fix

The fix is potentially very simple; the variables in `usb_descriptors.h` should be prepended with `extern`, which will indicate that those variables are only declared in `usb_descriptors.h`, while being defined elsewhere (specifically: `usb_descriptors.c`). As `usb_descriptors.c` is included in `libtinyusb.a`, anyone using tinyusb should get those definitions during linking.

chegewara, 2020-06-16:
Maybe this is a bit off topic, but tinyusb library has been updated a lot in past 2 months (since esp-idf example and port prepared):
https://github.com/hathach/tinyusb

You can check my PR, its easier to use and you have few more examples included:
https://github.com/espressif/esp-idf/pull/5355

sAlexander, 2020-06-16:
Hi @chegewara, thanks for the comment! I saw your PR when I was searching for duplicates of the issue I describe above before filing. I'm very excited to dig into the new examples (I'm implementing CDC myself). Thank you!

Before filing this issue, I looked at `components/tinyusb/port/common/include/usb_descriptors.h` in your PR, and it appeared to suffer from the same issue I describe in this issue (though I haven't checked too carefully). I'm preparing a PR to fix this, and I'll confirm that it works with your PR as well.

chegewara, 2020-06-16:
No worries, my PR most likely wont get merged.
You can also check arduino as component, where i am using tinyusb without esp-idf port. Its all C++.
Another off topic, sorry.
https://github.com/chegewara/esp32s2-tinyusb-arduino-as-component

Alvin1Zhang, 2020-06-17:
Thanks for reporting and providing the contribution and PR. We will look into.

dobairoland, 2020-06-17:
We have a couple of work-in-progress changes internally which hasn't been merged or pushed to Github yet (and got delayed for some other technical issues). We are sorry for that. They also addresses C++ compatibility issues.

> No worries, my PR most likely wont get merged.

Your contribution is highly appreciated @chegewara. I've noticed some incompatibilities with our internal changes. Your PR can be considered after those got merged. We are sorry for the delay.

chegewara, 2020-06-17:
> Your contribution is highly appreciated @chegewara. I've noticed some incompatibilities with our internal changes. Your PR can be considered after those got merged. We are sorry for the delay.

No worries, its not complaining. I made this PR not to be merged, but to give other people, just like me, who wants to play with USB to test it. Im not making PRs in esp-idf because i believe my programming style does not fit here. I know there is a lot better than me programmers at espressif (all of them) and i respect it and awaiting updates and fixes patiently.

sAlexander, 2020-06-21:
@dobairoland, thanks for the info! Is there an issue or other resource that we could follow to understand more about the progress you describe? I'm looking forward to the improvements.

Thank you!

dobairoland, 2020-06-22:
@sAlexander No. Probably the best way the get informed through this issue, the related PR, and https://github.com/espressif/esp-idf/pull/5355. My colleague will get back to you.

Alvin1Zhang, 2021-08-25:
Thanks for reporting, fix is available https://github.com/espressif/esp-idf/commit/e97e7281b260cdd8b963cb503ad296d38cba348f, feel free to reopen.

closed 2021-08-25 by Alvin1Zhang