espressif/esp-idf #5461: Passing serial port as ENV variable is inconsistent

kzyapkov, 2020-06-17:
If one wants to specify the serial port for idf.py in environment instead of cli option, they have to set `ESPPORT`. However, idf_monitor.py probes `ESPTOOL_PORT`. None of those are in the docs or `--help` output so I had to dig through the code.

I think this should be added to the documentation, output of `idf.py --help` and made consistent accross tools.

Alvin1Zhang, 2020-06-17:
Thanks for reporting, we will look into.

dobairoland, 2020-06-17:
Hi @kzyapkov. Have you read [this page](https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-guides/build-system.html) as well? It is mentioned at several places.

But you are right that it could useful to mention this in `idf.py --help` as well.

kzyapkov, 2020-06-17:
Thanks, saw the note. That's a lot of text to read for something like this though, please add a note in the `-p` option docstring.

closed 2020-07-10 by espressif-bot