espressif/esp-idf #5463: mesh: A stack overflow in task MEVT has been detected

dshil, 2020-06-17:
## Environment

- Module or chip used:  ESP32-WROVER-B
- IDF version: v4.0.1
- Build System: idf.py (cmake - the same)
- Compiler version: xtensa-esp32-elf-gcc (crosstool-NG esp-2019r2) 8.2.0
- Operating System: Linux
- Using an IDE: No
- Power Supply: USB

## Problem Description

I setup my mesh as:

```cpp
    ESP_ERROR_CHECK(esp_mesh_init());
    ESP_ERROR_CHECK(esp_mesh_set_max_layer(CONFIG_MESH_MAX_LAYER));
    ESP_ERROR_CHECK(esp_mesh_set_vote_percentage(1));
    ESP_ERROR_CHECK(esp_mesh_set_ap_assoc_expire(10));
    
    // I don't want my nodes to connect to the internet on the mesh layer.
    esp_mesh_fix_root(true);

    mesh_cfg_t cfg;
    memset(&cfg, 0, sizeof(cfg));
    cfg.crypto_funcs = &g_wifi_default_mesh_crypto_funcs;
    cfg.channel = CONFIG_MESH_CHANNEL;
    cfg.mesh_ap.max_connection = CONFIG_MESH_AP_CONNECTIONS;

    memcpy((uint8_t*)&cfg.mesh_id, MESH_ID, 6);

    ESP_ERROR_CHECK(esp_mesh_set_config(&cfg));

    ESP_ERROR_CHECK(
        esp_event_handler_register(MESH_EVENT, ESP_EVENT_ANY_ID, &handle_event_, this));
```

And when I flash and then run the program it is crashed each time with the following:

```plain
***ERROR*** A stack overflow in task MEVT has been detected
```

I guess when I call `esp_mesh_start()` internal task that is responsible for mesh management service is started. If yes it would be nice to have an option to configure size of this task, otherwise my app will crash each time.

P.S. I was able to temporary avoid crash reported in #5440 by disabling BLE stack (I still need BLE stack in production environment).

cc: @dongdong004, @ESP-iPENCIL
Relates: #5440

ydesp, 2020-06-18:
There should be no such task named MEVT on IDF version  v4.0.1.

Jiangyafeng, 2020-06-23:
> ## Environment
> * Module or chip used:  ESP32-WROVER-B
> * IDF version: v4.0.1
> * Build System: idf.py (cmake - the same)
> * Compiler version: xtensa-esp32-elf-gcc (crosstool-NG esp-2019r2) 8.2.0
> * Operating System: Linux
> * Using an IDE: No
> * Power Supply: USB
> 
> ## Problem Description
> I setup my mesh as:
> 
> ```c++
>     ESP_ERROR_CHECK(esp_mesh_init());
>     ESP_ERROR_CHECK(esp_mesh_set_max_layer(CONFIG_MESH_MAX_LAYER));
>     ESP_ERROR_CHECK(esp_mesh_set_vote_percentage(1));
>     ESP_ERROR_CHECK(esp_mesh_set_ap_assoc_expire(10));
>     
>     // I don't want my nodes to connect to the internet on the mesh layer.
>     esp_mesh_fix_root(true);
> 
>     mesh_cfg_t cfg;
>     memset(&cfg, 0, sizeof(cfg));
>     cfg.crypto_funcs = &g_wifi_default_mesh_crypto_funcs;
>     cfg.channel = CONFIG_MESH_CHANNEL;
>     cfg.mesh_ap.max_connection = CONFIG_MESH_AP_CONNECTIONS;
> 
>     memcpy((uint8_t*)&cfg.mesh_id, MESH_ID, 6);
> 
>     ESP_ERROR_CHECK(esp_mesh_set_config(&cfg));
> 
>     ESP_ERROR_CHECK(
>         esp_event_handler_register(MESH_EVENT, ESP_EVENT_ANY_ID, &handle_event_, this));
> ```
> 
> And when I flash and then run the program it is crashed each time with the following:
> 
> ```
> ***ERROR*** A stack overflow in task MEVT has been detected
> ```
> 
> I guess when I call `esp_mesh_start()` internal task that is responsible for mesh management service is started. If yes it would be nice to have an option to configure size of this task, otherwise my app will crash each time.
> 
> P.S. I was able to temporary avoid crash reported in #5440 by disabling BLE stack (I still need BLE stack in production environment).
> 
> cc: @dongdong004, @ESP-iPENCIL
> Relates: #5440

Hi, @dshil 
I follow your mesh configuration here, and there is no stack overflow.
1. Are you willing to provide your sample code, so that you can quickly locate the source of the problem?
2. Could you provide configuration configuration, maybe it is caused by some settings.
3. Are you willing to provide the log of the problem, we can use the log to locate the problem.

Alvin1Zhang, 2020-07-01:
@dshil Thanks for reporting, would you please help share if any updates for this issue? Thanks.

dshil, 2020-07-03:
Hi @Jiangyafeng, and thanks again for coming back to me.
Sorry, my initial report was not clear, let me clarify what is going on.

>  I follow your mesh configuration here, and there is no stack overflow.

You were right. First, I want to make it clear that I live C++ environment. The code posted above was called in the object constructor. I have the following workflow:

```cpp
#include <stdio.h>

#include "pipeline.h"
#include "board.h"

extern "C" void app_main() {
    pipeline::Pipeline pipeline;
    app::Board board(pipeline);
    
    configASSERT(board.start());
    configASSERT(pipeline.start());
}
```

`Pipeline` and `Board` are 2 classes that I can't share but inside then there are tones of components and near about 10 different tasks. You can be sure we use a lot of system resources. Lets move next. In the pipeline class I initialized the class that is responsible for mesh network management. I use almost the same code as from examples (examples/mesh/manual_networking). 

The interesting thing is that when I start a mesh it works fine, the problems appear when I handle first event about mesh start using the following code:

```cpp
    // Since I want to control routing by myself I disable the self-organization mode
    ESP_ERROR_CHECK(esp_mesh_set_self_organized(0, 0));

    mesh_addr_t id = {
        0,
    };
    esp_mesh_get_id(&id);

    ESP_LOGI(TAG, "mesh-network: started: heap=%d mac=" MACSTR "", MAC2STR(id.addr),
             heap_caps_get_minimum_free_size(MALLOC_CAP_INTERNAL));

    ESP_ERROR_CHECK(esp_wifi_scan_stop());
```

When I call ```ESP_ERROR_CHECK(esp_wifi_scan_stop());``` I got the posted stackoverflow error.

----

In the current section I will try to provide my step by step debugging process. I tried the following things:

- compile and flash the code from examples/mesh/manual_networking on my board - no problem;
- trace the stack size of different tasks - no problem;
- trace heap usage of different tasks - no problem;
- increase stack size of system components as - no result:

```cpp
CONFIG_LWIP_TCPIP_TASK_STACK_SIZE=4096
CONFIG_ESP_IPC_TASK_STACK_SIZE=4096
CONFIG_ESP_TIMER_TASK_STACK_SIZE=4096
CONFIG_ESP_SYSTEM_EVENT_TASK_STACK_SIZE=4096
```

----

After all this debugging stuff I had zero idea what is the problem and one day I tried to check the logging process (I don't know why I god this idea). We replace the standard logger with our own, since we need some pre and post processioning. When I disable custom logger we won't have any stackoverflow issue. I tried to print the remaining stack size of the task that log something and got the following result:

```cpp
esp-idf: I (1749) mesh: <SYSTEM_EVENT_SCAN_DONE>status:fail, num:0, id:129
bytes remain: 796

esp-idf: I (6439) mesh: 3367<scan>self-organized is disabled, users shall read out scan results by themselves.
bytes remain: 1772
```

I use the following code:

```cpp
int log_handler(const char* format, va_list args) {
    configASSERT(format);

    // In release build most of the time Info level is enabled, thus to ensure
    // won't miss any ESP-IDF specific log that does not contain level prefix,
    // LogInfo is used by default.
    LogLevel level = LogInfo;

    switch (format[0]) {
    case 'E':
        level = LogError;
        break;
    case 'I':
        level = LogInfo;
        break;
    case 'W':
        level = LogWarn;
        break;
    case 'D':
        level = LogDebug;
        break;
    case 'V':
        level = LogTrace;
        break;
    default:
        break;
    }

    if (level > core::Logger::instance().level()) {
        return 1;
    }

    char message[core::Logger::EntrySize] = {};
    vsnprintf(message, sizeof(message) - 1, format, args);

    fprintf(stdout, "esp-idf: %s: %d\n", message, uxTaskGetStackHighWaterMark(nullptr));

    return 0;
}
```

When I replace

```cpp
    fprintf(stdout, "esp-idf: %s: %d\n", message, uxTaskGetStackHighWaterMark(nullptr));
```

with my own:

```cpp
    my_log(level, "esp-idf: %s", message);
```

I got the stack overflow.

----

As you can see after we receive `SYSTEM_EVENT_SCAN_DONE` there is a few bytes remaining in stack. I think it would be viable if you can provide an option to set the size of mesh event stack size to be able to avoid such problems.

P.S. sorry for a long response, was experimenting a lot with this issue.
cc: @Alvin1Zhang

dshil, 2020-07-29:
Hello everyone, @Alvin1Zhang  @Jiangyafeng would you plan to take a look into this issue? If not I will close the issue.

Jiangyafeng, 2020-08-05:
First of all, if your API is incorrect, change 
ESP_LOGI(TAG, "mesh-network: started: heap=%d mac=" MACSTR "", MAC2STR(id.addr), heap_caps_get_minimum_free_size(MALLOC_CAP_INTERNAL));
 to 
ESP_LOGI(TAG, "mesh-network: started: heap=%d mac=" MACSTR "", heap_caps_get_minimum_free_size(MALLOC_CAP_INTERNAL), MAC2STR(id.addr));

Can you print the task-list regularly and check the task memory.

For memory problems, you can take the following measures:
1. You can enable "run freertos only on first core" under component config/freertos in menuconfig, which will change dual-core to single-core use;
2. You can enable "Enable'nano' formatting options for prinrf/scanf family" under component config/Newlib in menuconfig.

Jiangyafeng, 2020-08-05:
@dshil

Jiangyafeng, 2020-08-05:
I added your code in the MESH_EVENT_STARTED event. Your stack overflow does not appear, and you are not sure of your specific reasons, so I suggest you use the above methods to debug first to see if the problem can be solved.

Alvin1Zhang, 2020-08-25:
@dshil Thanks for reporting, and sorry for slow turnaround. Would you please help share if any updates for the issue? Thanks.

dshil, 2020-08-29:
Hello everyone, sorry for a delay, was quite busy.

@Jiangyafeng, 

> First of all, if your API is incorrect, change

Thanks, it was a typo.

> I added your code in the MESH_EVENT_STARTED event. Your stack overflow does not appear

Let's simplify the reproducing process by using the following:

- Use `example/mesh/internal_communication` from the latest master
- Add simple custom logger:

```c
int log_handler(const char* format, va_list args)
{
    char message[512] = {};

    if (vsnprintf(message, sizeof(message) - 1, format, args) < 0) {
        message[0] = '\0';
    }

    fprintf(stdout, "stack=%u msg=%s\n", uxTaskGetStackHighWaterMark(NULL), message);

    return 0;
}
```

- Set this logger at the beginning of app_main as:

```c
void app_main(void)
{
    esp_log_set_vprintf(log_handler);
    // other code
}
```

- Compile and flush the example, you should see something similar to:

```
stack=336 msg=I (12885) mesh_main: <MESH_EVENT_PARENT_CONNECTED>layer:0-->1, parent:e0:3f:49:d3:2c:30<ROOT>, ID:77:77:77:77:77:77, duty:0

stack=336 msg=I (13705) mesh_main: <IP_EVENT_STA_GOT_IP>IP:192.168.1.88

stack=684 msg=W (730045) mesh_main: [#RX:718/718][L:1] parent:e0:3f:49:d3:2c:30, receive from 24:6f:28:f0:10:80, size:1460, heap:169476, flag:0[err:0x0, proto:0, tos:0]
```

As you might see there is a little amount of stack left. Since custom logger provided above is very short and does not consume a lot of stack we won't run out of stack but in real application where custom logger can be more complicated the stack overflow issue can easily happen  (as happened in our application).

Jiangyafeng, 2020-08-31:
@dshil 
Thank you for your feedback,
1. First of all, the small stack that appears in the instance is determined by the size of the memory allocated by the corresponding running task. If you want to increase it, you can increase the corresponding running task memory. I use the method of increasing the size in the example, as follows:
E.g:

```c
        xTaskCreate(esp_mesh_p2p_rx_main, "MPRX", 3072, NULL, 5, NULL);
```

change:

```c
        xTaskCreate(esp_mesh_p2p_rx_main, "MPRX", 4096, NULL, 5, NULL);
```

you can see:

```log
stack=340 msg=I (13743) esp_netif_handlers: sta ip: 192.168.3.206, mask: 255.255.255.0, gw: 192.168.3.1

stack=340 msg=I (13743) mesh_main: <IP_EVENT_STA_GOT_IP>IP: 192.168.3.206

stack=1704 msg=W (13983) mesh_main: [#RX:2/2][L:1] parent:20:a6:80:98:dc:b4, receive from 98:f4:ab:76:32: c4, size:1460, heap:157196, flag:0[err:0x0, proto:0, tos:0]

stack=1704 msg=W (14983) mesh_main: [#RX:3/3][L:1] parent:20:a6:80:98:dc:b4, receive from 98:f4:ab:76:32: c4, size:1460, heap:157140, flag:0[err:0x0, proto:0, tos:0]
```

2. If there is not enough memory in the application, you can also use the following method to save some memory.
    - You can enable "run freertos only on first core" under component config/freertos in menuconfig, which will change dual-core to single-core use;
    - You can enable "Enable'nano' formatting options for prinrf/scanf family" under component config/Newlib in menuconfig.

closed 2023-05-24 by Alvin1Zhang