espressif/esp-idf #5465: WiFi Crash REST Server

rebeliousconformist, 2020-06-17:
## Environment

- Development Kit:      [ESP32-Wrover-Kit|
- Kit version v4]
- Module or chip used:  [ESP32-WROOM-32|ESP32-WROOM-32D|ESP32-WROOM-32U|ESP32-WROVER|ESP32-WROVER-I|ESP32-WROVER-B|ESP32-WROVER-IB|ESP32-SOLO-1|ESP32-PICO-D4|ESP32]
- IDF version (run ``git describe --tags`` to find it):
    // v3.2-dev-1148-g96cd3b75c
- Build System:         [idf.py]
- Compiler version (run ``xtensa-esp32-elf-gcc --version`` to find it):
    // 1.22.0-80-g6c4433a
- Operating System:     |Linux|

- Using an IDE?: eclipse
- Power Supply:         USB|

## Problem Description
<pre>wifi spiram 
Heap summary for capabilities 0x00000400:
  At 0x3f80496c len 4175507 free 4140912 allocated 33088 min_free 4118380
    largest_free_block 4126332 alloc_blocks 70 free_blocks 14 total_blocks 84
  Totals:
    free 4140912 allocated 33088 min_free 4118380 largest_free_block 4126332
I (85031) wifi:state: init -&gt; auth (b0)
I (85041) wifi:state: auth -&gt; assoc (0)
I (85051) wifi:state: assoc -&gt; run (10)
I (85351) wifi:connected with PEM, aid = 1, channel 1, BW20, bssid = be:db:4f:33:eb:dc
I (85351) wifi:security type: 3, phy: bgn, rssi: -76
I (85351) wifi:pm start, type: 1

I (85361) wifi softAP: ssid connected PEM 

I (85541) wifi:AP&apos;s beacon interval = 102400 us, DTIM period = 2
Guru Meditation Error: Core  0 panic&apos;ed (IllegalInstruction). Exception was unhandled.
Memory dump at 0x40081fc4: fff66502 0000f01d d2004136
Core  0 register dump:
PC      : 0x40081fc8  PS      : 0x00060c30  A0      : 0x00040020  A1      : 0x3ffc3310  
A2      : 0x3f80d370  A3      : 0x3ffc6070  A4      : 0x00000100  A5      : 0x00000000  
A6      : 0x3f80d418  A7      : 0x00000000  A8      : 0x80081fc8  A9      : 0x3ffc32e0  
A10     : 0x3f80d370  A11     : 0x40157e3c  A12     : 0x00000000  A13     : 0x00000000  
A14     : 0x000000fe  A15     : 0x00000001  SAR     : 0x00000019  EXCCAUSE: 0x00000000  
EXCVADDR: 0x00000000  LBEG    : 0x4008a816  LEND    : 0x4008a821  LCOUNT  : 0x00000000  

Backtrace:0x40081fc5:0x3ffc3310 0x0004001d:0x3ffc3330 |&lt;-CORRUPTED

ELF file SHA256: 3f98a9a537c7cf05

Rebooting...
ets Jun  8 2016 00:22:57

rst:0xc (SW_CPU_RESET),boot:0x1e (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0030,len:4
load:0x3fff0034,len:7168
ho 0 tail 12 room 4
load:0x40078000,len:14128
load:0x40080400,len:4060
</pre>
//Detailed problem description goes here.

### Expected Behavior
No Crash 
### Actual Behavior

### Steps to reproduce

1. connect to the esp and use the url to push an SSID and Password as a request 
2. This is then used to configure the Wifi STA mode config and then connect the station
3. i cannot share proprietery files on the SD CArds
[rest.zip](https://github.com/espressif/esp-idf/files/4794246/rest.zip)

### Code to reproduce this issue

Alvin1Zhang, 2020-06-18:
@rebeliousconformist Thanks for the detailed report and letting us know, we will look into.

rebeliousconformist, 2020-06-18:
I am hosting multiple webpages on the sd card. 
The ESP mode is STA and AP. 
I am using the AP mode to connect the ESP to a wifi network and hence in the AP mode i am passing a username and password . The ESP then init a new config and using this config connects to an AP in the STA mode.
When the ESP is connected to the Wifi AP an MQTT loop is started which starts to publish data to the MQTT broker. 
Now when i try to access the files on the SD card using the web server, it crashes.
@Alvin1Zhang

liuzfesp, 2020-06-18:
Hi @rebeliousconformist, could you paste the backtrace or the *elf file that generates the crash?

rebeliousconformist, 2020-06-18:
the error once again
<pre> (355607) httpd_txrx: httpd_resp_send_err: 500 Internal Server Error - Failed to read existing file
W (355619) httpd_uri: httpd_uri: uri handler execution failed
E (355709) esp-rest: Failed to open file : /www/ajax/var1
W (355710) httpd_txrx: httpd_resp_send_err: 500 Internal Server Error - Failed to read existing file
W (355721) httpd_uri: httpd_uri: uri handler execution failed
E (355807) esp-rest: Failed to open file : /www/ajax/var1
W (355808) httpd_txrx: httpd_resp_send_err: 500 Internal Server Error - Failed to read existing file
W (355817) httpd_uri: httpd_uri: uri handler execution failed
E (355907) esp-rest: Failed to open file : /www/ajax/var1
W (355908) httpd_txrx: httpd_resp_send_err: 500 Internal Server Error - Failed to read existing file
W (355917) httpd_uri: httpd_uri: uri handler execution failed
Guru Meditation Error: Core  0 panic&apos;ed (IllegalInstruction). Exception was unhandled.
Memory dump at 0x40081fbc: fff5a502 0000f01d d2004136
Core  0 register dump:
PC      : 0x40081fc0  PS      : 0x00060730  A0      : 0x00040020  A1      : 0x3ffc27b0  
A2      : 0x3f816258  A3      : 0x3f8040b0  A4      : 0x00000000  A5      : 0x3f8040b8  
A6      : 0x00000000  A7      : 0x00000000  A8      : 0x80081fc0  A9      : 0x3ffc2780  
A10     : 0x3f816258  A11     : 0x40108378  A12     : 0x00000000  A13     : 0x00000000  
A14     : 0x0000116f  A15     : 0x00000000  SAR     : 0x00000019  EXCCAUSE: 0x00000000  
EXCVADDR: 0x00000000  LBEG    : 0x4008a956  LEND    : 0x4008a961  LCOUNT  : 0x00000000  

Backtrace:0x40081fbd:0x3ffc27b0 0x0004001d:0x3ffc27d0 |&lt;-CORRUPTED

ELF file SHA256: b39b411bdb379c78

Rebooting...
ets Jun  8 2016 00:22:57

rst:0xc (SW_CPU_RESET),boot:0x1e (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0030,len:4
load:0x3fff0034,len:7168
ho 0 tail 12 room 4
load:0x40078000,len:14568
ho 0 tail 12 room 4
load:0x40080400,len:4156
entry 0x40080688
I (32) boot: ESP-IDF v4.2-dev-1905-g625bd5eb1 2nd stage bootloader
I (32) boot: compile time 14:29:25
I (33) boot: chip revision: 1
I (37) boot_comm: chip revision: 1, min. bootloader chip revision: 0
I (44) boot.esp32: SPI Speed      : 40MHz
I (49) boot.esp32: SPI Mode       : DIO
I (53) boot.esp32: SPI Flash Size : 4MB
I (58) boot: Enabling RNG early entropy source...
I (63) boot: Partition Table:
I (67) boot: ## Label            Usage          Type ST Offset   Length
I (74) boot:  0 nvs              WiFi data        01 02 00009000 00006000
I (82) boot:  1 phy_init         RF data          01 01 0000f000 00001000
I (89) boot:  2 factory          factory app      00 00 00010000 00120000
I (97) boot:  3 www              Unknown data     01 82 00130000 00200000
I (104) boot: End of partition table
I (108) boot_comm: chip revision: 1, min. application chip revision: 0
I (115) esp_image: segment 0: paddr=0x00010020 vaddr=0x3f400020 size=0x29a88 (170632) map
I (198) esp_image: segment 1: paddr=0x00039ab0 vaddr=0x3ffb0000 size=0x03f9c ( 16284) load
I (206) esp_image: segment 2: paddr=0x0003da54 vaddr=0x40080000 size=0x00404 (  1028) load
I (207) esp_image: segment 3: paddr=0x0003de60 vaddr=0x40080404 size=0x021b8 (  8632) load
I (218) esp_image: segment 4: paddr=0x00040020 vaddr=0x400d0020 size=0xb6e50 (749136) map
I (547) esp_image: segment 5: paddr=0x000f6e78 vaddr=0x400825bc size=0x1aa38 (109112) load
I (617) boot: Loaded app from partition at offset 0x10000
I (617) boot: Disabling RNG early entropy source...
I (617) psram: This chip is ESP32-D0WD
I (623) spiram: Found 64MBit SPI RAM device
I (626) spiram: SPI RAM mode: flash 40m sram 40m
I (632) spiram: PSRAM initialized, cache is in low/high (2-core) mode.
I (639) cpu_start: Pro cpu up.
I (643) cpu_start: Application information:
[... 29 lines cut ...]
I (1738) wifi:Init data frame dynamic rx buffer num: 16
I (1743) wifi:Init management frame dynamic rx buffer num: 16
I (1748) wifi:Init management short buffer num: 32
I (1753) wifi:Init static tx buffer num: 16
I (1757) wifi:Init static rx buffer size: 1600
I (1761) wifi:Init static rx buffer num: 10
I (1765) wifi:Init dynamic rx buffer num: 16
I (1770) wifi:set country: cc=DE schan=1 nchan=14 policy=0

I (1863) phy: phy_version: 4182, f1ba940, Jun  4 2020, 19:40:07, 0, 0
I (1865) wifi:mode : sta (bc:dd:c2:d1:c0:88) + softAP (bc:dd:c2:d1:c0:89)
I (1870) wifi:Total power save buffer number: 8
I (1871) wifi:Init max length of beacon: 752/752
I (1874) wifi:Init max length of beacon: 752/752
I (1879) wifi softAP: wifi_init_softap . SSID:accesspoint password:hello1234 
wifi started print heap details 
 I (1891) gpio: GPIO[13]| InputEn: 0| OutputEn: 1| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0 
Name: SN64G
Type: SDHC/SDXC
Speed: 20 MHz
Size: 60906MB
</pre>

the .elf file zipped
[rest.zip](https://github.com/espressif/esp-idf/files/4798555/rest.zip)

rebeliousconformist, 2020-06-18:
the size analysis yeilds the following:

<pre>Total sizes:
 DRAM .data size:   16284 bytes
 DRAM .bss  size:    7016 bytes
Used static DRAM:   23300 bytes ( 157436 available, 12.9% used)
Used static IRAM:  118769 bytes (  12303 available, 90.6% used)
      Flash code:  749147 bytes
    Flash rodata:  170632 bytes
Total image size:~1061848 bytes (.bin may be padded larger)
</pre>

rebeliousconformist, 2020-06-18:
when the ESP is in APSTA mode and no Station is connected then the following dump is shown

Task: Pre-Scheduler allocs -> CAP_8BIT: 5288 CAP_32BIT: 0
Task: esp_timer -> CAP_8BIT: 1908 CAP_32BIT: 0
Task: ipc0 -> CAP_8BIT: 8328 CAP_32BIT: 0
Task: main -> CAP_8BIT: 106652 CAP_32BIT: 0
Task: ipc1 -> CAP_8BIT: 12 CAP_32BIT: 0
Task: tiT -> CAP_8BIT: 768 CAP_32BIT: 0
Task: wifi -> CAP_8BIT: 51684 CAP_32BIT: 0
Task: mdns -> CAP_8BIT: 304 CAP_32BIT: 0
Task: sys_evt -> CAP_8BIT: 108 CAP_32BIT: 0
Task: httpd -> CAP_8BIT: 208 CAP_32BIT: 0

when a Station connects to the ESP(which is in AP mode), the following heap task info is shown

<pre>Task: Pre-Scheduler allocs -&gt; CAP_8BIT: 5288 CAP_32BIT: 0
Task: esp_timer -&gt; CAP_8BIT: 1908 CAP_32BIT: 0
Task: ipc0 -&gt; CAP_8BIT: 8328 CAP_32BIT: 0
Task: ���?���?xV�� ��?4&#x12;�������|��?\ -&gt; CAP_8BIT: 106636 CAP_32BIT: 0
Task: ipc1 -&gt; CAP_8BIT: 12 CAP_32BIT: 0
Task: tiT -&gt; CAP_8BIT: 2712 CAP_32BIT: 0
Task: wifi -&gt; CAP_8BIT: 52276 CAP_32BIT: 0
Task: mdns -&gt; CAP_8BIT: 112 CAP_32BIT: 0
Task: sys_evt -&gt; CAP_8BIT: 108 CAP_32BIT: 0
Task: httpd -&gt; CAP_8BIT: 12804 CAP_32BIT: 0
Task: mqtt_task -&gt; CAP_8BIT: 344 CAP_32BIT: 0
</pre>

rebeliousconformist, 2020-06-18:
Also there a way to force the mqtt task to clear the heap allocated after x seconds ?
Because in the range between 62000 and 70000 the MQTT task stops publishing.

<pre>receive timeout ID 1 
Task: Pre-Scheduler allocs -&gt; CAP_8BIT: 5288 CAP_32BIT: 0
Task: esp_timer -&gt; CAP_8BIT: 1908 CAP_32BIT: 0
Task: ipc0 -&gt; CAP_8BIT: 8328 CAP_32BIT: 0
Task:  -&gt; CAP_8BIT: 106636 CAP_32BIT: 0
Task: ipc1 -&gt; CAP_8BIT: 12 CAP_32BIT: 0
Task: tiT -&gt; CAP_8BIT: 2852 CAP_32BIT: 0
Task: wifi -&gt; CAP_8BIT: 54024 CAP_32BIT: 0
Task: mdns -&gt; CAP_8BIT: 112 CAP_32BIT: 0
Task: sys_evt -&gt; CAP_8BIT: 108 CAP_32BIT: 0
Task: httpd -&gt; CAP_8BIT: 12812 CAP_32BIT: 0
Task: mqtt_task -&gt; CAP_8BIT: 68632 CAP_32BIT: 0
</pre>

rebeliousconformist, 2020-06-19:
@Alvin1Zhang @liuzfesp  any updates ? 
Regards.

rebeliousconformist, 2020-06-22:
@igrr

Alvin1Zhang, 2020-06-22:
@rebeliousconformist Thanks for reporting and sorry for slow turnaround. We are now checking the issue, will share updates later. Thanks.

liuzfesp, 2020-06-22:
HI @rebeliousconformist the backtrace doesn't indicate too much useful info:

$ xtensa-esp32-elf-addr2line -piaf -e rest.elf 0x40081fbd:0x3ffc27b0 0x0004001d:0x3ffc27d0
0x40081fbd: __wrap_free at /home/pemubuntu/espressif/esp-idf/components/heap/include/heap_trace.inc:162
0x0004001d: ?? ??:0

I suspect the memory has been corrupted by application, is it possible for you to reproduce this issue by enabling memory corrupt debugging option: https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-reference/system/heap_debug.html?highlight=heap#heap-corruption-detection

rebeliousconformist, 2020-06-22:
@liuzfesp i have already attached the code . Wasnt it helpful ?

YouDONG-ESP, 2020-07-01:
hi @rebeliousconformist, i'm trying to reproduce this issue, your [code](https://github.com/espressif/esp-idf/files/4794246/rest.zip) cant be compiled propely. folder like <<front>> which is mentioned in Makefile is missing. I rewrite makefile without these file and it give me this type of log  rest/main/wireless.c:219:40: error: 'IP_EVENT' undeclared  in line 217 and 219. Can you take a look at it. Im going to use your code for debugging. 
Regards

rebeliousconformist, 2020-07-01:
hi @YouDONG-ESP im using the idf.py utility and also eclipse CMake as mentioned in https://docs.espressif.com/projects/esp-idf/en/latest/esp32/get-started-legacy/eclipse-setup.html

i have also cleaned up the code a bit. line 217 and 219 are errors not occuting from my side as they are @espressif specific. But i have cleaned up the makefiles mess. Please let me know how can i help you.
[err_prj.zip](https://github.com/espressif/esp-idf/files/4858839/err_prj.zip)

rebeliousconformist, 2020-07-06:
@YouDONG-ESP any progress/ updates

YouDONG-ESP, 2020-07-07:
Hi @rebeliousconformist, im trying to reproduce this issue but i do not really understand how it works. You got one esp running in AP/STA mode with the code u gave, it ran in ap mode at first, but how does it init a new config and goes to sta mode?

rebeliousconformist, 2020-07-08:
@YouDONG-ESP when you are in the AP mode then you need to push a username and password to the function via a web interface 

pushPassword(inputUsername,inputPassword) {
            $.post("/ajax/passwordPush",{ "username" : inputUsername, "password" : inputPassword, "end" : "" ,}, function(data,status) {
                console.log("post success ",data,status);                
            })
            
        }

Other

rebeliousconformist, 2020-07-08:
@liuzfesp i have it enabled, but i am getting the same output as below. I dont know where the application is getting corrupted

> HI @rebeliousconformist the backtrace doesn't indicate too much useful info:
>
> $ xtensa-esp32-elf-addr2line -piaf -e rest.elf 0x40081fbd:0x3ffc27b0 0x0004001d:0x3ffc27d0
> 0x40081fbd: __wrap_free at [...]

rebeliousconformist, 2020-07-09:
I switched of HEAP Debugging and the error has disappeared.

> HI @rebeliousconformist the backtrace doesn't indicate too much useful info:
>
> $ xtensa-esp32-elf-addr2line -piaf -e rest.elf 0x40081fbd:0x3ffc27b0 0x0004001d:0x3ffc27d0
> 0x40081fbd: __wrap_free at [...]

No it is not an application issue. it is your framework which has a bug in its Heap Debugging Tool

iltis42, 2020-11-30:
Same thing here in my system:
==================== THREAD 1 (TCB: 0x3ffd9d54, name: 'wifi') =====================
#0  0x40081a48 in __wrap_free (p=0x3ff8105c) at /home/nextpcb/esp/esp-idf/components/heap/include/heap_trace.inc:162
#1  0x40040020 in ?? ()
#2  0x400976d9 in ppTask () at /home/nextpcb/esp/esp-idf/components/esp32/crosscore_int.c:115